openbmc/openbmc #2390: env: can't execute 'witherspoon-psu-monitor': No such file or directory

gkeishin, 2017-09-28:
`Journald snippet`
```
Sep 19 20:40:31 witherspoon systemd[1]: witherspoon-power-supply-monitor@0.service: Main process exited, code=exited, status=127/n/a
Sep 19 20:40:31 witherspoon systemd[1]: witherspoon-power-supply-monitor@0.service: Unit entered failed state.
Sep 19 20:40:31 witherspoon systemd[1]: witherspoon-power-supply-monitor@0.service: Failed with result 'exit-code'.
Sep 19 20:40:31 witherspoon systemd[1]: Started Wait for /org/openbmc/HostIpmi/1.
Sep 19 20:40:31 witherspoon systemd[1]: witherspoon-power-supply-monitor@1.service: Main process exited, code=exited, status=127/n/a
Sep 19 20:40:31 witherspoon systemd[1]: witherspoon-power-supply-monitor@1.service: Unit entered failed state.
Sep 19 20:40:31 witherspoon systemd[1]: witherspoon-power-supply-monitor@1.service: Failed with result 'exit-code'.
```

```
root@witherspoon:~# systemctl status witherspoon-power-supply-monitor@0.service | cat
● witherspoon-power-supply-monitor@0.service - Witherspoon Power Supply 0 Monitor
   Loaded: loaded (/lib/systemd/system/obmc-standby.target.requires/../witherspoon-power-supply-monitor@.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Tue 2017-09-19 20:40:31 UTC; 8min ago
  Process: 1274 ExecStart=/usr/bin/env witherspoon-psu-monitor --path=${DEVPATH} --instance=${INSTANCE} --inventory=${INVENTORY} (code=exited, status=127)
 Main PID: 1274 (code=exited, status=127)

Sep 19 20:40:29 witherspoon systemd[1]: Started Witherspoon Power Supply 0 Monitor.
Sep 19 20:40:30 witherspoon env[1274]: env: can't execute 'witherspoon-psu-monitor': No such file or directory
Sep 19 20:40:31 witherspoon systemd[1]: witherspoon-power-supply-monitor@0.service: Main process exited, code=exited, status=127/n/a
Sep 19 20:40:31 witherspoon systemd[1]: witherspoon-power-supply-monitor@0.service: Unit entered failed state.
Sep 19 20:40:31 witherspoon systemd[1]: witherspoon-power-supply-monitor@0.service: Failed with result 'exit-code'.
root@witherspoon:~#
```

`Driver Info`
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.10-60"
VERSION_ID="v1.99.10-60-g5e61b98"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.10-60"
BUILD_ID="v1.99.10"
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-00cd954c889746a65f9022088c5d83f9f43bfddc #1 Fri Sep 22 12:34:59 UTC 2017 armv6l GNU/Linux
root@witherspoon:~#

```
- [x] Journald 
[power-supply-exit.txt](https://github.com/openbmc/openbmc/files/1342101/power-supply-exit.txt)

gkeishin, 2017-09-28:
This one look like coming from this code drop https://github.com/openbmc/openbmc/commit/360ebeba3452a43086aa157b43789c18dca8c2b8

bjwyman, 2017-09-28:
Yes, that is due to the recipe updates for #1731 being merged prior to the code that builds the executable it needs. I thought it was clearly spelled out that it required updates from another repo, but somehow it got through.

I had a revert ready to push up, but the only "bad" consequence that seemed likely to occur was a trace similar to what you saw. Andrew pointed out the change passing HW CI.

I left yesterday without pushing up a revert.

gkeishin, 2017-09-29:
Thanks @bjwyman

bjwyman, 2017-10-02:
This should be resolved now that #1731 is really closed.
Round about here: https://github.com/openbmc/witherspoon-pfault-analysis/commit/aad73e92f9f9044a70c77e42d7ba635637e804fa
This: https://github.com/openbmc/openbmc/commit/ef940735e4d4caea90f5091d434590bbd6563ff4, is close enough, but not fully covering the #1731 function.

bjwyman, 2017-10-04:
60034dc
That commit should resolve the problem. Updates to have #1731 and #1732, and not flood the journal with sysfs read fail messages.

bjwyman, 2017-10-10:
v1.99.10-100-gb66a627
That build should be fixed.

closed 2017-10-11 by spinler