openbmc/openbmc #2394: SSLError: [SSL: BAD_LENGTH] unknown error

gkeishin, 2017-10-01:
Running a long run doing the following to mimic xcat checking for systems 
- [x] Boot to OS
- [x] Enumerate every 10 seconds
- [x] Check for error logs

This seems to happen when the BMC REST server is under pressure and presumably returns invalid ssl responses and crashing.

`Journald snapshot`
```
Sep 30 20:16:58 witherspoon phosphor-gevent[1387]:  'bottle.request.urlparts': SplitResult(scheme='https', netloc='wsbmc011.aus.stglabs.ibm.com', path='/xyz/openbmc_project/inventory/enumerate', query='', fragment=''),
Sep 30 20:16:58 witherspoon phosphor-gevent[1387]:  'bottle.route': <GET '<path:path>/enumerate' <obmc.wsgi.apps.rest_dbus.ListHandler object at 0x76563630>>,
Sep 30 20:16:58 witherspoon phosphor-gevent[1387]:  'route.handle': <GET '<path:path>/enumerate' <obmc.wsgi.apps.rest_dbus.ListHandler object at 0x76563630>>,
Sep 30 20:16:58 witherspoon phosphor-gevent[1387]:  'route.url_args': {'path': '/xyz/openbmc_project/inventory'},
Sep 30 20:16:58 witherspoon phosphor-gevent[1387]:  'wsgi.errors': <open file '<stderr>', mode 'w' at 0x76f640d0>,
Sep 30 20:16:58 witherspoon phosphor-gevent[1387]:  'wsgi.input': <gevent.pywsgi.Input object at 0x759517b0>,
Sep 30 20:16:58 witherspoon phosphor-gevent[1387]:  'wsgi.multiprocess': False,
Sep 30 20:16:58 witherspoon phosphor-gevent[1387]:  'wsgi.multithread': False,
Sep 30 20:16:58 witherspoon phosphor-gevent[1387]:  'wsgi.run_once': False,
Sep 30 20:16:58 witherspoon phosphor-gevent[1387]:  'wsgi.url_scheme': 'https',
Sep 30 20:16:58 witherspoon phosphor-gevent[1387]:  'wsgi.version': (1, 0)} failed with error
Sep 30 20:16:58 witherspoon phosphor-gevent[1387]: ::60d9:b976:8e91:f402%1128049428 - - [2017-09-30 20:16:58] "GET /xyz/openbmc_project/inventory/enumerate HTTP/1.1" socket - 3.829771
Sep 30 20:16:58 witherspoon phosphor-gevent[1387]: Traceback (most recent call last):
Sep 30 20:16:58 witherspoon phosphor-gevent[1387]:   File "/usr/lib/python2.7/site-packages/gevent/greenlet.py", line 327, in run
Sep 30 20:16:58 witherspoon phosphor-gevent[1387]:     result = self._run(*self.args, **self.kwargs)
Sep 30 20:16:58 witherspoon phosphor-gevent[1387]:   File "/usr/lib/python2.7/site-packages/gevent/server.py", line 103, in wrap_socket_and_handle
Sep 30 20:16:58 witherspoon phosphor-gevent[1387]:     return self.handle(ssl_socket, address)
Sep 30 20:16:58 witherspoon phosphor-gevent[1387]:   File "/usr/lib/python2.7/site-packages/gevent/pywsgi.py", line 658, in handle
Sep 30 20:16:58 witherspoon phosphor-gevent[1387]:     handler.handle()
Sep 30 20:16:58 witherspoon phosphor-gevent[1387]:   File "/usr/lib/python2.7/site-packages/gevent/pywsgi.py", line 184, in handle
Sep 30 20:16:58 witherspoon phosphor-gevent[1387]:     result = self.handle_one_request()
Sep 30 20:16:58 witherspoon phosphor-gevent[1387]:   File "/usr/lib/python2.7/site-packages/gevent/pywsgi.py", line 321, in handle_one_request
Sep 30 20:16:58 witherspoon phosphor-gevent[1387]:     self.handle_one_response()
Sep 30 20:16:58 witherspoon phosphor-gevent[1387]:   File "/usr/lib/python2.7/site-packages/gevent/pywsgi.py", line 515, in handle_one_response
Sep 30 20:16:58 witherspoon phosphor-gevent[1387]:     self.handle_error(*sys.exc_info())
Sep 30 20:16:58 witherspoon phosphor-gevent[1387]:   File "/usr/lib/python2.7/site-packages/gevent/pywsgi.py", line 528, in handle_error
Sep 30 20:16:58 witherspoon phosphor-gevent[1387]:     self.write(_INTERNAL_ERROR_BODY)
Sep 30 20:16:58 witherspoon phosphor-gevent[1387]:   File "/usr/lib/python2.7/site-packages/gevent/pywsgi.py", line 376, in write
Sep 30 20:16:58 witherspoon phosphor-gevent[1387]:     self._write(data)
Sep 30 20:16:58 witherspoon phosphor-gevent[1387]:   File "/usr/lib/python2.7/site-packages/gevent/pywsgi.py", line 369, in _write
Sep 30 20:16:58 witherspoon phosphor-gevent[1387]:     self._sendall(data)
Sep 30 20:16:58 witherspoon phosphor-gevent[1387]:   File "/usr/lib/python2.7/site-packages/gevent/pywsgi.py", line 355, in _sendall
Sep 30 20:16:58 witherspoon phosphor-gevent[1387]:     self.socket.sendall(data)
Sep 30 20:16:58 witherspoon phosphor-gevent[1387]:   File "/usr/lib/python2.7/site-packages/gevent/socket.py", line 458, in sendall
Sep 30 20:16:58 witherspoon phosphor-gevent[1387]:     data_sent += self.send(_get_memory(data, data_sent), flags)
Sep 30 20:16:58 witherspoon phosphor-gevent[1387]:   File "/usr/lib/python2.7/site-packages/gevent/ssl.py", line 192, in send
Sep 30 20:16:58 witherspoon phosphor-gevent[1387]:     v = self._sslobj.write(data)
Sep 30 20:16:58 witherspoon phosphor-gevent[1387]: SSLError: [SSL: BAD_LENGTH] unknown error (_ssl.c:1645)
Sep 30 20:16:58 witherspoon phosphor-gevent[1387]: <Greenlet at 0x759a6440: <bound method WSGIServer.wrap_socket_and_handle of <WSGIServer at 0x76563bb0 fileno=5 address=('::2cc2:3d43:d8cd:bb76%1992019400', 443, 0, 1992019400)>>(<socket at 0x75951c50 fileno=-1>, ('::60d9:b976:8e91:f402%1128049428', 59741, 0, 112)> failed with SSLError
```
```
{
	"__CURSOR" : "s=797b7e75de654c9ab9562d2d9446b6ed;i=41f7;b=c2e715e442ef44bc835686fecdeb0f78;m=92d21a70f;t=55a642526000d;x=d8a27d9f54bcedb3",
	"__REALTIME_TIMESTAMP" : "1506761050030093",
	"__MONOTONIC_TIMESTAMP" : "39411885839",
	"_BOOT_ID" : "c2e715e442ef44bc835686fecdeb0f78",
	"_TRANSPORT" : "stdout",
	"PRIORITY" : "6",
	"SYSLOG_FACILITY" : "3",
	"SYSLOG_IDENTIFIER" : "phosphor-gevent",
	"_PID" : "1387",
	"_UID" : "0",
	"_GID" : "0",
	"_COMM" : "python",
	"_EXE" : "/usr/bin/python2.7",
	"_CMDLINE" : "python /usr/sbin/phosphor-gevent rest_dbus",
	"_CAP_EFFECTIVE" : "3fffffffff",
	"_SYSTEMD_CGROUP" : "/system.slice/phosphor-gevent.service",
	"_SYSTEMD_UNIT" : "phosphor-gevent.service",
	"_SYSTEMD_SLICE" : "system.slice",
	"_SYSTEMD_INVOCATION_ID" : "1001716e58634d0080d9983daf597597",
	"_MACHINE_ID" : "5bf77ffe0452445b8701f894e75a90a7",
	"_HOSTNAME" : "witherspoon",
	"MESSAGE" : "SSLError: [SSL: BAD_LENGTH] unknown error (_ssl.c:1645)"
}
```

`Driver Info`
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="ibm-v1.99.10"
VERSION_ID="ibm-v1.99.10-0-r10-0-g243649d"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) ibm-v1.99.10"
BUILD_ID="ibm-v1.99.10-0-r10"
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-c4227b43277bbd0992d10d764d4958239786503d #1 Fri Sep 29 17:06:02 CDT 2017 armv6l GNU/Linux
root@witherspoon:~#
```

- [x] Journald  [SSl_ERROR_FFDC.txt](https://github.com/openbmc/openbmc/files/1347436/SSl_ERROR_FFDC.txt)

rfrandse, 2017-10-16:
<https://gerrit.openbmc-project.xyz/7383> openpower-software-manager: Add service to clear volatile partitions
Resolves  #2394 SSLError: [SSL: BAD_LENGTH] unknown error

anoo1, 2017-12-05:
Putting this bug back in the backlog. I mistakenly had a gerrit commit referencing this bug.

rfrandse, 2018-07-18:
Not seeing anymore.

closed 2018-07-18 by rfrandse