openbmc/openbmc #2397: Clear volatile PNOR sections on power offs

geissonator, 2017-10-02:
PNOR sections have a "volatile" flag which indicate it should be cleared outside of the automatic reboot path.  This story will be used 

https://github.com/open-power/pnor/blob/master/p9Layouts/defaultPnorLayout_128.xml
```
    <volatile/>     -> Indicates Partition is not preserved across power offs, 
                       but is across reboots. BMC will clear on power off/on
```

anoo1, 2017-10-09:
- [ ] Get partition with volatile flag and add this new flag to the pnor table of contents.
- [ ] Create function to get volatile partition and clear it.
- [ ] Call function during power off.

rfrandse, 2018-03-14:
<https://gerrit.openbmc-project.xyz/9544> Service files to clear volatile PNOR partitions on host reboot
Resolves  #2397 Clear volatile PNOR sections on power offs

closed 2018-04-25 by geissonator