openbmc/openbmc #2398: Allow in band IPMI to enable/disable automatic reboots

geissonator, 2017-10-02:
The host requires the ability to control https://github.com/openbmc/phosphor-dbus-interfaces/blob/master/xyz/openbmc_project/Control/Boot/RebootPolicy.interface.yaml
- AutoReboot

The team is working on defining a sensor for this, once done we just need to update host ipmid to associated the new sensor with this setting.

bjwyman, 2017-10-04:
# **STATUS**
**10/04**
- Today: Informed by Spinler that I am the lucky new owner of this. Started looking things over, assuming this has something to do with openbmc/phosphor-host-ipmid repo. Not sure how I will know when the new sensor for this is defined.
- Next: Figure out how to tell when the sensor is defined, and what all is entailed with updating host ipmid to associate the new sensor with the setting.
- Blockers: ???

bjwyman, 2017-10-05:
Entity ID: 0x21
Sensor Type: 0x09

Looks like one can confirm the definition via this commit: https://github.com/open-power/witherspoon-xml/commit/24d2f6d98568f0b23d1eb3bb9f876a638be4ae37

bjwyman, 2017-10-18:
Without my changes:
```
[v2cib530@gfwr703 ~]$ ipmitool -I lanplus -H $BMC -P 0penBmc raw 0x04 0x2D 0xda
Unable to send RAW command (channel=0x0 netfn=0x4 lun=0x0 cmd=0x2d rsp=0xcb): Requested sensor, data, or record not found
[v2cib530@gfwr703 ~]$
```
With my changes:
```
root@witherspoon:~# busctl tree xyz.openbmc_project.Settings --no-pager
└─/xyz
  └─/xyz/openbmc_project
    ├─/xyz/openbmc_project/control
    │ ├─/xyz/openbmc_project/control/host0
    │ │ ├─/xyz/openbmc_project/control/host0/TPMEnable
    │ │ ├─/xyz/openbmc_project/control/host0/auto_reboot
    │ │ ├─/xyz/openbmc_project/control/host0/boot
    │ │ │ └─/xyz/openbmc_project/control/host0/boot/one_time
    │ │ ├─/xyz/openbmc_project/control/host0/power_cap
    │ │ ├─/xyz/openbmc_project/control/host0/power_restore_policy
    │ │ └─/xyz/openbmc_project/control/host0/restriction_mode
    │ └─/xyz/openbmc_project/control/minimum_ship_level_required
    ├─/xyz/openbmc_project/network
    │ └─/xyz/openbmc_project/network/host0
    │   └─/xyz/openbmc_project/network/host0/intf
    │     └─/xyz/openbmc_project/network/host0/intf/addr
    └─/xyz/openbmc_project/time
      ├─/xyz/openbmc_project/time/owner
      └─/xyz/openbmc_project/time/sync_method
root@witherspoon:~# busctl introspect xyz.openbmc_project.Settings /xyz/openbmc_project/control/host0/auto_reboot  --no-pager
NAME                                          TYPE      SIGNATURE RESULT/VALUE FLAGS
org.freedesktop.DBus.Introspectable           interface -         -            -
.Introspect                                   method    -         s            -
org.freedesktop.DBus.Peer                     interface -         -            -
.GetMachineId                                 method    -         s            -
.Ping                                         method    -         -            -
org.freedesktop.DBus.Properties               interface -         -            -
.Get                                          method    ss        v            -
.GetAll                                       method    s         a{sv}        -
.Set                                          method    ssv       -            -
.PropertiesChanged                            signal    sa{sv}as  -            -
xyz.openbmc_project.Control.Boot.RebootPolicy interface -         -            -
.AutoReboot                                   property  b         true         emits-change writable
root@witherspoon:~#
root@witherspoon:~# busctl get-property xyz.openbmc_project.Settings /xyz/openbmc_project/control/host0/auto_reboot xyz.openbmc_project.Control.Boot.RebootPolicy AutoReboot
b true
root@witherspoon:~#

[v2cib530@gfwr703 ~]$ ipmitool -I lanplus -H $BMC -P 0penBmc raw 0x04 0x2d 0xda
 00 00 02 00
[v2cib530@gfwr703 ~]$

root@witherspoon:~# busctl set-property xyz.openbmc_project.Settings /xyz/openbmc_project/control/host0/auto_reboot xyz.openbmc_project.Control.Boot.RebootPolicy AutoReboot b false
root@witherspoon:~# busctl get-property xyz.openbmc_project.Settings /xyz/openbmc_project/control/host0/auto_reboot xyz.openbmc_project.Control.Boot.RebootPolicy AutoReboot
b false
root@witherspoon:~#

[v2cib530@gfwr703 ~]$ ipmitool -I lanplus -H $BMC -P 0penBmc raw 0x04 0x2d 0xda
 00 00 01 00
[v2cib530@gfwr703 ~]$
```

bjwyman, 2017-10-18:
Did some testing with @mderkse1 with a HostBoot patch. Somehow a flag gets set, and if that is set, certain code paths will disable AutoReboot in order to stop on errors. The AutoReboot was set to true, but then after a poweron, it was set to false, so apparently the path from HostBoot to the BMC is working to change the AutoReboot value.

rfrandse, 2017-10-19:
<https://gerrit.openbmc-project.xyz/7301> AutoReboot: Add support for AutoReboot sensor type
Resolves:  #2398 Allow in band IPMI to enable/disable automatic reboots

closed 2017-10-26 by geissonator