openbmc/openbmc #2400: MCBISTFIR2/MBCALFIR1 checkstop during IPL not working

pandusiva, 2017-10-03:
Injected checkstop bits during IPL memdiag. System checkstopped as expected and issued reboot. but did not get expected error logs for the error injected. Getting below "no active bits found log for checkstop" . Looks like error is not isolated to checkstop or all FFDC data not collected before reipl. 

 | PRD SRC Type             : PRD Detected Hardware Indication                  |
 | PRD SRC Class            : Software likely caused a hardware error condition,|
 |                          :  smaller possibility of a hardware cause.         |
 |                                                                              |
 | PRD Signature            : 0x00250000 0x7917DD02                             |
 | Signature Description    : mcb(n0p0c0) No active error bits found            |
 |------------------------------------------------------------------------------|
 |                              User Defined Data                               |
 |------------------------------------------------------------------------------|
 | Section Version          : 1                                                 |
 | Sub-section type         : 6                                                 |
 | Created by               : errl                                              |
 | Callout type             : Hardware Callout                                  |
 | Target                   : Physical:/Sys0/Node0/Proc0/MCBIST0                |
 | CPU id                   : 0x3                                               |
 | Deconfig State           : NO_DECONFIG                                       |
 | GARD Error Type          : GARD_NULL                                         |
 | Priority                 : SRCI_PRIORITY_LOW                                 |

Attaching esel for reference.
[esel_r217.out_MCBISTbit2_did_notwork.txt](https://github.com/openbmc/openbmc/files/1352063/esel_r217.out_MCBISTbit2_did_notwork.txt)

mdmillerii, 2017-11-21:
This is an issue with host error log analysis and should be in a different repository.

anoo1, 2018-07-19:
If this is still an issue please open it against https://github.com/open-power/hostboot

closed 2018-07-19 by anoo1