openbmc/openbmc #2402: Logging exited with 'core-dump' during boot to standby

gkeishin, 2017-10-03:
`Journald snippet`
```
Sep 19 20:40:53 witherspoon phosphor-log-manager[1013]: Failed to read 4 bytes from input stream! Read 0
Sep 19 20:40:53 witherspoon phosphor-log-manager[1013]: Failed to read 4 bytes from input stream! Read 0
Sep 19 20:40:53 witherspoon systemd[1]: xyz.openbmc_project.Logging.service: Main process exited, code=dumped, status=11/SEGV
Sep 19 20:40:53 witherspoon systemd[1]: Failed to start Phosphor Log Manager.
Sep 19 20:40:53 witherspoon systemd[1]: xyz.openbmc_project.Logging.service: Unit entered failed state.
Sep 19 20:40:53 witherspoon systemd[1]: xyz.openbmc_project.Logging.service: Failed with result 'core-dump'.
Sep 19 20:40:53 witherspoon systemd[1]: Started Phosphor Dump Manager.
Sep 19 20:40:54 witherspoon systemd[1]: xyz.openbmc_project.Logging.service: Service hold-off time over, scheduling restart.
```
`Pretty print`
```
{
	"__CURSOR" : "s=cca7a9dc9b0f4a2982c4e77f792c1401;i=229;b=fea1979d625f49f0ba1de3b2660232b7;m=2952f54;t=55990e018fffa;x=666271be40caa22a",
	"__REALTIME_TIMESTAMP" : "1505853653450746",
	"__MONOTONIC_TIMESTAMP" : "43331412",
	"_BOOT_ID" : "fea1979d625f49f0ba1de3b2660232b7",
	"_MACHINE_ID" : "69e161dbbf234a53b357e714ae1489dc",
	"_HOSTNAME" : "witherspoon",
	"PRIORITY" : "4",
	"SYSLOG_FACILITY" : "3",
	"SYSLOG_IDENTIFIER" : "systemd",
	"_UID" : "0",
	"_GID" : "0",
	"_TRANSPORT" : "journal",
	"_CAP_EFFECTIVE" : "3fffffffff",
	"_SYSTEMD_SLICE" : "-.slice",
	"_PID" : "1",
	"_COMM" : "systemd",
	"_EXE" : "/lib/systemd/systemd",
	"_CMDLINE" : "/sbin/init",
	"_SYSTEMD_CGROUP" : "/init.scope",
	"_SYSTEMD_UNIT" : "init.scope",
	"UNIT" : "xyz.openbmc_project.Logging.service",
	"CODE_FILE" : "../git/src/core/service.c",
	"CODE_LINE" : "1470",
	"CODE_FUNCTION" : "service_enter_dead",
	"MESSAGE" : "xyz.openbmc_project.Logging.service: Failed with result 'core-dump'.",
	"_SOURCE_REALTIME_TIMESTAMP" : "1505853653388833"
}
```

`Driver info`
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.10-85"
VERSION_ID="v1.99.10-85-g7b8e664"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.10-85"
BUILD_ID="v1.99.10"
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-00cd954c889746a65f9022088c5d83f9f43bfddc #1 Tue Sep 26 18:05:33 UTC 2017 armv6l GNU/Linux
root@witherspoon:~# cat /etc/timestamp
20171002204212
root@witherspoon:~#
```

- [x] Jounrald log 
[logging_crashed_standby.txt](https://github.com/openbmc/openbmc/files/1352441/logging_crashed_standby.txt)

ojayanth, 2017-10-03:
@gkeishin problem is related to incomplete cereal generated files. 
 Please provide details of  previous ipl type inf.

gkeishin, 2017-10-03:
- [x] Code update to latest Master build 
- [x] Power on

that's the step..

ojayanth, 2017-10-03:
Problem could be same as  https://github.com/openbmc/openbmc/issues/2337.
We usually see this issue , incase some one issue reboot during ceral related data save.

gkeishin, 2017-10-16:
@ojayanth I dont see it fixed with the above in the latest Master..

ngorugan, 2017-10-27:
I have run code update jenkin job with images provided  by @gkeishin, but unable to recreate this issue.    -Tried  with elogs created first then updated the code.
-One observation: log entries are not getting restored  after code update.

gkeishin, 2017-10-27:
That's a good observation. perhaps we can try fix that @ngorugan

ojayanth, 2017-10-27:
@ngorugan Please create new issue related code update
@gkeishin As discussed yesterday please close this issue.

ojayanth, 2017-10-27:
@ngorugan Please check this particular you found is related cereal version support related code upgrade..

Code update from version unsupported Cereal version  driver to   cereal un-supported version to version enabled driver this is  expected behavior.

ngorugan, 2017-10-30:
@gkeishin please close this issue. I will open one more issue for "log entries are not getting after code update."

gkeishin, 2017-10-30:
Let me know the issue number and I close this one up @ngorugan  Thanks

ngorugan, 2017-10-30:
Issue#2532 is opened @gkeishin

closed 2017-10-30 by gkeishin