openbmc/openbmc #2405: sbefifo, BUS access failure when attempted from the sbe_interfaces library code(openpower-sbe-interface))

nmuruli, 2017-10-03:
I would like Christopher Bostic or Edward James help me to get pass through the below issue.

When I am attempting to test run my test my SBE interface library which provides interfaces to access the SBE via SBE FIFO device driver , I see that I consistently get BUS access failure.

Steps used:
1) obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.On
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Running

2)root@witherspoon:/tmp# ./sbei_inter
Poll for write completed    ====> Polling even for  POLLOUT | POLLERR;
Writing 16 bytes completed   ==> I see the write passed on /dev/sbefifo1
Polling for the read    
Hit a POLL Error errno:0x0     ==> But we fail when we are polling for the POLLIN|POLLERR event.

dmesg: (I see bus access failure)
[56442.640000]  slave@00:00: status: 0xc4000501, sisc: 0x08040008
[56442.660000]  slave@00:00: status: 0xc4000501, sisc: 0x08040008
[56935.250000] sbefifo 00:00:00:06: Fatal bus access failure: -5
[56935.250000] occ-hwmon occ1-dev0: occ bus op failed rc:-5
"dmsg_2" 187L, 13473C                                                                                                                                                      

Full Dmsg file is at : /afs/rch/usr5/murunata/public/dmsgFile

STRACE Output:
root@witherspoon:/tmp# bin/strace -e trace=open,write,read,poll ./sbei_inter
open("/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
open("/usr/lib/libstdc++.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\3\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0,\333\6\0004\0\0\0"..., 512) = 512
open("/lib/libgcc_s.so.1", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\240\322\0\0004\0\0\0"..., 512) = 512
open("/lib/tls/v6l/libc.so.6", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/lib/tls/libc.so.6", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/lib/v6l/libc.so.6", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/lib/libc.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0 h\1\0004\0\0\0"..., 512) = 512
open("/lib/libm.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\220=\0\0004\0\0\0"..., 512) = 512
open("/dev/sbefifo1", O_RDWR|O_NONBLOCK) = 3
poll([{fd=3, events=POLLOUT|POLLERR}], 1, -1) = 1 ([{fd=3, revents=POLLOUT}])
write(3, "\4\0\0\0\0\0\242\1\0\0\0\0\0 \20\7", 16) = 16                                                      ==================> I see the write passed
poll([{fd=3, events=POLLIN|POLLERR}], 1, -1) = 1 ([{fd=3, revents=POLLERR}])=================> This is line 76 in the image below
write(1, "Hit a POLL Error errno:0x0\n", 27Hit a POLL Error errno:0x0                       =================>  This is line 87 in the image below
) = 27

https://gerrit.openbmc-project.xyz/#/c/5210/12/sbe_chipOp_handler.cpp  Line 71 I am hitting the failure

cbostic, 2017-10-20:
@rfrandse this is a duplicate of https://github.com/openbmc/openbmc/issues/2405 can you please close this one?

amboar, 2017-10-22:
@nmuruli A few comments on your use of the github issue tracker to help you and everyone use it effectively:

First, please use the triple back-tick markdown when pasting code or console output into github issues. It prevents github's rendering engine interpreting your log output as markdown, visually separates it from your own comments, and makes things look tidy. What you write looks like:

    ```
    My code, console or log output here
    ```

A guide to using github-flavoured markdown is here:

https://guides.github.com/features/mastering-markdown/

Second, this is a public bug tracker. If you have something to add to it, please attach it directly to the issue. Statements such as:

> Full Dmsg file is at : /afs/rch/usr5/murunata/public/dmsgFile

Are fundamentally not useful, as this file is only reachable by (IBM) internal methods.

Third, instead of highlighting issues in output with markers such as ` =================>`, please use the formatting options to make quotes of the output and make your comments below. See the link on mastering github-flavoured markdown above.

Cheers,

Andrew

amboar, 2017-10-22:
> @rfrandse this is a duplicate of #2405 can you please close this one?

@cbostic this *is* #2405.

amboar, 2017-10-22:
@cbostic having looked through the issue tracker, I assume you meant #2343?

cbostic, 2017-10-23:
@rfrandse @amboar  My mistake, yes I meant this is a duplicate of #2343

rfrandse, 2017-11-06:
dup of  #2343

closed 2017-11-06 by rfrandse