openbmc/openbmc #2412: STC:OP910:Witherspoon: System shutdown for nonfunctional fans, however fans are functional

cdeadmin, 2017-10-03:


msbarth, 2017-10-03:
This defect covers 2 issues in the same area of code within phosphor-dbus-monitor
- `The inventory is nonfunctional` is being logged for fans after they've returned back to being functional
- The shutdown timer is not cancelling upon enough fans returning back to functional status

First issue:
```
Oct 02 02:13:02 witherspoon phosphor-dbus-monitor[1165]: The inventory is nonfunctional.
Oct 02 02:13:03 witherspoon phosphor-dbus-monitor[1165]: The inventory is nonfunctional.
Oct 02 02:13:03 witherspoon phosphor-dbus-monitor[1165]: The inventory is nonfunctional.
Oct 02 02:13:03 witherspoon phosphor-fan-monitor[3151]: Setting a fan back to functional
Oct 02 02:13:03 witherspoon phosphor-fan-monitor[3151]: Setting a fan back to functional
Oct 02 02:13:04 witherspoon phosphor-fan-monitor[3151]: Setting a fan back to functional
Oct 02 02:13:04 witherspoon phosphor-dbus-monitor[1165]: The inventory is nonfunctional.
Oct 02 02:13:05 witherspoon phosphor-dbus-monitor[1165]: The inventory is nonfunctional.
Oct 02 02:13:06 witherspoon phosphor-dbus-monitor[1165]: The inventory is nonfunctional.
Oct 02 02:13:08 witherspoon phosphor-dbus-monitor[1165]: Shutting down system.  There are not enough functional fans.
```

Second issue
```
Oct 02 12:56:39 witherspoon timemanager[1269]: Request to get HOST time
Oct 02 12:56:39 witherspoon timemanager[1269]: Mon Oct  2 12:56:39 2017 GMT
Oct 02 12:56:39 witherspoon timemanager[1269]:  Host_time_str: [ Mon Oct  2 12:56:39 2017 GMT ] host_time usec: [ 1506948999231301 ] host_offset: [ 0 ]
Oct 02 13:02:13 witherspoon phosphor-fan-monitor[3151]: Setting a fan to nonfunctional
Oct 02 13:02:14 witherspoon phosphor-dbus-monitor[1137]: The inventory is nonfunctional.
Oct 02 13:02:14 witherspoon phosphor-fan-monitor[3151]: Setting a fan to nonfunctional
Oct 02 13:02:14 witherspoon dbus[994]: [system] Activating via systemd: service name='org.freedesktop.journal1' unit='dbus-org.freedesktop.journal1.service'
Oct 02 13:02:14 witherspoon systemd[1]: Starting Journal Dbus Service...
Oct 02 13:02:14 witherspoon phosphor-fan-monitor[3151]: Setting a fan to nonfunctional
Oct 02 13:02:15 witherspoon dbus[994]: [system] Successfully activated service 'org.freedesktop.journal1'
Oct 02 13:02:15 witherspoon systemd[1]: Started Journal Dbus Service.
Oct 02 13:02:15 witherspoon phosphor-fan-monitor[3151]: Setting a fan back to functional
Oct 02 13:02:16 witherspoon phosphor-fan-monitor[3151]: Setting a fan back to functional
Oct 02 13:02:18 witherspoon phosphor-fan-monitor[3151]: Setting a fan back to functional
Oct 02 13:02:20 witherspoon phosphor-log-manager[957]: Failed to find metadata
Oct 02 13:02:20 witherspoon phosphor-fru-fault-monitor[1186]: /xyz/openbmc_project/logging/entry/9
Oct 02 13:02:20 witherspoon phosphor-dbus-monitor[1137]: The inventory is nonfunctional.
Oct 02 13:02:22 witherspoon phosphor-ledmanager[956]: Asserting LEDs
Oct 02 13:02:24 witherspoon phosphor-log-manager[957]: Failed to find metadata
Oct 02 13:02:24 witherspoon phosphor-fru-fault-monitor[1186]: /xyz/openbmc_project/logging/entry/10
Oct 02 13:02:24 witherspoon phosphor-dbus-monitor[1137]: The inventory is nonfunctional.
Oct 02 13:02:26 witherspoon phosphor-ledmanager[956]: Asserting LEDs
Oct 02 13:02:27 witherspoon phosphor-log-manager[957]: Failed to find metadata
Oct 02 13:02:27 witherspoon phosphor-fru-fault-monitor[1186]: /xyz/openbmc_project/logging/entry/11
Oct 02 13:02:28 witherspoon systemd[1]: Stopped target Start Host0.
Oct 02 13:02:28 witherspoon systemd[1]: Stopped PCIE Slot Presence Detection For Chassis0.
Oct 02 13:02:28 witherspoon systemd[1]: Stopped target Host0 (Started).
Oct 02 13:02:28 witherspoon systemd[1]: Stopped target Host0 (Starting).
Oct 02 13:02:28 witherspoon systemd[1]: Stopped Start host0 SBE.
Oct 02 13:02:28 witherspoon systemd[1]: Stopped target Start Host0 (Pre).
Oct 02 13:02:28 witherspoon systemd[1]: Stopped target Chassis0 (Power On).
Oct 02 13:02:28 witherspoon systemd[1]: Stopping Phosphor Fan Presence Tach Daemon...
Oct 02 13:02:28 witherspoon systemd[1]: Stopped Power on bind ir35221 device driver devs.
Oct 02 13:02:28 witherspoon systemd[1]: Stopped Witherspoon Fan Watchdog Reset.
Oct 02 13:02:28 witherspoon systemd[1]: Stopped Turn VCS on.
Oct 02 13:02:28 witherspoon systemd[1]: Stopped Run VCS workaround on host0.
Oct 02 13:02:28 witherspoon systemd[1]: Stopped Scan FSI hub device.
Oct 02 13:02:28 witherspoon systemd[1]: Stopped Scan FSI devices.
Oct 02 13:02:28 witherspoon systemd[1]: Starting OpenPOWER OCC Active Disable...
Oct 02 13:02:28 witherspoon systemd[1]: Starting Soft power off of the host...
Oct 02 13:02:28 witherspoon phosphor-dbus-monitor[1137]: Shutting down system.  There are not enough functional fans.
Oct 02 13:02:28 witherspoon phosphor-dbus-monitor[1137]: The inventory is nonfunctional.
```

rfrandse, 2017-10-27:
sueb on 26 October 2017 
can not re-create, closing defect

spinler, 2017-10-27:
Reopening because there is still an issue with dbus-monitor.

spinler, 2017-11-16:
This seems to have went away with the  latest fixes to the fan rules, but maybe more testing is warranted:

```
Nov 16 22:35:25 witherspoon phosphor-fan-monitor[4452]: Setting a fan to nonfunctional
Nov 16 22:35:25 witherspoon phosphor-dbus-monitor[3851]: The inventory is nonfunctional.
Nov 16 22:35:26 witherspoon phosphor-fan-monitor[4452]: Setting a fan to nonfunctional
Nov 16 22:35:26 witherspoon phosphor-fan-monitor[4452]: Setting a fan to nonfunctional
Nov 16 22:35:27 witherspoon phosphor-log-manager[982]: Failed to find metadata
Nov 16 22:35:27 witherspoon phosphor-fru-fault-monitor[1217]: /xyz/openbmc_project/logging/entry/16
Nov 16 22:35:27 witherspoon phosphor-dbus-monitor[3851]: The inventory is nonfunctional.
Nov 16 22:35:28 witherspoon phosphor-log-manager[982]: Failed to find metadata
Nov 16 22:35:28 witherspoon phosphor-fru-fault-monitor[1217]: /xyz/openbmc_project/logging/entry/17
Nov 16 22:35:28 witherspoon phosphor-dbus-monitor[3851]: The inventory is nonfunctional.
Nov 16 22:35:29 witherspoon phosphor-fan-monitor[4452]: Setting a fan back to functional
Nov 16 22:35:29 witherspoon phosphor-log-manager[982]: Failed to find metadata
Nov 16 22:35:29 witherspoon phosphor-fru-fault-monitor[1217]: /xyz/openbmc_project/logging/entry/18
Nov 16 22:35:30 witherspoon phosphor-fan-monitor[4452]: Setting a fan back to functional
Nov 16 22:35:30 witherspoon phosphor-fan-monitor[4452]: Setting a fan back to functional

```
So here, the system stayed up after all fans were nonfunctional, add errors logged against them, and then became functional again.

spinler, 2017-11-17:
Closing.  Can't recreate.

closed 2017-11-17 by spinler