openbmc/openbmc #2413: BMC failed to boot on reset

gkeishin, 2017-10-04:
- [x] Update system with https://openpower.xyz/job/openbmc-build/1025/
- [x] BMC reaches standby
- [x] Reset BMC and it hangs in boot

`Serial console log`
```
         Mounting /var...
[  OK  ] Mounted /usr/local.
[   22.950000] UBIFS (ubi0:2): background thread "ubifs_bgt0_2" started, PID 654
[  OK  ] Mounted /media.
[   23.170000] UBIFS (ubi0:2): UBIFS: mounted UBI device 0, volume 2, name "rwfs"
[   23.170000] UBIFS (ubi0:2): LEB size: 65408 bytes (63 KiB), min./max. I/O unit sizes: 8 bytes/256 bytes
[   23.170000] UBIFS (ubi0:2): FS size: 3597440 bytes (3 MiB, 55 LEBs), journal size 719488 bytes (0 MiB, 11 LEBs)
[   23.170000] UBIFS (ubi0:2): reserved for root: 0 bytes (0 KiB)
[   23.170000] UBIFS (ubi0:2): media format: w4/r0 (latest is w5/r0), UUID 99D6E8FF-58C0-4873-BFB5-D379BE96B8E3, small LPT model
[  OK  ] Mounted /var.
         Mounting /var/volatile...
         Starting Prepare for overlay fs...
         Starting Flush Journal to Persistent Storage...
[  OK  ] Mounted /var/volatile.
[  OK  ] Started udev Kernel Device Manager.
         Starting Load/Save Random Seed...
[   24.490000] systemd-journald[539]: Received request to flush runtime journal from PID 1
[  OK  ] Started Flush Journal to Persistent Storage.
[  OK  ] Started Load/Save Random Seed.
[  OK  ] Started udev Coldplug all Devices.
[  OK  ] Started Prepare for overlay fs.
[  OK  ] Found device /dev/aspeed-lpc-ctrl.
[  OK  ] Found device /dev/aspeed-mbox.
[  OK  ] Found device /dev/ttyS4.
[  OK  ] Found device /dev/ttyVUART0.
         Starting Prepare for overlay fs...
[  OK  ] Created slice system-xyz.openbmc_project.Hwmon.slice.
[  OK  ] Started Prepare for overlay fs.
         Starting Prepare for overlay fs...
[  OK  ] Started Prepare for overlay fs.
[  OK  ] Created slice system-xyz.openbmc_project.led.controller.slice.
[FAILED] Failed to start Prepare for overlay fs.
See 'systemctl status prepare-overlay.service' for details.
[DEPEND] Dependency failed for /etc.
[DEPEND] Dependency failed for Mount machine-id.
[DEPEND] Dependency failed for Local File Systems.
[DEPEND] Dependency failed for /home.
[DEPEND] Dependency failed for Emergency Shell.
[DEPEND] Dependency failed for Emergency Mode.
[  OK  ] Reached target Timers.
[FAILED] Failed to start Prepare for overlay fs.
See 'systemctl status prepare-overlay.service' for details.
[  OK  ] Closed Syslog Socket.
[  OK  ] Reached target Login Prompts.
[  OK  ] Reached target Host0 running after reset.
[  OK  ] Reached target Chassis0 power on after reset.
[  OK  ] Reached target Phosphor Object Mapper.
[  OK  ] Reached target Webserver (pre).
[  OK  ] Reached target Sockets.
         Starting Network Service...
         Starting Create Volatile Files and Directories...
[FAILED] Failed to start Create Volatile Files and Directories.
See 'systemctl status systemd-tmpfiles-setup.service' for details.
         Starting Network Time Synchronization...
[  OK  ] Started Network Time Synchronization.
[  OK  ] Reached target System Time Synchronized.
[  OK  ] Started Network Service.
[  OK  ] Reached target Network.
[   30.300000] 8021q: adding VLAN 0 to HW filter on device eth0
[   34.020000] ftgmac100 1e660000.ethernet eth0: NCSI interface up

````

gkeishin, 2017-10-04:
- [x] BMC hangs serial output 
[BMC_failed_boot_standby.txt](https://github.com/openbmc/openbmc/files/1354422/BMC_failed_boot_standby.txt)

geissonator, 2017-10-04:
We saw this in CI when some changes from @mtritz came through related to factory reset I believe.  Starting with @anoo1 to take an initial look.

gkeishin, 2017-10-04:
PDU recovers from the hang. continuing other tests.

saqibkh, 2017-10-16:
I verified this with the latest master and we don't see any boot hangs.
@rfrandse please close this defect as it is not reproducible. Thanks

gkeishin, 2017-10-16:
Sure I ll close it up.. If we see this one ahead..

closed 2017-10-16 by gkeishin

sivassrr, 2017-11-02:
SW406842 - Check this one. Symptom looks same.

amboar, 2017-11-13:
@sivassrr please do not reference IBM internal issue identifiers in the public bug tracker. It's completely meaningless to anyone outside of IBM. If the information is relevant, please copy it here.