openbmc/openbmc #2414: openpower-read-vpd: PIM Notify() failed

gkeishin, 2017-10-04:
`Journald sinppet`
```
openpower-read-vpd[1164]: PIM Notify() failed
```
`Driver Info`
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.10-95"
VERSION_ID="v1.99.10-95-g274f25a"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.10-95"
BUILD_ID="v1.99.10"
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-a4c9af76ceec1d7f638da84dc8a0b8b7b3adb8ac #1 Thu Sep 28 02:20:28 UTC 2017 armv6l GNU/Linux
root@witherspoon:~#
```
- [x] Journald 
[PIM_VPD_read_failure.txt](https://github.com/openbmc/openbmc/files/1355736/PIM_VPD_read_failure.txt)

gkeishin, 2017-10-16:
Keep seeing this in latter build too.. FYI

```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.10-156"
VERSION_ID="v1.99.10-156-ga1a81a0"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.10-156"
BUILD_ID="v1.99.10"
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-703c3c181790b20c48c4d02b9fed901379671fb7 #1 Mon Oct 16 16:32:33 UTC 2017 armv6l GNU/Linux
root@witherspoon:~# cat /etc/timestamp
20171016163705
root@witherspoon:~#
```

dkodihal, 2017-10-19:
This is seen on a specific Witherspoon that has bad VPD.

closed 2017-10-19 by dkodihal