openbmc/openbmc #2415: ipmid getHostNetworkData failed for get_sys_boot_options.

gkeishin, 2017-10-04:
- [x] Power on the host to boot OS

and we see this. The OS does boot eventually but we would need to see why this error

Ad per our @ratagupt we need to ask from the Host that why they keep sending the request when BMC have indicated that nothing is configured.

[18:49] 
So during poweron the box...Host asks from the BMC that get me the host network data

`Journald snippet`

```
Sep 19 20:44:14 witherspoon ipmid[1261]: getHostNetworkData failed for get_sys_boot_options.
Sep 19 20:44:14 witherspoon ipmid[1261]: ERROR:[0xFF] handling NetFn:[0x0], Cmd:[0x9]
Sep 19 20:44:14 witherspoon ipmid[1261]: Failed to call the method: Input/output error
Sep 19 20:44:14 witherspoon ipmid[1261]: Dest: :1.10, Path: /org/openbmc/HostIpmi/1
Sep 19 20:44:23 witherspoon ipmid[1261]: getHostNetworkData failed for get_sys_boot_options.
Sep 19 20:44:23 witherspoon ipmid[1261]: ERROR:[0xFF] handling NetFn:[0x0], Cmd:[0x9]
Sep 19 20:44:24 witherspoon ipmid[1261]: Failed to call the method: Input/output error
Sep 19 20:44:24 witherspoon ipmid[1261]: Dest: :1.10, Path: /org/openbmc/HostIpmi/1
Sep 19 20:44:33 witherspoon ipmid[1261]: getHostNetworkData failed for get_sys_boot_options.
Sep 19 20:44:33 witherspoon ipmid[1261]: ERROR:[0xFF] handling NetFn:[0x0], Cmd:[0x9]
Sep 19 20:44:34 witherspoon ipmid[1261]: Failed to call the method: Input/output error
Sep 19 20:44:34 witherspoon ipmid[1261]: Dest: :1.10, Path: /org/openbmc/HostIpmi/1
Sep 19 20:44:36 witherspoon ipmid[1261]: Failed to call the method: Input/output error
Sep 19 20:44:37 witherspoon ipmid[1261]: Dest: :1.10, Path: /org/openbmc/HostIpmi/1
Sep 19 20:44:49 witherspoon ipmid[1261]: getHostNetworkData failed for get_sys_boot_options.
Sep 19 20:44:49 witherspoon ipmid[1261]: ERROR:[0xFF] handling NetFn:[0x0], Cmd:[0x9]
Sep 19 20:44:50 witherspoon ipmid[1261]: Failed to call the method: Input/output error
Sep 19 20:44:50 witherspoon ipmid[1261]: Dest: :1.10, Path: /org/openbmc/HostIpmi/1
Sep 19 20:44:50 witherspoon ipmid[1261]: Failed to call the method: Input/output error
Sep 19 20:44:50 witherspoon ipmid[1261]: Dest: :1.10, Path: /org/openbmc/HostIpmi/1
Sep 19 20:44:50 witherspoon ipmid[1261]: Failed to call the method: Input/output error
Sep 19 20:44:50 witherspoon ipmid[1261]: Dest: :1.10, Path: /org/openbmc/HostIpmi/1
Sep 19 20:44:51 witherspoon ipmid[1261]: No Registered handlers for NetFn:[0x3A],Cmd:[0xFF]
Sep 19 20:44:51 witherspoon ipmid[1261]: ERROR:[0xC1] handling NetFn:[0x3A], Cmd:[0x1A]
Sep 19 20:45:18 witherspoon ipmid[1261]: getHostNetworkData failed for get_sys_boot_options.
Sep 19 20:45:18 witherspoon ipmid[1261]: ERROR:[0xFF] handling NetFn:[0x0], Cmd:[0x9]
Sep 19 20:45:18 witherspoon ipmid[1261]: Failed to call the method: Input/output error
Sep 19 20:45:18 witherspoon ipmid[1261]: Dest: :1.10, Path: /org/openbmc/HostIpmi/1
```

`Driver Info`
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.10-95"
VERSION_ID="v1.99.10-95-g274f25a"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.10-95"
BUILD_ID="v1.99.10"
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-a4c9af76ceec1d7f638da84dc8a0b8b7b3adb8ac #1 Thu Sep 28 02:20:28 UTC 2017 armv6l GNU/Linux
root@witherspoon:~# cat /media/pnor-ro-27b41bda/VERSION
IBM-witherspoon-sequoia-ibm-OP9_v1.19_1.41
        op-build-v1.7-1383-g487ffd1
        buildroot-2017.08-6-g319c6e1
        skiboot-v5.8-90-g398a98123641-pb7fe7a5
        hostboot-3d6c541
        linux-4.13-openpower1-p1b975d8
        petitboot-v1.5.1-p65d605b
        machine-xml-f8aaa73-p03357dc
        occ-a43395b
        hostboot-binaries-edd568f
        capp-ucode-9c73e9f
        sbe-8d90ab2
root@witherspoon:~#
```

- [x] Journald 
[IPMI_getnetwork_error.txt](https://github.com/openbmc/openbmc/files/1356141/IPMI_getnetwork_error.txt)

ratagupt, 2017-10-05:
@shenki :-Did we make any change recently in this path,,I thought it was fixed before..BMC is returning the IPMI_UNSPECIFIED_ERROR..But host is keep sending the request..

sammj, 2017-10-05:
Petitboot definitely only checks the IPMI network override option once during init, but it's possible that the `discover/ipmi` code is polling a few times waiting for a certain response. I'll have a look.

sammj, 2017-10-05:
Indeed we can see that Petitboot sees an error in the response and moves on:
```
IPMI: netfn(0->1), cmd(9->9)
IPMI net override resp [0][27]:
 ff 01 61 80 21 70 62 21  00 01 06 04 00 00 00 00
 00 00 00 00 00 00 00 00  00 00 00
platform: non-zero completion code 255 from IPMI network req
```

sammj, 2017-10-05:
`Sep 19 20:44:50 witherspoon ipmid[1261]: Failed to call the method: Input/output error`
Actually that's a little interesting. From ipmid.cpp:
```
    // Call the IPMI responder on the bus so the message can be sent to the CEC
    r = sd_bus_call(bus, m, 0, &error, &reply);
    if (r < 0) {
        fprintf(stderr, "Failed to call the method: %s\n", strerror(-r));
        fprintf(stderr, "Dest: %s, Path: %s\n", dest, path);
        goto final;
    }
```

Is this occurring since OpenBMC is failing to send the response?

ratagupt, 2017-10-24:
It is a false positive,I have seen through IPMItool that BMC do send the return code which ipmitool sees.
pasting the response below.

root@witherspoon:/tmp# /tmp/ipmitool -I dbus raw 0x00 0x09 0x61 0x80 0x00
Unable to send RAW command (channel=0x0 netfn=0x0 lun=0x0 cmd=0x9 rsp=0xff): Unspecified error

ratagupt, 2017-10-24:
in the above case it is failing as the Ip address and gatway is set to 0.0.0.0.

closed 2017-10-24 by dkodihal