openbmc/openbmc #2416: Need a method to create an internal error in error logs

rahulmah, 2017-10-04:
Need a method to create an internal error in error logs. This is required for dump testing. My main intention is to see a dump getting created once this entry is logged.

Development support for the dump to get created is already in place. All we need is a method to generate an internal error.

rahulmah, 2017-10-04:
@mtritz : As discussed, here is the task which we need for our testing.

rfrandse, 2017-10-05:
<https://gerrit.openbmc-project.xyz/7076> Add InternalFailure to logging-test
Resolves  #2416 Need a method to create an internal error in error logs

rahulmah, 2017-10-10:
@mtritz : To be exact, we want to see below error getting logged. 
`Message = xyz.openbmc_project.Common.Error.InternalFailure`

rahulmah, 2017-10-13:
@rfrandse :  As discussed, this is something we need for verifying internal failure dump. Please move this up in the priority list.

mtritz, 2017-12-13:
commit is actually here - https://gerrit.openbmc-project.xyz/#/c/7389/

closed 2018-07-18 by rfrandse