openbmc/openbmc #2417: MFG:Witherspoon, "Common.Error.InternalFailureError" SEL logged after flashing 1738D

cdeadmin, 2017-10-04:


anoo1, 2017-10-04:
The service to update the symlinks for mboxd started before the remount of the partitions, so
no symlinks were created a mboxd failed to start. Need to add a dependency to those services.
```
Oct 03 20:28:01 witherspoon systemd[1]: Starting Updates symlinks for active PNOR version...
Oct 03 20:28:04 witherspoon systemd[1]: Starting Remount the squashfs and ubi volumes after a reboot...
Oct 03 20:28:07 witherspoon obmc-flash-bios[1003]: ls: /media/pnor-ro-*: No such file or directory
Oct 03 20:28:05 witherspoon systemd[1]: Started Updates symlinks for active PNOR version.
Oct 03 20:28:15 witherspoon systemd[1]: Started Remount the ubi volume after a reboot.
..
Oct 03 20:28:55 witherspoon systemd[1]: Started Phosphor MBOX Daemon.
Oct 03 20:28:57 witherspoon mboxd[1243]: The operation failed internally.
```

anoo1, 2017-10-04:
With the service change:
```
Oct 04 19:17:22 witherspoon systemd[1]: Starting Remount the squashfs and ubi volumes after a reboot...
Oct 04 19:17:48 witherspoon systemd[1]: Started Remount the squashfs and ubi volumes after a reboot.
Oct 04 19:18:12 witherspoon systemd[1]: Starting OpenPower Software Update Manager...
Oct 04 19:18:19 witherspoon systemd[1]: Started OpenPower Software Update Manager.
Oct 04 19:18:19 witherspoon systemd[1]: Starting Create a ubi volume for pnor patches...
Oct 04 19:18:19 witherspoon systemd[1]: Starting Updates symlinks for active PNOR version...
Oct 04 19:18:20 witherspoon systemd[1]: Started Updates symlinks for active PNOR version.
Oct 04 19:18:24 witherspoon systemd[1]: Started Create a ubi volume for pnor patches.
Oct 04 19:18:24 witherspoon systemd[1]: Started Phosphor MBOX Daemon.
```

rfrandse, 2017-10-04:
<https://gerrit.openbmc-project.xyz/7074> openpower-software-manager: Add service dependencies
Resolves  #2417 MFG:Witherspoon, "Common.Error.InternalFailureError" SEL logged after flashing 1738D

closed 2017-10-06 by geissonator