openbmc/openbmc #2419: [Witherspoon] System is not powering on after shut down on OS

swe12345, 2017-10-05:
Witherspoon system is not powering off after shutting down from the OS.
Code Update was successful. 
Reboot on the OS was successful. 
Shut down on OS failed and the system is not coming up.

```
root@witherspoon:~# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.On
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Running
```
Shutting down from the OS:
```
root@rcx2d508:~# shutdown -h now

Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  .Ubuntu 17.04.  .  .  
```
System Configuration:
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.10-105"
VERSION_ID="v1.99.10-105-g53edba2"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.10-105"
BUILD_ID="v1.99.10"
```
PNOR details:
```
 "data": {
    "/xyz/openbmc_project/software/27b41bda": {
      "Activation": "xyz.openbmc_project.Software.Activation.Activations.Active",
      "ExtendedVersion": "op-build-v1.7-1383-g487ffd1,buildroot-2017.08-6-g319c6e1,skiboot-v5.8-90-g398a98123641-pb7fe7a5,hostboot-3d6c541,linux-4.13-openpower1-p1b975d8,petitboot-v1.5.1-p65d605b,machine-xml-f8aaa73-p03357dc,occ-a43395b,hostboot-binaries-edd568f,capp-ucode-9c73e9f,sbe-8d90ab2",
      "Path": "",
      "Priority": 0,
      "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.Host",
      "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.None",
      "Version": "IBM-witherspoon-sequoia-ibm-OP9_v1.19_1.41",
      "associations": [
        [
          "inventory",
          "activation",
          "/xyz/openbmc_project/inventory/system/chassis"
        ]
      ]
    },
```
[journalctl_5oct.txt](https://github.com/openbmc/openbmc/files/1358222/journalctl_5oct.txt)

sivassrr, 2017-10-05:
@swe12345 pls change the title to power off.

sivassrr, 2017-10-05:
Capture myffdc and upload it here

geissonator, 2017-10-05:
This issue needs to go over to OPAL, looks like they're failing to shutdown.  If you wait 45 minutes, the timer will hit in openbmc and it will forcefully shut it down.

closed 2017-10-05 by geissonator

sivassrr, 2017-10-06:
not able to recreate with today's driver 
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.10-113"
VERSION_ID="v1.99.10-113-g65edf7d"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.10-113"
BUILD_ID="v1.99.10"
root@witherspoon:~# cat /etc/timestamp
20171006022726
root@witherspoon:~# cat /media/pnor-ro-776f6a74/VERSION
IBM-witherspoon-ibm-OP9_v1.19_1.52
	op-build-v1.7-1421-g2f4cbfe
	buildroot-2017.08-6-g319c6e1
	skiboot-v5.8-130-g712837cedca0
	hostboot-8e7e2ad
	linux-4.13.4-openpower1-p1c668f0
	petitboot-v1.6.0-p85a1eab
	machine-xml-1bd247d-p7c6c471
	occ-60c09cd
	hostboot-binaries-b7d8417
	capp-ucode-p9-dd2-v1
	sbe-ff713df
root@witherspoon:~#

root@witherspoon:~#
root@witherspoon:~# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.On
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Running
root@witherspoon:~# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.Off
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Off

```
[  OK  ] Deactivated swap /dev/dm-1.
[  OK  ] Deactivated swap /dev/mapper/01301--vg-swap_1.
[  OK  ] Stopped Update UTMP about System Boot/Shutdown.
[  OK  ] Stopped Network Time Synchronization.
[  OK  ] Stopped Create Volatile Files and Directories.
[  OK  ] Stopped target Local File Systems.
         Unmounting /run/user/0...
         Unmounting /boot...
[  OK  ] Unmounted /run/user/0.
[  OK  ] Unmounted /boot.
[  OK  ] Reached target Unmount All Filesystems.
[  OK  ] Stopped File System Check on /dev/d…60ea1-7740-45e8-bfc7-400f1ea0b8b4.
[  OK  ] Removed slice system-systemd\x2dfsck.slice.
[  OK  ] Stopped target Local File Systems (Pre).
[  OK  ] Stopped Create Static Device Nodes in /dev.
         Stopping Monitoring of LVM2 mirrors…ng dmeventd or progress polling...
[  OK  ] Stopped Remount Root and Kernel File Systems.
[  OK  ] Reached target Shutdown.
[  246.172246] reboot: Power down
[  360.099157208,5] OPAL: Shutdown request type 0x0...```

amboar, 2017-10-08:
Hi @sivassrr, please use the triple back-tick markdown when pasting code or console output into github issues. It prevents github's rendering engine interpreting your log output as markdown. What you write looks like:

    ```
    My code, console or log output here
    ```