openbmc/openbmc #2420: IPL: Flood of core_phosphor-fan-controls on w51 (DD21) with 910.1740.20171002b

cdeadmin, 2017-10-05:


msbarth, 2017-10-10:
Fan control init consistently is unable to access fan3_0 FanSpeed target to set its speed causing the application to core by design. This is on the init service running fan control which only sets fans to full speed on poweron.

"MESSAGE" : "Error in mapper response for getting service name",
"PATH" : "/xyz/openbmc_project/sensors/fan_tach/fan3_0",
"INTERFACE" : "xyz.openbmc_project.Control.FanSpeed",
"SYSLOG_IDENTIFIER" : "phosphor-fan-control",

I dug thru all the FFDC provided and could not find anything in regard to why the `Target` property for fan3_0 wasnt created along with why fan3_1 sensor was never created. These are the root causes to the fan control init cores.

        "/xyz/openbmc_project/sensors/fan_tach/fan0_0": {
            "Scale": 0,
            "Target": 10500,
            "Unit": "xyz.openbmc_project.Sensor.Value.Unit.RPMS",
            "Value": 0
        },
        "/xyz/openbmc_project/sensors/fan_tach/fan0_1": {
            "Scale": 0,
            "Unit": "xyz.openbmc_project.Sensor.Value.Unit.RPMS",
            "Value": 0
        },
        "/xyz/openbmc_project/sensors/fan_tach/fan1_0": {
            "Scale": 0,
            "Target": 10500,
            "Unit": "xyz.openbmc_project.Sensor.Value.Unit.RPMS",
            "Value": 0
        },
        "/xyz/openbmc_project/sensors/fan_tach/fan1_1": {
            "Scale": 0,
            "Unit": "xyz.openbmc_project.Sensor.Value.Unit.RPMS",
            "Value": 0
        },
        "/xyz/openbmc_project/sensors/fan_tach/fan2_0": {
            "Scale": 0,
            "Target": 10500,
            "Unit": "xyz.openbmc_project.Sensor.Value.Unit.RPMS",
            "Value": 0
        },
        "/xyz/openbmc_project/sensors/fan_tach/fan2_1": {
            "Scale": 0,
            "Unit": "xyz.openbmc_project.Sensor.Value.Unit.RPMS",
            "Value": 0
        },
        "/xyz/openbmc_project/sensors/fan_tach/fan3_0": {
            "Scale": 0,
            "Unit": "xyz.openbmc_project.Sensor.Value.Unit.RPMS",
            "Value": 0
        },

A recreate of this issue to stop on this error so sysfs can be investigated to narrow in on whether this is on the user(hwmon) or device driver/kernel(max31785) side is needed. At a minimum, some additional FFDC on fails here should be added to the logs.

amboar, 2017-10-11:
I suspect the problem outlined in series [pmbus: Fix error propagation for register check functions]
(http://patchwork.ozlabs.org/project/openbmc/list/?series=1500&state=%2A&archive=both) might be involved. It can lead to mysterious absence of functionality.

msbarth, 2017-10-27:
This is a dup of #2518 where @amboar has a workaround fix that's being tested/merged to perform a retry when setting the page to the max31785 fails.

msbarth, 2017-11-14:
This issue can be closed, it has not reoccurred since @amboar page retry on the max31785 has been merged.

closed 2017-11-14 by spinler