openbmc/openbmc #2423: OpenBMC:Witherspoon:FVT- Getting few medium and low priority vulnerabilities

cdeadmin, 2017-10-05:


ratagupt, 2017-11-16:
1) These are the security vulnerabilities on the openbmc which we need to work on to fix.
     I was taking help from team to run the security tools on the openbmc stack and add the report so that we can work on it..

ratagupt, 2018-01-14:
Few more is there..

==== State: Open by: prkatti1 on 15 November 2017 04:38:11 ==== 12

we re-ran security scan on latest build v2 r7, still we found same issues.
Also found new medium sev vulnerability

Web Server Generic XSS

Synopsis :

The remote web server is affected by a cross-site scripting
vulnerability.

Description :

The remote host is running a web server that fails to adequately
sanitize request strings of malicious JavaScript. A remote attacker
can exploit this issue, via a specially crafted request, to execute
arbitrary HTML and script code in a user's browser within the security
context of the affected site.

See also :

https://en.wikipedia.org/wiki/Cross-site_scripting

Solution :

Contact the vendor for a patch or upgrade.

Risk factor :

Medium / CVSS Base Score : 4.3
(CVSS2#AV:N/AC:M/Au:N/C:N/I:P/A:N)
CVSS Temporal Score : 3.6
(CVSS2#E:F/RL:OF/RC:C)
Public Exploit Available : true

ratagupt, 2018-01-15:
Cross Site Scripting: https://gerrit.openbmc-project.xyz/#/c/8483/
Disable Weak ciphers from Webserver : https://gerrit.openbmc-project.xyz/#/c/8484/1
Expired Certificate: https://gerrit.openbmc-project.xyz/#/c/8485/1

There are still some vulnerabilities in dropbear ssh server which needs to be handled.

rfrandse, 2018-01-17:
<https://gerrit.openbmc-project.xyz/8485> Security: Configure new ssl certificate
Partially Resolves  #2423 Getting few medium and low priority vulnerabilities

closed 2018-01-23 by geissonator