openbmc/openbmc #2427: Enable use of watchdog within systemd in userspace

geissonator, 2017-10-05:
#314 enabled the watchdogs.  #1974 tracks being able to boot from the alternate flash chip.  This story will bring it all together.
- Systemd needs to open watchdog so it takes over pinging it
- Need to patch in https://github.com/shenki/systemd/tree/watchdog-config to yocto and utilize this so systemd will use the 2nd watchdog (first one is used for fan control)
- Need to verify the end to end flow by ensuring watchdog expiration results in expected behavior (reset, side switch)

This should also resolve #1909 but since this is new function, wanted an enhancement.

geissonator, 2017-10-09:
@saqibkh has some availability and will be helping out

eddiejames, 2017-10-10:
# **STATUS**

10/10 - Working on an update for systemd to use a specific watchdog device, rather than /dev/watchdog only. Joel's patch got me started, but doesn't do the whole thing. My initial test of my patch was unsuccessful, so I need to add some tracing to systemd and keep working

mdmillerii, 2017-10-11:
I'll reiterate I think we should just choose which watchdog we want to be used for reset and make it `/dev/watchdog0` (which `/dev/watchdog` always aliases) via the device tree `/aliases` node.

rfrandse, 2017-10-12:
<https://gerrit.openbmc-project.xyz/7267> systemd: Patch in watchdog device number and enable it
Resolves  #2427 Enable use of watchdog within systemd in userspace

amboar, 2017-10-12:
> I'll reiterate I think we should just choose which watchdog we want to be used for reset and make it /dev/watchdog0 (which /dev/watchdog always aliases) via the device tree /aliases node.

But why? systemd is using a legacy interface to the watchdog - I think it's entirely reasonable that we fix systemd and *don't* mangle the device node associations.

mdmillerii, 2017-10-12:
For that matter we should have `/aliases` to make sure with a random ordered device tree the kernel uses known instances if it matters, so we need to do that for all systems.  We also need to teach the kernel which watchdog to use for reboot if we don't use `watchdog0` no?  Otherwise we will just kick the fans up.

It just seems adding unnecessary complication code when the system designer can specify which watchdog to use via the device tree today.  And the patches show it is a non-trivial amount of code being added.

amboar, 2017-10-12:
> For that matter we should have /aliases to make sure with a random ordered device tree the kernel uses known instances if it matters, so we need to do that for all systems. 

Agreed.

> We also need to teach the kernel which watchdog to use for reboot if we don't use watchdog0 no? Otherwise we will just kick the fans up.

IIRC the kernel just tries to trigger all available watchdogs. In this case as you suggest it'll hit the fan watchdog kicking the fans up, but also hit the system reset watchdog. Having the fans high across a BMC reset probably isn't a bad idea in the event that the BMC doesn't come back.

> It just seems adding unnecessary complication code when the system designer can specify which watchdog to use via the device tree today. And the patches show it is a non-trivial amount of code being added.

I think that's for upstream systemd to decide. If they say "no, go away" then we at least have this fallback. Maybe they say exactly what you suggest? We need to start the conversation with them.

eddiejames, 2017-11-14:
Trying to get this patchset finished up today. The shutdown watchdog wasn't working correctly. Once that's working, I will create a pull request with systemd.

amboar, 2017-11-19:
FWIW the systemd pull-req is here: https://github.com/systemd/systemd/pull/7352

rfrandse, 2018-01-30:
patch reference information:  https://gerrit.openbmc-project.xyz/#/c/8098/

bradbishop, 2018-04-13:
This was reverted, so reopening.

amboar, 2018-04-24:
Revert was tracked in #3094 - I've reproduced that behaviour in qemu, investigating now.

geissonator, 2018-07-18:
Where'd we land with this one @eddiejames @amboar ?

closed 2018-08-06 by geissonator