openbmc/openbmc #2430: Debug tarball excludes libraries installing into /lib, breaking some applications

amboar, 2017-10-06:
Recently I was looking at adding latencytop to the debugtools tarball. Simply adding the package to `packagegroup-obmc-phosphor-debugtools.bb` doesn't help: latencytop won't execute as the dynamic linker can't find libncurses.so.5. As it turns out, ncurses installs its libraries into `/lib`, not `/usr/lib`, and so

```
IMAGE_CMD_tar = "${IMAGE_CMD_TAR} -cvf ${IMGDEPLOYDIR}/${IMAGE_NAME}${IMAGE_NAME_SUFFIX}.tar -C ${IMAGE_ROOTFS}/usr ."
```
in `obmc-phosphor-debug-tarball.bbclass` excludes the required library.

We need a way where we can trivially add applications for debugging without hacks like copying libraries manually. This may be a broader problem than just hacking the way we build the debug tools tarball in some way.

amboar, 2017-10-06:
I've pushed some PoC changes here: https://gerrit.openbmc-project.xyz/#/q/project:openbmc/openbmc+branch:master+topic:latencytop

mdmillerii, 2017-10-07:
In addition, the tool to regen the ld.so.conf was removed as unnecessary recently in addition as it was claimed a static path was sufficient.

bradbishop, 2018-06-06:
latencytop was dropped upstream:  http://cgit.openembedded.org/openembedded-core/commit/?id=36aae56e7f86a4d5ce93e4528e7dcc42f60c705e

amboar, 2018-06-07:
I don't think that means we shouldn't solve this problem.

amboar, 2019-03-28:
Still needs investigation/resolution as far as I'm aware.

closed 2020-04-30 by stale[bot]