openbmc/openbmc #2433: network interface getting reset multiple times on way to ready state

geissonator, 2017-10-07:
This hit in CI, causing a false failure.   The test automation waits on the REST interface, and once working it attempts an ssh into the system.  When I review the network and dropbear traces, I see the phosphor-network-manager logging the same stuff to the journal multiple times and the eth0 network apparently losing and gaining the network multiple times.  There were no changes to the network config.  What's causing this continuous gain and loss of the network?

```
[witherspoon]# tracf | grep -e network -e dropbear
Oct 07 16:22:47 witherspoon kernel: Driver for 1-wire Dallas network protocol.
Oct 07 16:22:47 witherspoon kernel: netconsole: network logging started
Oct 07 16:22:55 witherspoon systemd[1]: Listening on dropbear.socket.
Oct 07 16:23:03 witherspoon systemd-networkd[988]: Enumeration completed
Oct 07 16:23:03 witherspoon systemd-networkd[988]: eth0: IPv6 enabled for interface: Success
Oct 07 16:23:03 witherspoon systemd-networkd[988]: eth0: Gained carrier
Oct 07 16:23:05 witherspoon systemd-networkd[988]: eth0: Gained IPv6LL
Oct 07 16:23:02 witherspoon kernel[1015]: [    2.160000] Driver for 1-wire Dallas network protocol.
Oct 07 16:23:02 witherspoon kernel[1015]: [    7.730000] netconsole: network logging started
Oct 07 16:23:18 witherspoon systemd-networkd[988]: eth0: Configured
Oct 07 16:23:18 witherspoon systemd[1]: Created slice system-dropbear.slice.
Oct 07 16:23:19 witherspoon dropbear[1098]: Failed loading /var/lib/dropbear/dropbear_rsa_host_key
Oct 07 16:23:19 witherspoon dropbear[1098]: Failed loading /etc/dropbear/dropbear_dss_host_key
Oct 07 16:23:19 witherspoon dropbear[1098]: Failed loading /etc/dropbear/dropbear_ecdsa_host_key
Oct 07 16:23:19 witherspoon dropbear[1098]: Child connection from ::ffff:9.41.165.94:37578
Oct 07 16:23:27 witherspoon dropbear[1098]: PAM password auth succeeded for 'root' from ::ffff:9.41.165.94:37578
Oct 07 16:23:56 witherspoon systemd[1]: Starting Wait for /xyz/openbmc_project/network/config...
Oct 07 16:24:02 witherspoon phosphor-network-manager[1171]: ConfigParser: Key not found
Oct 07 16:24:02 witherspoon phosphor-network-manager[1171]: The operation failed internally.
Oct 07 16:24:02 witherspoon phosphor-network-manager[1171]: Exception occured during getting of DHCP value
Oct 07 16:24:48 witherspoon systemd[1]: Started Wait for /xyz/openbmc_project/network/config.
Oct 07 16:25:00 witherspoon systemd-networkd[988]: eth0: Lost carrier
Oct 07 16:25:00 witherspoon systemd-networkd[988]: eth0: Gained carrier
Oct 07 16:25:01 witherspoon phosphor-network-manager[1171]: ConfigParser: Key not found
Oct 07 16:25:01 witherspoon phosphor-network-manager[1171]: The operation failed internally.
Oct 07 16:25:01 witherspoon phosphor-network-manager[1171]: Exception occured during getting of DHCP value
Oct 07 16:25:01 witherspoon systemd-networkd[988]: eth0: Gained IPv6LL
Oct 07 16:25:01 witherspoon systemd-networkd[988]: eth0: Configured
Oct 07 16:25:01 witherspoon phosphor-network-manager[1171]: Timer expired
Oct 07 16:25:28 witherspoon systemd-networkd[988]: eth0: Lost carrier
Oct 07 16:25:29 witherspoon systemd-networkd[988]: eth0: Gained carrier
Oct 07 16:25:30 witherspoon phosphor-network-manager[1171]: ConfigParser: Key not found
Oct 07 16:25:30 witherspoon phosphor-network-manager[1171]: The operation failed internally.
Oct 07 16:25:30 witherspoon phosphor-network-manager[1171]: Exception occured during getting of DHCP value
Oct 07 16:25:30 witherspoon phosphor-network-manager[1171]: Timer expired
Oct 07 16:25:30 witherspoon systemd-networkd[988]: eth0: Gained IPv6LL
Oct 07 16:25:30 witherspoon systemd-networkd[988]: eth0: Configured
Oct 07 16:25:34 witherspoon dropbear[1475]: Failed loading /var/lib/dropbear/dropbear_rsa_host_key
Oct 07 16:25:34 witherspoon dropbear[1475]: Failed loading /etc/dropbear/dropbear_dss_host_key
Oct 07 16:25:34 witherspoon dropbear[1475]: Failed loading /etc/dropbear/dropbear_ecdsa_host_key
Oct 07 16:25:34 witherspoon dropbear[1475]: Child connection from ::ffff:9.41.165.94:37602
Oct 07 16:25:35 witherspoon dropbear[1475]: PAM password auth succeeded for 'root' from ::ffff:9.41.165.94:37602
Oct 07 16:25:52 witherspoon dropbear[1098]: Exit (root): Exited normally
Oct 07 16:25:52 witherspoon dropbear[1475]: Exit (root): Exited normally
Oct 07 16:26:21 witherspoon dropbear[1501]: Failed loading /var/lib/dropbear/dropbear_rsa_host_key
Oct 07 16:26:21 witherspoon dropbear[1501]: Failed loading /etc/dropbear/dropbear_dss_host_key
Oct 07 16:26:21 witherspoon dropbear[1501]: Failed loading /etc/dropbear/dropbear_ecdsa_host_key
Oct 07 16:26:21 witherspoon dropbear[1501]: Child connection from ::ffff:9.3.62.215:44259
Oct 07 16:26:21 witherspoon dropbear[1501]: PAM password auth succeeded for 'root' from ::ffff:9.3.62.215:44259
Oct 07 16:26:32 witherspoon dropbear[1509]: Failed loading /var/lib/dropbear/dropbear_rsa_host_key
Oct 07 16:26:32 witherspoon dropbear[1509]: Failed loading /etc/dropbear/dropbear_dss_host_key
Oct 07 16:26:32 witherspoon dropbear[1509]: Failed loading /etc/dropbear/dropbear_ecdsa_host_key
Oct 07 16:26:32 witherspoon dropbear[1509]: Child connection from ::ffff:9.41.165.103:41156
Oct 07 16:26:32 witherspoon dropbear[1509]: PAM password auth succeeded for 'root' from ::ffff:9.41.165.103:41156
Oct 07 16:26:35 witherspoon dropbear[1513]: Failed loading /var/lib/dropbear/dropbear_rsa_host_key
Oct 07 16:26:35 witherspoon dropbear[1513]: Failed loading /etc/dropbear/dropbear_dss_host_key
Oct 07 16:26:35 witherspoon dropbear[1513]: Failed loading /etc/dropbear/dropbear_ecdsa_host_key
Oct 07 16:26:35 witherspoon dropbear[1513]: Child connection from ::ffff:9.41.165.103:41166
Oct 07 16:26:35 witherspoon dropbear[1513]: PAM password auth succeeded for 'root' from ::ffff:9.41.165.103:41166
Oct 07 16:26:38 witherspoon dropbear[1509]: Exit (root): Exited normally
Oct 07 16:26:38 witherspoon dropbear[1513]: Exit (root): Exited normally
```

This was seen on w54 but I think we pretty much see this on all systems with the latest master code.

ratagupt, 2017-10-07:
I can see that systemd-networkd has not been restarted here..and the process is 988 for systemd-networkd..If it would have been restarted from phosphor-networkd,the process id would have been changed for systemd-networkd.
Seems to me genuine network up/down.
I would see the behaviour on other system also.

geissonator, 2017-10-09:
Can we get to root cause on why this up/down happens so much Ratan?  I wonder if there's something we can do to avoid it (like brining systemd-networkd up later in the boot)?  I'd prefer it come up later, but stay up if possible.

ratagupt, 2017-10-24:
Seems to me NCSI interface is going up/down multiple times,that can be the cause.

sammj, 2017-11-15:
While working on NCSI logging I saw on machines where this is happening that the NCSI driver was being instructed to stop via ncsi_stop_dev(), which is only called in turn via ftgmac100_stop(). So it doesn't sound like the NCSI link is going down, it sounds like something else is setting the interface down and up - likely the systemd/openbmc network stack.

closed 2019-10-27 by stale[bot]