openbmc/openbmc #2434: Error while running bitbake for Raspberry PI 3 build.

xavinux, 2017-10-09:
Trying to build OpenBMC image for Raspberry PI 3 on Ubuntu 16.04 / 64 bits machine.

I've followed these steps:

**_tester@obmc:~$ sudo apt-get install -y git build-essential libsdl1.2-dev texinfo gawk chrpath diffstat

tester@obmc:~$ git clone git@github.com:openbmc/openbmc.git

tester@obmc:~$ cd openbmc

tester@obmc:/openbmc$ export TEMPLATECONF=meta-openbmc-machines/meta-evb/meta-evb-raspberrypi/conf

 tester@obmc:/openbmc$ . oe-init-build-env ~/my-build-raspberrypi

tester@obmc:~/my-build-raspberrypi$ bitbake obmc-phosphor-image

Build Configuration:
BB_VERSION        = "1.32.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "Ubuntu-16.04"
TARGET_SYS        = "arm-openbmc-linux-gnueabi"
MACHINE           = "raspberrypi"
DISTRO            = "openbmc-phosphor"
DISTRO_VERSION    = "0.1.0"
TUNE_FEATURES     = "arm armv6  vfp arm1176jzfs callconvention-hard"
TARGET_FPU        = "hard"
meta              
meta-poky         
meta-oe           
meta-networking   
meta-python       
meta-virtualization 
meta-phosphor     
meta-raspberrypi  = "master:65edf7d796c34012982632d0a724d80ff44c9e4a"_**

The following error appears:

**_ERROR: linux-raspberrypi-1_4.4.48+gitAUTOINC+7ddf96fbb7-r0 do_kernel_metadata: Function failed: do_kernel_metadata (log file is located at /home/tester/my-build-raspberrypi/tmp/work/raspberrypi-openbmc-linux-gnueabi/linux-raspberrypi/1_4.4.48+gitAUTOINC+7ddf96fbb7-r0/temp/log.do_kernel_metadata.7276)
ERROR: Logfile of failure stored in: /home/tester/my-build-raspberrypi/tmp/work/raspberrypi-openbmc-linux-gnueabi/linux-raspberrypi/1_4.4.48+gitAUTOINC+7ddf96fbb7-r0/temp/log.do_kernel_metadata.7276
Log data follows:
| DEBUG: Executing shell function do_kernel_metadata
| ERROR. input file "phosphor-gpio-keys" does not exist
| ERROR: could not process input files: /home/tester/openbmc/import-layers/meta-raspberrypi/recipes-kernel/linux/linux-raspberrypi/defconfig /home/tester/openbmc/import-layers/meta-raspberrypi/recipes-kernel/linux/linux-raspberrypi-4.4/0001-fix-dtbo-rules.patch phosphor-gpio-keys
|        See /tmp/tmp.GA2zgVbszZ for details
| ERROR. input file "phosphor-gpio-keys" does not exist
| ERROR: could not process input files: /home/tester/openbmc/import-layers/meta-raspberrypi/recipes-kernel/linux/linux-raspberrypi/defconfig /home/tester/openbmc/import-layers/meta-raspberrypi/recipes-kernel/linux/linux-raspberrypi-4.4/0001-fix-dtbo-rules.patch phosphor-gpio-keys
|        See /tmp/tmp.BEzIPRvVMy for details
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_kernel_metadata (log file is located at /home/tester/my-build-raspberrypi/tmp/work/raspberrypi-openbmc-linux-gnueabi/linux-raspberrypi/1_4.4.48+gitAUTOINC+7ddf96fbb7-r0/temp/log.do_kernel_metadata.7276)
ERROR: Task (/home/tester/openbmc/import-layers/meta-raspberrypi/recipes-kernel/linux/linux-raspberrypi_4.4.bb:do_kernel_metadata) failed with exit code '1'
NOTE: Tasks Summary: Attempted 632 tasks of which 0 didn't need to be rerun and 1 failed.
Summary: 1 task failed:
  /home/tester/openbmc/import-layers/meta-raspberrypi/recipes-kernel/linux/linux-raspberrypi_4.4.bb:do_kernel_metadata
Summary: There were 6 WARNING messages shown.
Summary: There was 1 ERROR message shown, returning a non-zero exit code._**

Regards,

Javier

amboar, 2017-10-09:
Hi @xavinux, thanks for the bug report! As a note for the future, please use the triple back-tick markdown when pasting code or console output into github issues. It prevents github's rendering engine interpreting your log output as markdown. What you write looks like:

    ```
    My code, console or log output here
    ```

A guide to using github-flavoured markdown is here:

https://guides.github.com/features/mastering-markdown/

xavinux, 2017-10-09:
Hi @amboar , thanks for your advice. I've corrected the output text.

an146, 2017-11-10:
Want to give me a cheap VDS if I fix this? Preferably Selectel, Russia

amboar, 2017-11-10:
@an146 you get the simple pleasure of having fixed it and contributing to an open source project, but nothing more

an146, 2017-11-10:
it doesn't even has a BMC, what's the point

amboar, 2017-11-10:
@an146 the raspberry pi can be used *as* a BMC, with some hardware hacking, to control say power application to another machine. Now maybe that's a bit of an elaborate solution, but it can serve as a useful introduction to OpenBMC

amboar, 2018-01-02:
No one has fixed this to my knowledge. I was hoping someone with an interest in making the port work could use this issue to contribute their first patch

alehed, 2018-09-27:
A workaround is to append the line
```
KERNEL_FEATURES_remove = "phosphor-gpio-keys"
```
to the `local.conf` (or `local.conf.sample`). This is what some other targets do to disable gpio-keys.

jyangchisyan, 2018-10-08:
Hi @alehed ,
 I try to follow the workaround, but when **do_generate_static** exectue error.
 
 Below are log information:

```
DEBUG: Executing python function extend_recipe_sysroot
NOTE: Direct dependencies are ['/home/samueljiang/openbmc/meta-raspberrypi/recipes-kernel/linux/linux-raspberrypi_4.14.bb:do_populate_sysroot', '/home/samueljiang/openbmc/meta/recipes-devtools/mklibs/mklibs-native_0.1.43.bb:do_populate_sysroot', 'virtual:native:/home/samueljiang/openbmc/meta/recipes-extended/bc/bc_1.06.bb:do_populate_sysroot', '/home/samueljiang/openbmc/meta/recipes-devtools/gcc/gcc-cross_7.3.bb:do_populate_sysroot', 'virtual:native:/home/samueljiang/openbmc/meta/recipes-bsp/u-boot/u-boot-mkimage_2018.01.bb:do_populate_sysroot', 'virtual:native:/home/samueljiang/openbmc/meta/recipes-devtools/rpm/rpm_4.14.1.bb:do_populate_sysroot', '/home/samueljiang/openbmc/meta/recipes-core/os-release/os-release.bb:do_populate_sysroot', 'virtual:native:/home/samueljiang/openbmc/meta/recipes-devtools/bison/bison_3.0.4.bb:do_populate_sysroot', '/home/samueljiang/openbmc/meta/recipes-devtools/qemu/qemuwrapper-cross_1.0.bb:do_populate_sysroot', 'virtual:native:/home/samueljiang/openbmc/meta/recipes-devtools/mtd/mtd-utils_git.bb:do_populate_sysroot', '/home/samueljiang/openbmc/meta/recipes-core/glibc/ldconfig-native_2.12.1.bb:do_populate_sysroot', '/home/samueljiang/openbmc/meta/recipes-bsp/u-boot/u-boot_2018.01.bb:do_populate_sysroot', 'virtual:native:/home/samueljiang/openbmc/meta/recipes-devtools/dnf/dnf_2.7.5.bb:do_populate_sysroot', '/home/samueljiang/openbmc/meta/recipes-kernel/kmod/kmod-native_git.bb:do_populate_sysroot', 'virtual:native:/home/samueljiang/openbmc/meta/recipes-devtools/prelink/prelink_git.bb:do_populate_sysroot', '/home/samueljiang/openbmc/meta/recipes-kernel/kmod/depmodwrapper-cross_1.0.bb:do_populate_sysroot', 'virtual:native:/home/samueljiang/openbmc/meta/recipes-devtools/opkg/opkg_0.3.6.bb:do_populate_sysroot', '/home/samueljiang/openbmc/meta/recipes-devtools/binutils/binutils-cross_2.30.bb:do_populate_sysroot', 'virtual:native:/home/samueljiang/openbmc/meta/recipes-devtools/makedevs/makedevs_1.0.1.bb:do_populate_sysroot', 'virtual:native:/home/samueljiang/openbmc/meta/recipes-devtools/pseudo/pseudo_git.bb:do_populate_sysroot', 'virtual:native:/home/samueljiang/openbmc/meta/recipes-kernel/dtc/dtc_1.4.5.bb:do_populate_sysroot', 'virtual:native:/home/samueljiang/openbmc/meta/recipes-devtools/createrepo-c/createrepo-c_git.bb:do_populate_sysroot', 'virtual:native:/home/samueljiang/openbmc/meta/recipes-extended/xz/xz_5.2.3.bb:do_populate_sysroot', 'virtual:native:/home/samueljiang/openbmc/meta/recipes-devtools/opkg-utils/opkg-utils_0.3.6.bb:do_populate_sysroot', 'virtual:native:/home/samueljiang/openbmc/meta/recipes-devtools/squashfs-tools/squashfs-tools_git.bb:do_populate_sysroot', '/home/samueljiang/openbmc/meta/recipes-devtools/quilt/quilt-native_0.65.bb:do_populate_sysroot', '/home/samueljiang/openbmc/meta/recipes-kernel/kern-tools/kern-tools-native_git.bb:do_populate_sysroot', 'virtual:native:/home/samueljiang/openbmc/meta/recipes-extended/pigz/pigz_2.4.bb:do_populate_sysroot', 'virtual:native:/home/samueljiang/openbmc/meta/recipes-core/update-rc.d/update-rc.d_0.7.bb:do_populate_sysroot', 'virtual:native:/home/samueljiang/openbmc/meta/recipes-support/lzop/lzop_1.03.bb:do_populate_sysroot']
NOTE: Installed into sysroot: []
NOTE: Skipping as already exists in sysroot: ['linux-raspberrypi', 'mklibs-native', 'bc-native', 'gcc-cross-arm', 'u-boot-mkimage-native', 'rpm-native', 'os-release', 'bison-native', 'qemuwrapper-cross', 'mtd-utils-native', 'ldconfig-native', 'u-boot', 'dnf-native', 'kmod-native', 'prelink-native', 'depmodwrapper-cross', 'opkg-native', 'binutils-cross-arm', 'makedevs-native', 'pseudo-native', 'dtc-native', 'createrepo-c-native', 'xz-native', 'opkg-utils-native', 'squashfs-tools-native', 'quilt-native', 'kern-tools-native', 'pigz-native', 'update-rc.d-native', 'lzop-native', 'systemd-systemctl-native', 'shadow-native', 'gettext-minimal-native', 'libtool-native', 'autoconf-native', 'automake-native', 'gnu-config-native', 'flex-native', 'texinfo-dummy-native', 'zlib-native', 'gmp-native', 'mpfr-native', 'libmpc-native', 'glibc', 'linux-libc-headers', 'openssl-native', 'popt-native', 'pkgconfig-native', 'dbus-native', 'db-native', 'libarchive-native', 'bzip2-native', 'python3-native', 'elfutils-native', 'file-native', 'nss-native', 'qemu-native', 'lzo-native', 'e2fsprogs-native', 'acl-native', 'util-linux-native', 'rpi-u-boot-scr', 'gcc-runtime', 'librepo-native', 'libdnf-native', 'python3-iniparse-native', 'cmake-native', 'ninja-native', 'libcomps-native', 'gtk-doc-native', 'binutils-native', 'libsolv-native', 'expat-native', 'libxml2-native', 'glib-2.0-native', 'curl-native', 'sqlite3-native', 'attr-native', 'lz4-native', 'm4-native', 'autoconf-archive-native', 'readline-native', 'gdbm-native', 'nspr-native', 'alsa-lib-native', 'pixman-native', 'ncurses-native', 'libgcc', 'gpgme-native', 'libcheck-native', 'gobject-introspection-native', 'python3-six-native', 're2c-native', 'libffi-native', 'libpcre-native', 'libpng-native', 'util-macros-native', 'swig-native', 'libassuan-native', 'libgpg-error-native', 'python3-setuptools-native', 'unzip-native']
DEBUG: Python function extend_recipe_sysroot finished
DEBUG: Executing python function do_generate_static
DEBUG: Executing shell function do_mk_static_nor_image
32768+0 records in
32768+0 records out
33554432 bytes (34 MB, 32 MiB) copied, 0.0317138 s, 1.1 GB/s
DEBUG: Shell function do_mk_static_nor_image finished
392+1 records in
392+1 records out
401640 bytes (402 kB, 392 KiB) copied, 0.000563654 s, 713 MB/s
ERROR: Image '/home/samueljiang/openbmc/pi-build/tmp/deploy/images/raspberrypi/fitImage-obmc-phosphor-initramfs-raspberrypi.bin' is too large!
DEBUG: Python function do_generate_static finished
ERROR: Function failed: do_generate_static
```
how could I fix that?

Regards,

Samuel Jiang.

alehed, 2018-10-08:
For it to compile you still have to give it the right flash offsets (as seen in evb-zx3-pm3.conf).

For example the evb-zx3-pm3 has those offsets:
```
FLASH_UBOOT_OFFSET = "0"
FLASH_KERNEL_OFFSET = "6144"
FLASH_UBI_OFFSET = "${FLASH_KERNEL_OFFSET}"
FLASH_ROFS_OFFSET = "12288"
FLASH_RWFS_OFFSET = "65024"
FLASH_SIZE = "65536"
```
On the RBP the actual numbers might be different.

mdmillerii, 2018-10-09:
I would not expect rpi to boot a mtd flash image.  

I was going to say they didn't have a flash controller, but then I remembered you can attach a SPI chip.

Their boot process starts with the GPU reading files from a FAT16 partition on a SD card (or other source - they can do usb or usb device mode too).  The kernel (or u-boot) and device tree are loaded from the fat partition.

My suggestion:
- The first step is to build and boot an upstream rpi image.
- Optionally add a step to boot a u-boot rpi image.
- Then try to replace the root file system image in that phosphor root file system.

amboar, 2018-11-08:
Someone recently put the effort in to make OpenBMC run on the Raspberry Pis:

https://lists.ozlabs.org/pipermail/openbmc/2018-October/013856.html

Still needs upstreaming, but it's a great starting point.

closed 2019-06-09 by stale[bot]

morchaos, 2019-08-15:
> A workaround is to append the line
> 
> ```
> KERNEL_FEATURES_remove = "phosphor-gpio-keys"
> ```
> 
> to the `local.conf` (or `local.conf.sample`). This is what some other targets do to disable gpio-keys.

the workaround works for me. it continue building still... took me hours
I am wondering why not create the file "phosphor-gpio-keys.cfg" for raspberrypi?
the contain I checked for Aspeed and Nuvoton are the same, there are only two samples I found in the source code below.
>cat ../meta-phosphor/aspeed-layer/recipes-kernel/linux/linux-aspeed/phosphor-gpio-keys.cfg
CONFIG_INPUT_KEYBOARD=y
# CONFIG_INPUT_LEDS is not set
# CONFIG_INPUT_MOUSDEV is not set
CONFIG_INPUT_EVDEV=y
# CONFIG_KEYBOARD_ATKBD is not set
CONFIG_KEYBOARD_GPIO=y
# CONFIG_INPUT_MOUSE is not set

amboar, 2019-08-16:
@morchaos Good question! Are you able to submit a patch? Please take a look at the contributing guidelines here: https://github.com/openbmc/docs/blob/master/CONTRIBUTING.md#submitting-changes-via-gerrit-server

jyangchisyan, 2019-08-16:
> @morchaos Good question! Are you able to submit a patch? Please take a look at the contributing guidelines here: https://github.com/openbmc/docs/blob/master/CONTRIBUTING.md#submitting-changes-via-gerrit-server

Hi @amboar, I could help submit this patch and I want confirmed I could create raspbetrrypi-layer in meta-phosphor folder.

By the way, the raspberrypi layer have another obmc-phosphor-image default image size problem.  
I still traced that in free time.

amboar, 2019-08-19:
Hi @jyangchisyan - thanks! I've added @mine260309 to review the patch as IIRC he did a bunch of the work to support the raspi. Hopefully that can move it along.