openbmc/openbmc #2437: witherspoon-power-supply-monitor@.service capitalization issue with SysLogIdentifier

geissonator, 2017-10-09:
The SysLogIdentifier= should be SyslogIdentifier  (lowercase l)

I noticed this in the logs:
```
Oct 09 15:29:01 witherspoon systemd[1]: [/lib/systemd/system/obmc-standby.target.requires/../witherspoon-power-supply-monitor@.service:12] Unknown lvalue 'SysLogIdentifier' in section 'Service'
Oct 09 15:29:01 witherspoon systemd[1]: [/lib/systemd/system/obmc-standby.target.requires/../witherspoon-power-supply-monitor@.service:12] Unknown lvalue 'SysLogIdentifier' in section 'Service'
root@witherspoon:~# 

```

bjwyman, 2017-10-09:
Simple enough fix to check and correct. Code pushed up.

rfrandse, 2017-10-10:
<https://gerrit.openbmc-project.xyz/7182> Fix capitalization issue with SyslogIdentifier
Resolves:  #2437 witherspoon-power-supply-monitor@.service capitalization issue with SysLogIdentifier

bjwyman, 2017-10-10:
# **STATUS**
**10/10**
- Today: See all +1 review done.
- Tomorrow: Check for review feedback.
- Blockers: +2 review and merge.

closed 2017-10-11 by geissonator