openbmc/openbmc #2439: IPL: Error log cleanup: Pervaisve Power.Fault.Error.PowerSupplyInputFault / PowerSupplyShouldBeOn on w51?, w55, wsbmc010, wsbmc016?, WS-Seq-FW-1 with 910.1740.20171006c

cdeadmin, 2017-10-09:


bjwyman, 2017-10-09:
The PowerSupplyInputFault looked to be due to the test starting just prior to the top power supply being plugged into a good power source.

The PowerSupplyShouldBeOn looks to be due to some unaccounted delay from the UCD Sequencer to the power supply ENABLE# line, which also has a delay to 12V in spec, then a delay from that to DC_GOOD going active which also changes the POWER_GOOD Negated in the STATUS_WORD.

rfrandse, 2017-10-10:
<https://gerrit.openbmc-project.xyz/7201> Update the delay from state=1 to powerOn=true
Resolves:  #2439 IPL: Error log cleanup: Pervaisve Power.Fault.Error.PowerSupplyInputFault / PowerSupplyShouldBeOn on w51?, w55, wsbmc010, wsbmc016?, WS-Seq-FW-1 with 910.1740.20171006c

closed 2017-10-11 by geissonator