openbmc/openbmc #2444: Inventory manager is not catering for empty interfaces

devenrao, 2017-10-10:
Catered for empty interfaces as part of 1786, looks like it is broken. Do not see empty interfaces being created by phosphor-inventory-manager.

This should not have any functional impact as of now.

rfrandse, 2017-10-28:
<https://gerrit.openbmc-project.xyz/7662> Rework CPU watch rule to watch on type-only CPU interface
Resolves  #2444 Inventory manager is not catering for empty interfaces

devenrao, 2017-12-14:
# **Status**
12/14 - Pushed code changes for review. Solution need to be relooked with recent changes done to occ by deepak. There may be chances that this story might be invalidated.

devenrao, 2018-01-03:
At present we are not watching on cpu objects created to create occ objects. So this changes are not longer valid. Need to abandon the story.

closed 2018-01-03 by dkodihal