openbmc/openbmc #2448: DeleteAll REST interface does not delete anything

charleshofer, 2017-10-10:
1. Upload and activate a BMC image
2. Call `curl -b cjar -k -H 'Content-Type: application/json' -X POST -d '{"data":[]}' https://<bmc ip>/xyz/openbmc_project/software/action/DeleteAll`
3. Enumerate software. Notice that the image just uploaded is still there.

DeleteAll works fine if you use the interface on port 3000, just not working through REST.

```
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.10-116"
VERSION_ID="v1.99.10-116-g7870c80"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.10-116"
BUILD_ID="v1.99.10"
```

[journal.txt](https://github.com/openbmc/openbmc/files/1372976/journal.txt)

mtritz, 2017-10-10:
Following the steps above, I was not able to recreate the issue on my own. Granted, I was using this firmware -
```
ID="openbmc-phosphor"`
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.10-119"
VERSION_ID="v1.99.10-119-g9dc05e5"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.10-119"
BUILD_ID="v1.99.10"
```
Subsequently, I tried the same machine with the same firmware that Charlie used above and confirmed that the issue does occur. Interestingly, it seems that the function is run - the cleanup service file starts - but no activations are removed. Since the function does work on the later firmware, the status of this issue isn't completely clear to me.

charleshofer, 2017-10-11:
The issue still exists in
```
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.10-121"
VERSION_ID="v1.99.10-121-gb5da93f"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.10-121"
BUILD_ID="v1.99.10"
```

mtritz, 2017-10-11:
This is a problem with the REST server. What's happening in Charlie's case is that both the host and BMC Object.Deletes are implemented under the same path. The curl command, specifying only the function and the path, should DeleteAll for both host and BMC, but stops after one.

saqibkh, 2017-11-08:
Fixed by https://gerrit.openbmc-project.xyz/#/c/7788/

closed 2017-11-14 by rfrandse