openbmc/openbmc #2453: IPL: The specified resource cannot be created: '/xyz/openbmc_project/watchdog/host0 during AutoReboot test on WS-Seq-FW-1 with 910.1740.20171006c

cdeadmin, 2017-10-11:


geissonator, 2017-10-12:
Summary of issue:

Automated test cases are encountering this error from the REST api during testing

geissonator, 2017-10-12:
Dug into the traces a bit more, the fail on /xyz/openbmc_project/watchdog/host0 is because the host watchdog is not running.

# Power down requested and watchdog disabled
Oct 06 20:09:54 witherspoon systemd[1]: Stopped Phosphor poweron watchdog.

# Waiting for host to shutdown
Oct 06 20:10:11 witherspoon phosphor-softpoweroff[1890]: Timer started waiting for host to shutdown

# Attempt by test case to read watchdog
Oct 06 20:27:08 witherspoon phosphor-gevent[1168]: ::6039:c276:8e91:f402%1243261716 - - [2017-10-06 20:27:08] "PUT /xyz/openbmc_project/watchdog/host0/attr/Enabled HTTP/1.1" 403 289 0.054789

I see this in the OS_msglog.txt, which show the shutdown was processed, but then the host just kept on booting.  Probably related to https://github.com/open-power/skiboot/issues/81#issuecomment-335355802

[   44.027813631,5] IPMI: Soft shutdown requested

My guess is the test case sees the state as still chassis on and host running so tries to read the host watchdog.  Once the power off path is requested though, the host watchdog is stopped because it will no longer be pinging the watchdog, and we want to give it the full 45 minutes to shutdown properly.  I think the host hang issue is captured elsewhere so I'm going to assign this over to the automated test team to either not read the host watchdog after a power off request or to just ignore the error if not available.

geissonator, 2017-10-12:
Closing this one out as test team will be handling the changes to the test case.

closed 2017-10-12 by geissonator