openbmc/openbmc #2454: IPL: OS is up following IPMI MC Reset Warm (off) with a Hard Power Off prior on w55 with 910.1738.20170908u

cdeadmin, 2017-10-11:


geissonator, 2017-10-12:
Summary of issue:

The basic issue here is the last host requested state was a "IPMI Power On".  This is saved within the host state object.

The tests then do a chassis hard power off.  This is like holding down the power button on your computer and not letting it shut down.  It's not a host requested state, it's a shutdown-immediately type request.

The BMC is then rebooted.  The policy for this system is programmed to RESTORE last state.  That means, restore the last host requested state, which is this case was a power on.  That's why we power on.

I can definitely see this being argued both ways.  Our struggle is with supporting these concepts some day in blade center and multi-node systems, we don't want to box ourselves in with lots of complex logic and dependencies between objects.  At a minimum, the documentation should be updated to be clear on the design here.

geissonator, 2017-10-12:
One point I thought of as I went through the doc for https://github.com/openbmc/phosphor-dbus-interfaces/blob/master/xyz/openbmc_project/Control/Power/RestorePolicy.interface.yaml is that we should really only be enforcing the policy when the BMC is A/C cycled, not just rebooted.  Looking into whether we have a way to determine this in BMC code.

mdmillerii, 2017-10-12:
I don't think we want to go down "why did we reboot" path.  After power is applied something may cause the bmc to reboot before the system was booted an we want to get the host back to running.

It sounds more like "should chassis off be considered a host state?" question.  Its probably complicated a bit by having host booting imply chassis power on.

amboar, 2017-10-12:
> Its probably complicated a bit by having host booting imply chassis power on.

Right, this seems like a convenience. If we split them out (obviously the physical dependency still remains), having the conflict state (chassis off but host booted) simply means that the host doesn't get booted.

How splitting that out works in terms of user interactions is another question though.

geissonator, 2017-10-17:
Added some clarification to restore policy in https://gerrit.openbmc-project.xyz/#/c/7417/

rfrandse, 2017-10-18:
<https://gerrit.openbmc-project.xyz/7417> Clarify restore policy behavior
Partially Resolves  #2454 IPL: OS is up following IPMI MC Reset Warm (off) with a Hard Power Off prior on w55 with 910.1738.20170908u

closed 2017-11-06 by geissonator