openbmc/openbmc #2455: missing traces in console output

cdeadmin, 2017-10-11:


geissonator, 2017-10-12:
Details:
```
IBM-witherspoon-ibm-OP9_v1.19_1.52
	op-build-v1.7-1422-g7bccb07
	buildroot-2017.08-6-g319c6e1
	skiboot-v5.8-130-g712837cedca0
	hostboot-8e7e2ad
	linux-4.13.4-openpower1-p1c668f0
	petitboot-v1.6.0-p85a1eab
	machine-xml-1bd247d-p7c6c471
	occ-60c09cd
	hostboot-binaries-b7d8417
	capp-ucode-p9-dd2-v1
	sbe-ff713df

We can give you examples of where traces are missing, but I don't think they would be useful without context. This was observed in the PRD and MDIA traces during step 14.1. Sometimes we are missing single line traces. Other times we are missing blocks of 10-20 traces. Makes it really hard to debug.

We've also done some experimentation where we will put two traces back-to-back in the code, with nothing in between them, so both traces should always be printed. We've seen all combinations. Sometimes only the first shows, sometimes only the second shows, sometimes neither shows.
```

jk-ozlabs, 2017-10-12:
Just as a helpful datapoint: are the logs missing entire lines (ie, dropping on newline boundaries), or do you see the starts/ends of dropped lines in the logs too?

mine260309, 2017-10-12:
We met similar issue during some P8 machine's bring up. At that time it was the buffer full due to too many logs. We tried to increase the buffer size and then the issue is gone.

If the root cause is the same, you can increase the buffer and have a try. @geissonator 
```
diff --git a/console-server.c b/console-server.c
index fb84e19..d0ba0da 100644
--- a/console-server.c
+++ b/console-server.c
@@ -68,7 +68,7 @@ struct poller {
 static const int n_internal_pollfds = 1;
 
 /* size of the shared backlog ringbuffer */
-const size_t buffer_size = 128 * 1024;
+const size_t buffer_size = 128 * 1024 * 8;
 
 /* state shared with the signal handler */
 static bool sigint;
@@ -463,6 +463,7 @@ static void sighandler(int signal)
 		sigint = true;
 }
 
+static uint8_t buf[4096 * 16];
 int run_console(struct console *console)
 {
 	sighandler_t sighandler_save;
@@ -473,7 +474,6 @@ int run_console(struct console *console)
 	rc = 0;
 
 	for (;;) {
-		uint8_t buf[4096];
 
 		BUILD_ASSERT(sizeof(buf) <= buffer_size);
 

```

jk-ozlabs, 2017-10-12:
@mine260309 : we drop to blocking reads/writes in the console server once buffers are full, so changing the buffer size won't affect dropped characters.

jk-ozlabs, 2017-10-12:
(that change to drop to blocking writes has probably happened since the original bringup work though)

zane131, 2017-10-12:
@jk-ozlabs We are seeing entire lines missing. No segmented lines that we know of.

zane131, 2017-10-24:
Now we are seeing segmented lines:
```
 96.91154|PRDF|E>HOMER: EC Level:21
.C: Loading 24b random seed map index 1 
 97.24862|FAPI|mcbist.C: Loading 24b random seed map index 2 
```

jk-ozlabs, 2018-04-19:
So this should be fixed by https://github.com/openbmc/linux/commit/158c8e15d577cebdb047f5d553e12054b15b1b84 ; might be worth a re-test with a recent build.

closed 2018-07-18 by rfrandse