openbmc/openbmc #2458: Latest phosphor-host-ipmid fails to compile in openbmc

geissonator, 2017-10-11:
This is a bump to the latest code in host-ipmid - https://gerrit.openbmc-project.xyz/#/c/7263/.  It failed to compile with this error.
```
18:09:05 Log data follows:
18:09:05 | DEBUG: SITE files ['endian-little', 'bit-32', 'arm-common', 'arm-32', 'common-linux', 'common-glibc', 'arm-linux', 'arm-linux-gnueabi', 'common']
18:09:05 | DEBUG: Executing shell function do_compile
18:09:05 | NOTE: make -j16
18:09:05 | ERROR: oe_runmake failed
18:09:05 | /bin/bash ../git/generate_whitelist.sh /tmp/openbmc/sysroots/x86_64-linux/usr/share/phosphor-ipmi-host/phosphor-ipmi-fru-whitelist-native.conf /tmp/openbmc/sysroots/x86_64-linux/usr/share/phosphor-ipmi-host/openpower-ipmi-oem-whitelist-native.conf /tmp/openbmc/work/armv6-openbmc-linux-gnueabi/phosphor-ipmi-host/1.0-r1/git/host-ipmid-whitelist.conf > ipmiwhitelist.cpp
18:09:05 | /tmp/openbmc/sysroots/x86_64-linux/usr/bin/python-native/python ../git/scripts/sensor_gen.py -i /tmp/openbmc/sysroots/x86_64-linux/usr/share/phosphor-ipmi-host/sensor/sensor.yaml -o . generate-cpp
18:09:05 | /tmp/openbmc/sysroots/x86_64-linux/usr/bin/python-native/python ../git/scripts/inventory-sensor.py -i /tmp/openbmc/sysroots/x86_64-linux/usr/share/phosphor-ipmi-host/sensor/invsensor.yaml -o . generate-cpp
18:09:05 | /tmp/openbmc/sysroots/x86_64-linux/usr/bin/python-native/python ../git/scripts/fru_gen.py -i /tmp/openbmc/sysroots/x86_64-linux/usr/share/phosphor-ipmi-host/config/fru_config.yaml -o . generate-cpp
18:09:05 | Traceback (most recent call last):
18:09:05 |   File "../git/scripts/sensor_gen.py", line 60, in <module>
18:09:05 |     main()
18:09:05 |   File "../git/scripts/sensor_gen.py", line 55, in main
18:09:05 |     function(args.sensor_yaml, args.outputdir)
18:09:05 |   File "../git/scripts/sensor_gen.py", line 24, in generate_cpp
18:09:05 |     fd.write(t.render(sensorDict=ifile))
18:09:05 |   File "/tmp/openbmc/sysroots/x86_64-linux/usr/lib/python2.7/site-packages/mako/template.py", line 445, in render
18:09:05 |     return runtime._render(self, self.callable_, args, data)
18:09:05 |   File "/tmp/openbmc/sysroots/x86_64-linux/usr/lib/python2.7/site-packages/mako/runtime.py", line 829, in _render
18:09:05 |     **_kwargs_for_callable(callable_, data))
18:09:05 |   File "/tmp/openbmc/sysroots/x86_64-linux/usr/lib/python2.7/site-packages/mako/runtime.py", line 864, in _render_context
18:09:05 |     _exec_template(inherit, lclcontext, args=args, kwargs=kwargs)
18:09:05 |   File "/tmp/openbmc/sysroots/x86_64-linux/usr/lib/python2.7/site-packages/mako/runtime.py", line 890, in _exec_template
18:09:05 |     callable_(context, *args, **kwargs)
18:09:06 |   File "_tmp_openbmc_work_armv6_openbmc_linux_gnueabi_phosphor_ipmi_host_1_0_r1_git_scripts_writesensor_mako_cpp", line 152, in render_body
18:09:06 | KeyError: 'Offsets'
18:09:06 | Makefile:1499: recipe for target 'sensor-gen.cpp' failed
18:09:06 | make: *** [sensor-gen.cpp] Error 1
18:09:06 | WARNING: exit code 1 from a shell command.
18:09:06 | ERROR: Function failed: do_compile (log file is located at /tmp/openbmc/work/armv6-openbmc-linux-gnueabi/phosphor-ipmi-host/1.0-r1/temp/log.do_compile.31813)
18:09:06 No currently running tasks (4298 of 4357)  98% ||
```

geissonator, 2017-10-12:
https://gerrit.openbmc-project.xyz/#/c/7270/ (mrw-rev) fails to compile as well with the following:
```
23:38:00 ERROR: Logfile of failure stored in: /tmp/openbmc/work/x86_64-linux/phosphor-ipmi-sensor-inventory-mrw-native/1.0-r1/temp/log.do_install.20136
23:38:00 Log data follows:
23:38:00 | DEBUG: Executing shell function do_install
23:38:00 | Can't use string ("uint32_t") as a HASH ref while "strict refs" in use at /tmp/openbmc/sysroots/x86_64-linux/usr/bin/gen_ipmi_sensor.pl line 148.
23:38:00 | Loading MRW XML: /tmp/openbmc/sysroots/x86_64-linux/usr/share/obmc-mrw/witherspoon.xml
23:38:00 | WARNING: exit code 255 from a shell command.
23:38:00 | ERROR: Function failed: do_install (log file is located at /tmp/openbmc/work/x86_64-linux/phosphor-ipmi-sensor-inventory-mrw-native/1.0-r1/temp/log.do_install.20136)
23:38:00 Currently 10 running tasks (3493 of 4357)  80% ||
```

dhruvibm, 2017-10-12:
it passed now, that was due to dependency on mrw and host-ipmid changes
https://gerrit.openbmc-project.xyz/#/c/7029/

closed 2017-10-13 by geissonator