openbmc/openbmc #2461: Update to Yocto 2.3

geissonator, 2017-10-12:
Getting to be near that time again.

FYI @saqibkh @anoo1

amboar, 2017-10-12:
Any thoughts on what we should do to avoid "that time" again? It would be good to do CI builds against passing upstream master commits so we can work on this over time, rather than ingest massive updates. That might lead to some restructuring of what we have now, e.g. moving away from the git-subtree imports we currently have.

dcobbley, 2017-10-13:
Just curious, YP released 2.4 back in August. Are you worried about making small incremental changes? 

Agree with moving away from sub-tree.

YP community recommends always having a CI pointed at master to ensure you stay aware of the direction they are going and that your project aligns well for the future with theirs.

saqibkh, 2017-10-13:
@dcobbley Yocto's current latest official release is Pyro 2.3.2.
We don't want to point to the latest master because master tends to be less stable.

dcobbley, 2017-10-13:
Sorry, getting ahead of myself - 2.4 targeted release is in 7 days. I had misread the timeline. https://wiki.yoctoproject.org/wiki/Yocto_Project_v2.4_Status#Milestone_4.2FFinal_-_Target_Oct._20.2C_2017

I'm not suggesting having openbmc supported on YP master, but simply an extra pipeline in Jenkins that builds the distribution with poky master. Letting it break, but giving early feedback of the direction YP is going.

dcobbley, 2017-11-15:
@saqibkh When do you suspect the project will be ready for YP 2.3? 
I see you've been busy :) https://gerrit.openbmc-project.xyz/#/q/topic:yocto-2.3

saqibkh, 2017-11-16:
@dcobbley It's difficult to give a timeline here.
The major remaining task now is to build images for all available platforms (barreleye, witherspoon etc) and fix all build errors along the way. Don't know how many errors there are and how complicated each one of them would be to resolve.

amboar, 2017-11-19:
> Don't know how many errors there are and how complicated each one of them would be to resolve.

Obviously this is a big motivator for doing CI against some form of upstream.

geissonator, 2018-01-03:
Brad did the update to 2.2.2, has started update to 2.3 so be sure to work with him there.

rfrandse, 2018-02-21:
<https://gerrit.openbmc-project.xyz/8792> yocto 2.3
Resolves:  #2461 Update to Yocto 2.3

closed 2018-03-13 by geissonator