openbmc/openbmc #2463: Internal server error when activating host firmware

cdeadmin, 2017-10-12:


anoo1, 2017-10-12:
There's a time window where if the user issues a REST enumerate request during a software activation, and before the call returns an interface is removed (like ProgressIndicator) as part of the activation, then the enumerate fails saying that interface does not exist.

Need a mapper update to handle short lived processes.

```
curl -s  -b cjar  -c cjar   -k  -H  'Content-Type: application/json'   -X GET https://9.5.10.64///
xyz/openbmc_project/software/6a0c1508
{
  "data": {
    "description": "Internal Server Error",
    "exception": "DBusException(\"Unknown interface 'xyz.openbmc_project.Software.ActivationProgre
ss'.\",)",
    "traceback": [
      "Traceback (most recent call last):",
      "  File \"/usr/lib/python2.7/site-packages/bottle.py\", line 862, in _handle",
      "    return route.call(**args)",
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 800, in __cal
l__",
      "    return self.callback(*a, **kw)",
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 857, in wrap"
,
      "    return callback(*a, **kw)",
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 1040, in __ca
ll__",
      "    return self.callback(*a, **kw)",
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 1014, in wrap
",
      "    return self.to_jsonp(callback(*a, **kw))",
      "  File \"/usr/lib/python2.7/site-packages/bottle.py\", line 1734, in wrapper",
      "    rv = callback(*a, **ka)",
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 979, in wrap"
,
      "    data = callback(*a, **kw)",
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 200, in __cal
l__",
      "    return getattr(self, 'do_' + request.method.lower())(**kw)",
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 538, in do_ge
t",
      "    mapper_data=request.route_data['map'])",
      "  File \"/usr/lib/python2.7/site-packages/obmc/mapper/bindings.py\", line 127, in enumerate
_object",
      "    path, owner, interfaces, match))",
      "  File \"/usr/lib/python2.7/site-packages/obmc/mapper/bindings.py\", line 111, in __get_pro
perties_on_bus",
      "    properties_iface, i))",
      "  File \"/usr/lib/python2.7/site-packages/obmc/mapper/bindings.py\", line 97, in __get_prop
erties_on_iface",
      "    properties_iface.GetAll, iface)",
      "  File \"/usr/lib/python2.7/site-packages/obmc/mapper/bindings.py\", line 83, in __try_prop
erties_interface",
      "    return f(*a)",
      "  File \"/usr/lib/python2.7/site-packages/dbus/proxies.py\", line 145, in __call__",
      "    **keywords)",
      "  File \"/usr/lib/python2.7/site-packages/dbus/connection.py\", line 651, in call_blocking"
,
      "    message, timeout)",
      "DBusException: org.freedesktop.DBus.Error.UnknownInterface: Unknown interface 'xyz.openbmc_
project.Software.ActivationProgress'."
    ]
  },
  "message": "500 Internal Server Error",
  "status": "error"
}
```

whowutwut, 2017-11-08:
Not sure if this is an exact match but looks like the above shows Unknown interface `ActivationProgress` but we see it on `Software.Version`. 

```
# rflash cn01 -a witherspoon.pnor.squashfs.tar
cn01: Uploading /firmware/Witherspoon/OP910_1742D/witherspoon.pnor.squashfs.tar ...
cn01: Firmware upload successful. Attempting to activate firmware: IBM-witherspoon-ibm-OP9_v1.19_1.73
cn01: rflash started, please wait...
cn01: Activating firmware . . . 10%
cn01: Activating firmware . . . 10%
cn01: Error: DBusException("Unknown interface 'xyz.openbmc_project.Software.Version'.",)
```

The polling query is done against the `/xyz/openbmc_project/software/<ID>` endpoint

Reported by customer here: https://github.com/xcat2/xcat-core/issues/4261

rfrandse, 2017-12-11:
<https://gerrit.openbmc-project.xyz/8235> bindings.py: Ignore unknown interface error
Resolves  #2463 Internal server error when activating host firmware

closed 2017-12-18 by geissonator