openbmc/openbmc #2465: Sprint 17.10.18 - Committed

rfrandse, 2017-10-12:


spinler, 2017-10-13:
***Squad 2 Sprint Plan***

**Matt S**

**Matt B**

**Gunnar**

**Brandon**

**Eddie**

dkodihal, 2017-10-16:
***Squad 1 Sprint Plan***

**Vacation**
Ratan - 16 Oct - 22 Oct
18th - holiday in Hyderabad
20th - holiday in Bangalore

**Deepak**
#2425 System boots to temporary boot policy setting even after powering on the system multiple times
#2414 openpower-read-vpd: PIM Notify() failed
#2377 phosphor-inventory: Encountered unsupported interface

**Jayanth**
#2350 UBI- empty file after unclean reboot
#2365 phosphor-mboxd: Flash read where there is no partition segfaults daemon

**Ratan**
#2163 [xCAT] Unable to modify existing Network Attributes Object
#2450 [xCAT] Setting hostname on firmware 1738 does not persis in the BMC - Removing network files and reboot causes this

**Devender**
#1960 phosphor-logging: Duplicated Event string in error message
#2444 Inventory manager is not catering for empty interfaces

**Nagaraju**
#2402 Logging exited with 'core-dump' during boot to standby 
#2333 If mac address is all 0's, code will still use it

**Dhruv**
#2298 IPMID : Invalid event data on Poweron
#2347 Failed to start SYNC system BMC ethernet.

**Tom**
#2368 OP910:OPRAS:Witherspoon:Procedure callouts don't show up in BMC error logs
- Finish pending events rework and recipe changes.

anoo1, 2017-10-17:
***Squad 3 Sprint Plan***

**Vacation**
Michael - 27 Oct - 5 Nov

**Adriana**

**Saqib**

**Michael**

**Chris**

**Charlie**

closed 2017-11-09 by zahrens