openbmc/openbmc #2470: [Witherspoon] xyz.openbmc_project.Power.Fault.Error.PowerSupplyShouldBeOn

swe12345, 2017-10-13:
Observing the following error logs after powering on the system.
```
{
  "data": {
    "/xyz/openbmc_project/logging/entry/2": {
      "AdditionalData": [
        "_PID=1310"
      ],
      "Id": 2,
      "Message": "xyz.openbmc_project.Power.Fault.Error.PowerSupplyInputFault",
      "Resolved": 0,
      "Severity": "xyz.openbmc_project.Logging.Entry.Level.Error",
      "Timestamp": 1505854825090,
      "associations": []
    },
    "/xyz/openbmc_project/logging/entry/3": {
      "AdditionalData": [
        "_PID=1310"
      ],
      "Id": 3,
      "Message": "xyz.openbmc_project.Power.Fault.Error.PowerSupplyShouldBeOn",
      "Resolved": 0,
      "Severity": "xyz.openbmc_project.Logging.Entry.Level.Error",
      "Timestamp": 1505854828176,
      "associations": []
    }
  },
  "message": "200 OK",
  "status": "ok"
```
```
root@witherspoon:~# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.On
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Running
```
System Configuration:
BMC Details:
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.10-139"
VERSION_ID="v1.99.10-139-g3202561"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.10-139"
BUILD_ID="v1.99.10"
```
PNOR Details:
```
root@witherspoon:~# cat /media/pnor-ro-46e65782/VERSION
IBM-witherspoon-ibm-OP9_v1.19_1.60
        op-build-v1.19-147-gc4823a9
        buildroot-2017.08-8-g5e23247
        skiboot-v5.8-150-gabb1d4e81f33
        hostboot-b01e6bb
        linux-4.13.5-openpower1-p402ce7c
        petitboot-v1.6.0-pf254bee
        machine-xml-389c56f
        occ-ba4e81e
        hostboot-binaries-d9d2910
        capp-ucode-p9-dd2-v2
        sbe-374add4
```
[journalctl_13oct.txt](https://github.com/openbmc/openbmc/files/1381465/journalctl_13oct.txt)
FFDC logs: /gsa/ausgsa/projects/i/indiateam02/swe12345/defects/poweronsupply

spinler, 2017-10-13:
@bjwyman I imagine this system just needs the PS plugged in?  If so, I will close this.

(Though, do you think it's even worth reporting the 2nd error if you already are reporting the 1st?)

bjwyman, 2017-10-13:
One of the power supplies is likely unplugged, but I cannot tell which one from the data provided. The error entry metadata and call-out would point to which supply to check.

[v2cib530@gfwr703 ~]$ ls /gsa/ausgsa/projects/i/indiateam02/swe12345/defects/poweronsupply
ls: cannot open directory /gsa/ausgsa/projects/i/indiateam02/swe12345/defects/poweronsupply: Permission denied
[v2cib530@gfwr703 ~]$

I thought it possibly made sense to report both of those errors. That was part of the reasoning for re-ordering some of the checks in the powered on path.

spinler, 2017-10-13:
This error is expected with an unplugged PS.  closing.

closed 2017-10-13 by spinler

amboar, 2017-10-15:
Just to clarify, `PowerSupplyInputFault` tells us that the powersupply is expected to be present but missing?

bjwyman, 2017-10-16:
PowerSupplyInputFault means that the input power to the supply is not within spec. Basically, this will hit on some kind of voltage or current issue (brownout) or if the supply is unplugged. Most of the faults I have seen are due to an unplugged power supply.

If the power supply is not present, the monitor will not be checking it for faults.