openbmc/openbmc #2471: How to send hard reset command to BMC using REST interface?

xuweibj, 2017-10-13:
Hi, is there REST interface to hard reset BMC?

geissonator, 2017-10-13:
No, there is no single command to do a hard reset (i.e. don't give host opportunity to shut down).  You'll need to do 2 commands.  One to the chassis state object to power off, then one to the host state object to power back on.  The host state object has a reboot option but that will give the host an opportunity to shut down cleanly.

Doc:
https://github.com/openbmc/phosphor-dbus-interfaces/tree/master/xyz/openbmc_project/State

closed 2017-10-13 by geissonator

whowutwut, 2017-10-16:
@geissonator  Did I also recall correct  in our conversations that we need to give the host time to power off and the API won't queue up requests that are sent to it? 

So we can't blindly send hard off to the chassis, then on to the host and expect the BMC to be 100% reliable to power off and on the server?

geissonator, 2017-10-16:
Correct, the chassis and state objects in BMC are independent so there is no queue possible behind those commands.  You'd need to do a hard power off, poll for chassis and host to be off, then request the power on.

geissonator, 2017-10-16:
Is this support other server provide a command for?

whowutwut, 2017-10-16:
@geissonator  Are there objects that do allow for a "Queue up" ? 

No, we never had this situation before because a hard reset on the box via IPMI was very quick and that command blocked. (1-2 seconds).  If I was behind an x86 server or even the OpenPower P8 LC boxes (Firestone/Minsky) the command issues to power off the server for hard off killed power within 1-2 seconds. 

Here with REST, even a hard off to the chassis takes about 15 seconds or more.. with power on being similar before the node goes into "Running " . After a flash of FW it's even worse and we have seen up to 2+ minutes before any IPL log messages show up. 

Because of this delay and we don't want to block on this command being send via REST, we were trying to implement a transition state where if the user queries the state of the power, they get something like `powering-on` or `powering-off` so they don't keep sending on/off commands while it's doing it's work... .  But as we discussed, a hard off doesn't update any transition states so we are trying to come up with other ways to save state in xCAT or only show transition states if we know they did a softoff.