openbmc/openbmc #2473: IPL: Intermittent Power.Fault.Error.PowerSupplyShouldBeOn errors on w55 with 910.1742.20171011a

cdeadmin, 2017-10-13:


spinler, 2017-10-13:
An update is being made to require 3 consecutive positives before a fault is reported.

bjwyman, 2017-10-13:
Tested a patch where the ShouldBeOn fault would only get logged if the STATUS_WORD had POWER_GOOD Negated or UNIT_IS_OFF bits in bad state three times in a row. I was not seeing the faults prior to the change, so decreased the power on delay timer. I would then hit the fault every power on on the system I was using to test things. 

I then patched up the 3 consecutive counter stuff. After four power on attempts, seeing some information tracing I had added in, every one would briefly have one or the other bit on. The fourth power on also had the 20 to 26 second glitch in UNIT_IS_OFF set, which was cleared a second later.

bjwyman, 2017-10-13:
I have a local commit that I was debating on just pushing up for a quick review, but the Gerrit server is still down.

rfrandse, 2017-10-16:
<https://gerrit.openbmc-project.xyz/7371> Implement de-glitch for PGOOD and UNIT_IS_OFF
Resolves  #2473 IPL: Intermittent Power.Fault.Error.PowerSupplyShouldBeOn errors on w55 with 910.1742.20171011a

rfrandse, 2017-10-19:
#**STATUS**
CQ is closed. 
Since the commit was cherry-picked leaving code-review until commit is merged.

closed 2017-10-25 by geissonator