openbmc/openbmc #2476: IPMID Coudn't find d-bus path

gkeishin, 2017-10-16:
During power on, we see this error, probably a cleanup is not done for the sensors in the inventory for IPMID

```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.10-154"
VERSION_ID="v1.99.10-154-g63d4f19"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.10-154"
BUILD_ID="v1.99.10"
root@witherspoon:~#
```

```
Oct 12 12:26:52 witherspoon ipmid[1273]: Failed to get a fru type: Invalid argumentFailed to create a method call: Invalid argumentBus: :1.17 Path:  Interface: org.openbmc.InventoryItem
Oct 12 12:26:53 witherspoon ipmid[1273]: Coudn't find d-bus path
Oct 12 12:26:53 witherspoon ipmid[1273]: The operation failed internally.
Oct 12 12:26:53 witherspoon ipmid[1273]: Set sensor failed
```

- [x] Journald log 
[ORG_IPMID_FAILURE.txt](https://github.com/openbmc/openbmc/files/1386726/ORG_IPMID_FAILURE.txt)

gkeishin, 2017-10-16:
@tomjoseph83 ^^^

gtmills, 2018-10-12:
@gkeishin @tomjoseph83 Still a problem?

closed 2019-05-14 by stale[bot]