openbmc/openbmc #2477: phosphor-discover-system-state@0.service: Main process exited, code=dumped, status=6/ABRT

gkeishin, 2017-10-16:
BMC r/r when Host is booted.. 

Post r/r :

`OS is up and running but BMC is at standby`
```
root@rcx2d508:~# uptime
 08:43:02 up 15 min,  1 user,  load average: 0.00, 0.03, 0.07
root@rcx2d508:~
```
`BMC state`
```
root@witherspoon:~# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.Off
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Off
root@witherspoon:~# obmcutil power
pgood = 1
state = 1
pgood_timeout = 10
root@witherspoon:~#
````
`System REST State enum`
```
/xyz/openbmc_project/state/enumerate
{
    "/xyz/openbmc_project/state/bmc0": {
        "CurrentBMCState": "xyz.openbmc_project.State.BMC.BMCState.Ready",
        "RequestedBMCTransition": "xyz.openbmc_project.State.BMC.Transition.None"
    },
    "/xyz/openbmc_project/state/chassis0": {
        "CurrentPowerState": "xyz.openbmc_project.State.Chassis.PowerState.Off",
        "RequestedPowerTransition": "xyz.openbmc_project.State.Chassis.Transition.Off"
    },
    "/xyz/openbmc_project/state/host0": {
        "AttemptsLeft": 3,
        "BootProgress": "xyz.openbmc_project.State.Boot.Progress.ProgressStages.OSStart",
        "CurrentHostState": "xyz.openbmc_project.State.Host.HostState.Off",
        "OperatingSystemState": "xyz.openbmc_project.State.OperatingSystem.Status.OSStatus.BootComplete",
        "RequestedHostTransition": "xyz.openbmc_project.State.Host.Transition.On"
    }
}
```

```
root@witherspoon:~# systemctl status phosphor-discover-system-state@0.service | cat
● phosphor-discover-system-state@0.service - Reboot If Enabled
   Loaded: loaded (/lib/systemd/system/obmc-standby.target.wants/../phosphor-discover-system-state@.service; disabled; vendor preset: enabled)
   Active: failed (Result: core-dump) since Thu 2017-10-12 12:02:25 UTC; 1min 57s ago
  Process: 1445 ExecStart=/usr/bin/env phosphor-discover-system-state --host %i (code=dumped, signal=ABRT)
 Main PID: 1445 (code=dumped, signal=ABRT)

Oct 12 12:02:06 witherspoon systemd[1]: Starting Reboot If Enabled...
Oct 12 12:02:19 witherspoon phosphor-discover-system-state[1445]: Host power is off, checking power policy
Oct 12 12:02:19 witherspoon phosphor-discover-system-state[1445]: power_policy=RESTORE, restoring last state
Oct 12 12:02:25 witherspoon phosphor-discover-system-state[1445]: Error in mapper call
Oct 12 12:02:25 witherspoon phosphor-discover-system-state[1445]: terminate called after throwing an instance of 'std::runtime_error'
Oct 12 12:02:25 witherspoon phosphor-discover-system-state[1445]:   what():  Error in mapper call
Oct 12 12:02:25 witherspoon systemd[1]: phosphor-discover-system-state@0.service: Main process exited, code=dumped, status=6/ABRT
Oct 12 12:02:25 witherspoon systemd[1]: Failed to start Reboot If Enabled.
Oct 12 12:02:25 witherspoon systemd[1]: phosphor-discover-system-state@0.service: Unit entered failed state.
Oct 12 12:02:25 witherspoon systemd[1]: phosphor-discover-system-state@0.service: Failed with result 'core-dump'.
```

`Driver Info`
```
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.10-154"
VERSION_ID="v1.99.10-154-g63d4f19"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.10-154"
BUILD_ID="v1.99.10"
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-97988a39b46d73304a5697b8d3a5cf08da287350 #1 Tue Oct 10 05:30:30 UTC 2017 armv6l GNU/Linux
root@witherspoon:~#
```

- [x] Journald log 
[RR_OS_booted_BUG.txt](https://github.com/openbmc/openbmc/files/1386815/RR_OS_booted_BUG.txt)

geissonator, 2017-10-16:
hmm, first real fail seems to be with mapper, which then is going to start to break everything else:
```
Oct 12 12:01:36 witherspoon bmc_update.py[1346]: Running Bmc Flash Control
Oct 12 12:01:36 witherspoon systemd[1]: Started Phosphor BMC Code Update.
Oct 12 12:01:38 witherspoon mapper[1290]: Error starting event loop: Connection timed out
Oct 12 12:01:38 witherspoon systemd[1]: mapper-wait@-xyz-openbmc_project-control-host0.service: Main process exited, code=exited, status=1/FAILURE
Oct 12 12:01:38 witherspoon systemd[1]: Failed to start Wait for /xyz/openbmc_project/control/host0.
Oct 12 12:01:38 witherspoon systemd[1]: mapper-wait@-xyz-openbmc_project-control-host0.service: Unit entered failed state.
Oct 12 12:01:38 witherspoon systemd[1]: mapper-wait@-xyz-openbmc_project-control-host0.service: Failed with result 'exit-code'.
Oct 12 12:01:45 witherspoon mapper[1301]: Error starting event loop: Connection timed out
Oct 12 12:01:45 witherspoon systemd[1]: mapper-wait@-org-openbmc-managers-System.service: Main process exited, code=exited, status=1/FAILURE
Oct 12 12:01:45 witherspoon systemd[1]: Failed to start Wait for /org/openbmc/managers/System.
Oct 12 12:01:45 witherspoon systemd[1]: mapper-wait@-org-openbmc-managers-System.service: Unit entered failed state.
Oct 12 12:01:45 witherspoon systemd[1]: mapper-wait@-org-openbmc-managers-System.service: Failed with result 'exit-code'.
```

gkeishin, 2017-10-16:
@geissonator anything recently went it that could have caused this one..

gkeishin, 2017-10-16:
We hit other failures in CT runs  which seems a mapper problem ..

```
root@witherspoon:~# systemctl status phosphor-max31785-msl@wspoon.service | cat
● phosphor-max31785-msl@wspoon.service - Phosphor MAX31785 MSL
   Loaded: loaded (/lib/systemd/system/obmc-standby.target.wants/../phosphor-max31785-msl@.service; static; vendor preset: enabled)
   Active: failed (Result: exit-code) since Thu 2017-10-12 12:01:44 UTC; 14min ago
  Process: 1311 ExecStart=/usr/bin/env max31785-msl -p ${PATHS} -b ${BUS} -r ${REVISION} (code=exited, status=1/FAILURE)
 Main PID: 1311 (code=exited, status=1/FAILURE)

Oct 12 12:01:18 witherspoon systemd[1]: Starting Phosphor MAX31785 MSL...
Oct 12 12:01:44 witherspoon max31785-msl[1311]: Error finding '/xyz/openbmc_project/inventory' service: Connection timed out
Oct 12 12:01:44 witherspoon systemd[1]: phosphor-max31785-msl@wspoon.service: Main process exited, code=exited, status=1/FAILURE
Oct 12 12:01:44 witherspoon systemd[1]: Failed to start Phosphor MAX31785 MSL.
Oct 12 12:01:44 witherspoon systemd[1]: phosphor-max31785-msl@wspoon.service: Unit entered failed state.
Oct 12 12:01:44 witherspoon systemd[1]: phosphor-max31785-msl@wspoon.service: Failed with result 'exit-code'.
root@witherspoon:~#
```

spinler, 2017-10-18:
Didn't recreate with:
```
VERSION="v1.99.10-158"
VERSION_ID="v1.99.10-158-g2a595c5"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.10-158"
BUILD_ID="v1.99.10"
```

gkeishin, 2017-10-18:
Anything from the FFDC we could find the reason. I ll try to recreate on the latter build once available.

spinler, 2017-12-18:
I think this went away, so closing.

closed 2017-12-18 by spinler