openbmc/openbmc #2478: [xCAT] Reset Network to Factory via REST API call does not clear out the network VLAN settings

cdeadmin, 2017-10-16:


anoo1, 2017-10-16:
From Victor Hu:

I finally got around to running the Factory Reset call on a node that was tagged VLAN

There's a file left around in the network folder for the VLAN: 00-bmc-eth0.11.network

Here's what it looked like when the BMC comes back up:

```
root@witherspoon:/etc/systemd/network# ls -ltr
total 8
-rw-r--r--    1 root     root            66 Oct 12 00:38 00-bmc-eth0.network
-rw-r--r--    1 root     root            69 Oct 12 00:38 00-bmc-eth0.11.network
root@witherspoon:/etc/systemd/network# cat 00-bmc-eth0.network
[Match]
Name=eth0
[Network]
DHCP=true
[DHCP]
ClientIdentifier=mac
root@witherspoon:/etc/systemd/network# cat 00-bmc-eth0.11.network
[Match]
Name=eth0.11
[Network]
DHCP=true
[DHCP]
ClientIdentifier=mac
```

From the RestAPI it looks OK,

```
[root@briggs01 p9]# ./list.sh
+ curl -c cjar -b cjar -k -H 'Content-Type: application/json' -X GET https://172.12.253.100/xyz/openbmc_project/network/enumerate
{
  "data": {
    "/xyz/openbmc_project/network/config": {
      "DefaultGateway": "172.12.253.27",
      "HostName": "witherspoon"
    },
    "/xyz/openbmc_project/network/config/dhcp": {
      "DNSEnabled": 1,
      "HostNameEnabled": 1,
      "NTPEnabled": 1
    },
    "/xyz/openbmc_project/network/eth0": {
      "AutoNeg": 0,
      "DHCPEnabled": 1,
      "DomainName": [],
      "InterfaceName": "eth0",
      "MACAddress": "70:e2:84:14:28:d",
      "Nameservers": [],
      "Speed": 0
    },
    "/xyz/openbmc_project/network/eth0/ipv4/8b2580a9": {
      "Address": "172.12.253.100",
      "Gateway": "0.0.0.0",
      "Origin": "xyz.openbmc_project.Network.IP.AddressOrigin.DHCP",
      "PrefixLength": 16,
      "Type": "xyz.openbmc_project.Network.IP.Protocol.IPv4"
    },
    "/xyz/openbmc_project/network/eth0/ipv6/28f8222a": {
      "Address": "fe80::72e2:84ff:fe14:280d",
      "Gateway": "",
      "Origin": "xyz.openbmc_project.Network.IP.AddressOrigin.DHCP",
      "PrefixLength": 64,
      "Type": "xyz.openbmc_project.Network.IP.Protocol.IPv6"
    },
    "/xyz/openbmc_project/network/host0/intf": {
      "MACAddress": "00:00:00:00:00:00"
    },
    "/xyz/openbmc_project/network/host0/intf/addr": {
      "Address": "0.0.0.0",
      "Gateway": "0.0.0.0",
      "Origin": "xyz.openbmc_project.Network.IP.AddressOrigin.Static",
      "PrefixLength": 0,
      "Type": "xyz.openbmc_project.Network.IP.Protocol.IPv4"
    }
  },
  "message": "200 OK",
```

But I don't think we should have that network file...

rfrandse, 2017-10-17:
<https://gerrit.openbmc-project.xyz/7387> Network factory reset: don't recreate VLAN interfaces
Resolves  #2478 [xCAT] Reset Network to Factory via REST API call does not clear out the network VLAN settings

closed 2017-10-25 by geissonator