openbmc/openbmc #2482: [LTCTest][OPAL][OP910] After erasing the GUARD partition we are not able to get the  full output of pflash --info

cdeadmin, 2017-10-17:


anoo1, 2017-11-14:
Reported issue:
```
[root@ltc-wspoon8 ~]# ./skiboot/external/pflash/pflash -r /mnt/gard.bin -P GUARD
Reading to "/mnt/gard.bin" from 0x00026000..0x0002b000 !
[==================================================] 100%

[root@ltc-wspoon8 ~]# ls -l /mnt/gard.bin
-rw-r--r--. 1 root root 20480 Oct  9 01:46 /mnt/gard.bin

Erase the GUARD partition.

[root@ltc-wspoon8 ~]# ./skiboot/external/pflash/pflash -P GUARD -e -f
About to erase 0x00026000..0x0002b000 !
Erasing...
[==================================================] 100%
Updating actual size in partition header...

Then check the PARTITION information using pflash tool.

[root@ltc-wspoon8 ~]# ./skiboot/external/pflash/pflash --info
Flash info:
-----------
Name          = /dev/mtd0
Total size    = 64MB     Flags E:ECC, P:PRESERVED, R:READONLY
Erase granule =  4KB           B:BACKUP, F:REPROVISION

The output gets stuck at the above point and we see the following messages in the console.

[ 3378.062541645,3] MBOX-FLASH: pos: 0x0, len: 0x30
[ 3378.062570733,3] MBOX-FLASH: win pos: 0x00000000 win size: 0x00000000
[ 3378.165022236,3] MBOX-FLASH: Move window is indicating size zero!
```

From Suraj:
```
So after pflash has finished erasing a partition it tries to update the actual size in the partition table header to zero.
It looks like in vpnor you can't write to the partition table, however this is failing silently.
As I understand it pflashing with vpnor is deprecated although I can't find any official comment to this effect.
```

mdmillerii, 2017-11-14:
I would guess Siraj is thinking of pflash from the BMC which is not allowed.
But as was stated elsewhere, any access the host can do to pnor should be supported.   vpnor is supposed to be transparent just more space efficent on the BMC.

closed 2018-03-01 by rfrandse

amboar, 2018-03-04:
@mdmillerii so I agree transparency would be ideal, however the implementation in phosphor-mboxd currently enforces the read-only property of partitions, so it's not transparent as is.

We can change this, though we should discuss it with @stewartsmith and the hostboot people.

anoo1, 2018-03-05:
This appears to have been fixed in the latest firmware levels:
```
[root@ws022os ~]# ./skiboot/external/pflash/pflash -r /mnt/gard.bin -P GUARD
Reading to "/mnt/gard.bin" from 0x0002c000..0x00031000 !
[==================================================] 100%

[root@ws022os ~]# ls -l /mnt/gard.bin
-rw-r--r--. 1 root root 20480 Jan  4 10:40 /mnt/gard.bin

[root@ws022os ~]# ./skiboot/external/pflash/pflash -P GUARD -e -f
About to erase 0x0002c000..0x00031000 !
Erasing...
[==================================================] 100%
Updating actual size in partition header...

[root@ws022os ~]# ./skiboot/external/pflash/pflash --info
Flash info:
-----------
Name          = /dev/mtd0
Total size    = 64MB	 Flags E:ECC, P:PRESERVED, R:READONLY
Erase granule =  4KB           B:BACKUP, F:REPROVISION, V:VOLATILE

TOC@0x00000000 Partitions:
-----------
ID=00            part 0x00000000..0x00001000 (actual=0x00001000) [------]
ID=01            HBEL 0x00008000..0x0002c000 (actual=0x00024000) [E-----]
ID=02           GUARD 0x0002c000..0x00031000 (actual=0x00000000) [EP----]
ID=03           NVRAM 0x00031000..0x000c1000 (actual=0x00090000) [-P----]
...
```