openbmc/openbmc #2483: VPD MAC data is not updated onto Inventory BMC ethernet property

gkeishin, 2017-10-17:
The VPD is populated but doesnt update the inventory property. We verified it works on GHE.

So there is a regression in the Master only for now.

```
/xyz/openbmc_project/inventory/system/chassis/motherboard/boxelder/bmc/ethernet
{
    "FieldReplaceable": 0,
    "MACAddress": "",
    "Present": 1,
    "PrettyName": ""
}
```

```
$ .UpdateBmcMac.sh wsbmcxxx -check
 MAC 70E2841XXXXX and data from VPD  70E2841XXXXX are same
 No need to update MAC
```

`Driver Info`
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.10-158"
VERSION_ID="v1.99.10-158-g2a595c5"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.10-158"
BUILD_ID="v1.99.10"
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-703c3c181790b20c48c4d02b9fed901379671fb7 #1 Mon Oct 16 07:26:12 UTC 2017 armv6l GNU/Linux
root@witherspoon:~#
```

dkodihal, 2017-10-17:
Also saw that the BMC object is not being created.

dkodihal, 2017-10-19:
This is seen on a specific Witherspoon that has bad VPD.

closed 2017-10-19 by dkodihal

gkeishin, 2018-03-13:
Looks like once we factory reset it, it seems to show up..