openbmc/openbmc #2486: IPL: Stuck with chassis power on and host power off on wsbcm016 with 910.1742.20171013a

cdeadmin, 2017-10-17:


rfrandse, 2017-10-17:
Eddie is working on a fix

spinler, 2017-10-18:
This defect is for a VRM bind failing on the power off:

```
Oct 14 03:16:43 witherspoon systemd[1]: Starting Power off mode bind ir35221 device driver devs...
Oct 14 03:16:43 witherspoon systemd[1]: Started Phosphor Hwmon Poller.
Oct 14 03:16:43 witherspoon systemd[1]: Started Phosphor Hwmon Poller.
Oct 14 03:16:43 witherspoon systemd[1]: Started Phosphor Hwmon Poller.
Oct 14 03:16:44 witherspoon ir35221-unbind-bind.sh[7982]: /usr/bin/ir35221-unbind-bind.sh: line 46: echo: write error: No such device
Oct 14 03:16:44 witherspoon systemd[1]: ir35221-off-bind@0.service: Main process exited, code=exited, status=1/FAILURE
Oct 14 03:16:44 witherspoon systemd[1]: Failed to start Power off mode bind ir35221 device driver devs.
Oct 14 03:16:44 witherspoon systemd[1]: Dependency failed for Chassis0 (Power Off).
Oct 14 03:16:44 witherspoon systemd[1]: Dependency failed for Power0 Host Off.
Oct 14 03:16:44 witherspoon systemd[1]: Dependency failed for Reboot Host0.
Oct 14 03:16:44 witherspoon kernel: ir35221 5-0071: Failed to read PMBUS_MFR_ID
Oct 14 03:16:44 witherspoon kernel: ir35221: probe of 5-0071 failed with error -71
Oct 14 03:16:44 witherspoon kernel[1004]: [ 8876.580000] ir35221 5-0071: Failed to read PMBUS_MFR_ID
Oct 14 03:16:44 witherspoon systemd[1]: obmc-host-reboot@0.target: Job obmc-host-reboot@0.target/start failed with result 'dependency'.
Oct 14 03:16:44 witherspoon kernel[1004]: [ 8876.580000] ir35221: probe of 5-0071 failed with error -71
Oct 14 03:16:44 witherspoon systemd[1]: obmc-host-reboot@0.target: Triggering OnFailure= dependencies.
Oct 14 03:16:44 witherspoon systemd[1]: obmc-host-shutdown@0.target: Job obmc-host-shutdown@0.target/start failed with result 'dependency'.
Oct 14 03:16:44 witherspoon systemd[1]: obmc-host-shutdown@0.target: Triggering OnFailure= dependencies.
Oct 14 03:16:44 witherspoon systemd[1]: obmc-chassis-poweroff@0.target: Job obmc-chassis-poweroff@0.target/start failed with result 'dependency'.```

It apparently happens quite a bit.

rfrandse, 2017-10-19:
fixed in the kernel

closed 2018-02-01 by rfrandse