openbmc/openbmc #2488: /home/root/.bash_history and /var/log/wtmp growing uncontrolled?

geissonator, 2017-10-17:
I'm not seeing anything obvious that keeps these from just continuing to grow?

For wtmp, I see reference to using logrotate but not sure we have that on openbmc

geissonator, 2017-10-17:
For the .bash_history file
```
root@witherspoon:~# echo $HISTSIZE
500
root@witherspoon:~# echo $HISTFILESIZE
500
```

geissonator, 2017-10-19:
Team discussion:

- Lets just disable wtmp logging for now.
.bbappend to package that installs wtmp, and basically just don't install it.  Probably something does a touch on wtmp to get 0 file size, which is what wtmp looks at to know if it should write the file.
- Look for way to reduce bash history to something like 10

geissonator, 2017-11-03:
Going to use this to just track the wtmp issue.  That file grows unchecked.  The .bash_history doesn't get much past 30KB before it starts wrapping.

rfrandse, 2017-11-04:
<https://gerrit.openbmc-project.xyz/7785> Disable wtmp function on openbmc systems
Resolves  #2488 /home/root/.bash_history and /var/log/wtmp growing uncontrolled?

geissonator, 2017-11-08:
Rebased fix onto a change from Eddie and put into https://gerrit.openbmc-project.xyz/#/c/7844/

rfrandse, 2017-11-10:
<https://gerrit.openbmc-project.xyz/7844> Disable wtmp function on openbmc systems
Resolves  #2488 /home/root/.bash_history and /var/log/wtmp growing uncontrolled?

geissonator, 2017-12-13:
After an afternoon of searching, I think I've tracked what we need down to https://github.com/systemd/systemd/pull/6006

This is in systemd v234, which is included in yocto 2.4.  That plus a change to pass in "EXTRA_OECONF += " --disable-utmp" should do it the "right way".

geissonator, 2017-12-14:
We have a temporary solution with https://gerrit.openbmc-project.xyz/#/c/7844/.  I'll need to defer the final solution documented above until we get yocto 2.4 in.

geissonator, 2018-03-26:
After moving to 2.4, this was fixed because systemd started honoring the utmp disable flag. https://github.com/openbmc/meta-phosphor/blob/master/common/recipes-core/systemd/systemd_%25.bbappend#L6 already has this in by default for openbmc.

closed 2018-03-26 by geissonator