openbmc/openbmc #2491: Building libapphandler_la-sensor-gen.lo results in memory exhaustion, ICE

amboar, 2017-10-19:
I hit memory exhaustion and received a compiler ICE on two separate runs while attempting to build openbmc/openbmc@51f10b670f4f

```
| virtual memory exhausted: Cannot allocate memory
| Makefile:866: recipe for target 'libapphandler_la-sensor-gen.lo' failed
| make[2]: *** [libapphandler_la-sensor-gen.lo] Error 1
| make[2]: Leaving directory '/home/andrew/src/openbmc/openbmc/build/tmp/work/armv6-openbmc-linux-gnueabi/phosphor-ipmi-host/1.0-r1/build'
| Makefile:1055: recipe for target 'all-recursive' failed
| make[1]: *** [all-recursive] Error 1
| make[1]: Leaving directory '/home/andrew/src/openbmc/openbmc/build/tmp/work/armv6-openbmc-linux-gnueabi/phosphor-ipmi-host/1.0-r1/build'
| Makefile:552: recipe for target 'all' failed
| make: *** [all] Error 2
| ERROR: Function failed: do_compile (log file is located at /home/andrew/src/openbmc/openbmc/build/tmp/work/armv6-openbmc-linux-gnueabi/phosphor-ipmi-host/1.0-r1/temp/log.do_compile.18948)
```

```
| ./arm-openbmc-linux-gnueabi-libtool  --tag=CXX   --mode=compile arm-openbmc-linux-gnueabi-g++  -march=armv6 --sysroot=/home/andrew/src/openbmc/openbmc/build/tmp/sysroots/witherspoon -DHAVE_CONFIG_
H -I. -I../git        -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/home/andrew/src/openbmc/openbmc/build/tmp/work/armv6-openbmc-linux-gnueabi/phosphor-ipmi-host/1.0-r1=/usr/src/de
bug/phosphor-ipmi-host/1.0-r1 -fdebug-prefix-map=/home/andrew/src/openbmc/openbmc/build/tmp/sysroots/x86_64-linux= -fdebug-prefix-map=/home/andrew/src/openbmc/openbmc/build/tmp/sysroots/witherspoon=
 -fvisibility-inlines-hidden -Wall -Werror -c -o libapphandler_la-sensor-gen.lo `test -f 'sensor-gen.cpp' || echo '../git/'`sensor-gen.cpp
| arm-openbmc-linux-gnueabi-libtool: compile:  arm-openbmc-linux-gnueabi-g++ -march=armv6 --sysroot=/home/andrew/src/openbmc/openbmc/build/tmp/sysroots/witherspoon -DHAVE_CONFIG_H -I. -I../git -O2 -
pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/home/andrew/src/openbmc/openbmc/build/tmp/work/armv6-openbmc-linux-gnueabi/phosphor-ipmi-host/1.0-r1=/usr/src/debug/phosphor-ipmi-host/1.0-
r1 -fdebug-prefix-map=/home/andrew/src/openbmc/openbmc/build/tmp/sysroots/x86_64-linux= -fdebug-prefix-map=/home/andrew/src/openbmc/openbmc/build/tmp/sysroots/witherspoon= -fvisibility-inlines-hidde
n -Wall -Werror -c sensor-gen.cpp  -fPIC -DPIC -o .libs/libapphandler_la-sensor-gen.o
| arm-openbmc-linux-gnueabi-g++: internal compiler error: Killed (program cc1plus)
| Please submit a full bug report,
| with preprocessed source if appropriate.
| See <http://gcc.gnu.org/bugs.html> for instructions.
| Makefile:866: recipe for target 'libapphandler_la-sensor-gen.lo' failed
| make[2]: *** [libapphandler_la-sensor-gen.lo] Error 1
| make[2]: Leaving directory '/home/andrew/src/openbmc/openbmc/build/tmp/work/armv6-openbmc-linux-gnueabi/phosphor-ipmi-host/1.0-r1/build'
| Makefile:1055: recipe for target 'all-recursive' failed
| make[1]: *** [all-recursive] Error 1
| make[1]: Leaving directory '/home/andrew/src/openbmc/openbmc/build/tmp/work/armv6-openbmc-linux-gnueabi/phosphor-ipmi-host/1.0-r1/build'
| Makefile:552: recipe for target 'all' failed
| make: *** [all] Error 2
| ERROR: Function failed: do_compile (log file is located at /home/andrew/src/openbmc/openbmc/build/tmp/work/armv6-openbmc-linux-gnueabi/phosphor-ipmi-host/1.0-r1/temp/log.do_compile.25327)
```

amboar, 2017-10-19:
Probably related to https://github.com/openbmc/openbmc/issues/1441, and the corresponding GCC bug https://gcc.gnu.org/bugzilla/show_bug.cgi?id=80290

tomjoseph83, 2017-10-19:
@amboar 1441 had fixed the problem. I haven't across this after that.

amboar, 2017-10-19:
@tomjoseph83 #1441 fixed the problem for the fru-gen.cpp case, in openbmc/phosphor-ipmi-fru. GCC is not fixed, and sensor-gen.cpp in openbmc/phosphor-host-ipmid suffers the same nested structure problem.

bazhenov-dn, 2017-11-01:
I was able to workaround the problem by removing "extern const" type specifiers from the correspondinf *.mako files and regenerating *-gen.cpp.

amboar, 2019-04-02:
Fixed for 7.4/8.2/9.

https://gcc.gnu.org/bugzilla/show_bug.cgi?id=80290#c32

closed 2019-04-02 by amboar