openbmc/openbmc #2492: [xCAT] Set BMC ipsrc dhcp failed using ipmitool

xuweibj, 2017-10-19:
We used ``ipmitool lan set 1 ipsrc dhcp`` and ``ipmitool lan set 1 access on`` to set BMC IP to DHCP. But the result as below:

```
[xCAT Genesis running on mid05tor12cn16 /]# ipmitool lan set 1 ipsrc dhcp
[xCAT Genesis running on mid05tor12cn16 /]# ipmitool lan set 1 access on
IPMI command failed: Command response could not be provided
Unable to Set Channel Access(volatile) for channel 1
[xCAT Genesis running on mid05tor12cn16 /]# ipmitool lan print 1
Set in Progress         : Set Complete
Auth Type Support       : MD5
Auth Type Enable        : Callback : MD5                        
: User     : MD5                       
: Operator : MD5                        
: Admin    : MD5                        
: OEM      : MD5IP 
Address Source       : Static Address
IP Address              : 0.0.0.0
Subnet Mask             : 0.0.0.0
MAC Address             : 70:e2:84:14:30:bf
Default Gateway IP      : 0.0.0.0
802.1q VLAN ID          : Disabled
Cipher Suite Priv Max   : Not Available
Bad Password Threshold  : Not Available
```

I ever tried to reboot BMC, but it's still "Static Address".

@whowutwut @zet809

geissonator, 2017-11-03:
Is this something we need to track internally @xuweibj ?  What's the impact here?

tomjoseph83, 2017-11-03:
@xuweibj The result is expected, as `ipmitool lan set 1 access on` failed, which means applying the DHCP configuration failed. Are you able to reproduce this?

whowutwut, 2017-11-03:
@tomjoseph83 Yes, the message comes out, but i think it actually applies. 
You had mentioned here: https://github.com/openbmc/openbmc/issues/1565#issuecomment-299777089 to use the raw command, and we had changed that initially,  Then around Sept 2017, we found that `access on` works... although we see that error. 

I think the error that @xuweibj  is reporting is the fact that DHCP is set, but `ipmitool lan print 1` still shows:

Address Source       : Static Address

@geissonator Yes, we should be opening CQ defects. .  I mentioned this on this weeks interlock, but this was opened earlier on.

ratagupt, 2017-11-04:
@whowutwut :- seems to me the problem is with set channel access on ..I was able to reproduce with set channel access on..
but the following command works fine..
/tmp/ipmitool -I dbus lan set 1 ipsrc dhcp
/tmp/ipmitool -I dbus raw 0x06 0x40

But however if I try with /tmp/ipmitool -I dbus lan set 1 access on it doesn't work..

@whowutwut ;- Can you also confirm the same?

whowutwut, 2017-11-06:
@xuweibj Can you confirm? 
If we change the access on back to the raw command, does this issue resolve itself?

xuweibj, 2017-11-06:
Get Error when run ipmitool with -I dbus.
```
# ipmitool -I dbus raw 0x06 0x40
Error loading interface dbus
```

Tried to run ``ipmitool raw 0x60 0x40``, it's OK:

```
[xCAT Genesis running on mid05tor12cn16 /]# ipmitool lan print 1
Set in Progress         : Set Complete
Auth Type Support       : MD5
Auth Type Enable        : Callback : MD5
                        : User     : MD5
                        : Operator : MD5
                        : Admin    : MD5
                        : OEM      : MD5
IP Address Source       : Static Address
IP Address              : 172.11.139.16
Subnet Mask             : 255.255.0.0
MAC Address             : 70:e2:84:14:30:bf
Default Gateway IP      : 172.11.253.27
802.1q VLAN ID          : 11
Cipher Suite Priv Max   : Not Available
Bad Password Threshold  : Not Available
[xCAT Genesis running on mid05tor12cn16 /]# ipmitool lan set 1 ipsrc dhcp
[xCAT Genesis running on mid05tor12cn16 /]# ipmitool raw 0x06 0x40
Unable to send RAW command (channel=0x0 netfn=0x6 lun=0x0 cmd=0x40 rsp=0xce): Command response could not be provided
[xCAT Genesis running on mid05tor12cn16 /]# ipmitool lan print 1
Set in Progress         : Set Complete
Auth Type Support       : MD5
Auth Type Enable        : Callback : MD5
                        : User     : MD5
                        : Operator : MD5
                        : Admin    : MD5
                        : OEM      : MD5
IP Address Source       : DHCP Address
IP Address              : 172.12.253.100
Subnet Mask             : 255.255.0.0
MAC Address             : 70:e2:84:14:30:bf
Default Gateway IP      : 172.12.253.27
802.1q VLAN ID          : Disabled
Cipher Suite Priv Max   : Not Available
Bad Password Threshold  : Not Available
```

The setting is successful, but still have error msg "Unable to send RAW command (channel=0x0 netfn=0x6 lun=0x0 cmd=0x40 rsp=0xce): Command response could not be provided"

whowutwut, 2017-11-06:
@xuweibj The `-I dbus` is not valid when we run from compute node host.  They are using this because they are running from within the BMC at the root prompt...  

Looks like this message comes out now... 
```
[xCAT Genesis running on mid05tor12cn16 /]# ipmitool raw 0x06 0x40
Unable to send RAW command (channel=0x0 netfn=0x6 lun=0x0 cmd=0x40 rsp=0xce): Command response could not be provided
```

So since this seems to make the function work better, perhaps we should change this back to using raw commands, but the error message still will mislead anyone looking for errors in the logs..

tomjoseph83, 2017-11-07:
@whowutwut 

a)

I tried to reproduce what @xuweibj explained in the first comment. I was able to reproduce and understood the root cause. The issue is with `lan set 1 access on` option. This ipmitool option sends the `Set Channel Access` command one for the non-volatile and volatile settings. When the command is send twice it leads to undefined behaviour. The plan is to fix the implementation of `Set Channel Access` to handle this appropriately. Once the fix is merged the raw command option would not work.

b) 

`
[xCAT Genesis running on mid05tor12cn16 /]# ipmitool raw 0x06 0x40
Unable to send RAW command (channel=0x0 netfn=0x6 lun=0x0 cmd=0x40 rsp=0xce): Command response could not be provided
`
I did not hit this problem.(I ran from the host, not with the -I dbus option). It seems you are able to reproduce it consistently. Can you share the setup and the exact sequence?

whowutwut, 2017-11-07:
@tomjoseph83  
a) 👍  Can you let us know when this would be fixed? Is there an issue we can track? 
b)  The xCAT genesis kernel on the p9 is a build on RHEL 7.4 snap for Power9. Here's what we are doing... 
```
[xCAT Genesis running on mid05tor12cn05 /]# time /bin/bmcsetup  
ipmitool version 1.8.18
Invalid channel: 16
/bin/bmcsetup: line 150:  1573 Terminated              allowcred.awk
xcat.genesis.bmcsetup: BMC Information obtained from xCAT
xcat.genesis.bmcsetup: NUMBMCS=1 ==> BMC IP=172.11.139.5/255.255.0.0, GW=172.11.253.27, VLAN=11
xcat.genesis.bmcsetup: IPMIVER=2.0, IPMIMFG=42817, XPROD=16975
xcat.genesis.bmcsetup: Auto detecting LAN channel...
xcat.genesis.bmcsetup: Detected LAN channel 1
++ let idev=NUMBMCS
++ '[' static=static ']'
++ '[' 1 -gt 0 ']'
++ let idev=idev-1
++ TRIES=0
++ ipmitool -d 0 lan set 1 ipsrc static
++ '[' 0 -gt 0 ']'
++ let idev=0
++ for b in '$BMCIP'
++ TRIES=0
++ ipmitool -d 0 lan set 1 ipaddr 172.11.139.5
Setting LAN IP Address to 172.11.139.5
++ let idev=idev+1
++ let idev=0
++ for m in '$BMCNM'
++ TRIES=0
++ ipmitool -d 0 lan set 1 netmask 255.255.0.0
Setting LAN Subnet Mask to 255.255.0.0
++ let idev=idev+1
++ '[' '!' -z 172.11.253.27 ']'
++ let idev=0
++ for g in '$BMCGW'
++ TRIES=0
++ ipmitool -d 0 lan set 1 defgw ipaddr 172.11.253.27
Setting LAN Default Gateway IP to 172.11.253.27
++ let idev=idev+1
++ '[' 11 = off ']'
++ let idev=0
++ for b in '$BMCVLAN'
++ TRIES=0
++ ipmitool -d 0 lan set 1 vlan id 11
++ let idev=idev+1
++ '[' '!' -z 1 ']'
++ ipmitool -d 0 raw 0x06 0x40 1 0x42 0x44
Unable to send RAW command (channel=0x0 netfn=0x6 lun=0x0 cmd=0x40 rsp=0xce): Command response could not be provided
++ '[' '!' -z 172.12.253.27 ']'
++ snooze
++ '[' -z 16975 ']'
++ '[' 16975 = 43707 -a 42817 '!=' 0 ']'
++ '[' 16975 = 43707 ']'
++ sleep 1
++ ipmitool lan print 1
++ grep 172.11.139.5
Get Device ID command failed: 0xce Command response could not be provided
++ updateflag.awk 172.12.253.27 3002 'installstatus bmcready'
++ rm -f /tmp/ipmicfg.xml
++ exit

real	0m26.424s
user	0m0.184s
sys	0m0.101s
[xCAT Genesis running on mid05tor12cn05 /]#
```

@zet809, I used the command that you used initially, do you know why `++ ipmitool -d 0 raw 0x06 0x40 1 0x42 0x44` we have `0x42 and 0x44` at the end?

rfrandse, 2017-11-07:
<https://gerrit.openbmc-project.xyz/7817> Apply network settings only if the channel settings is non-volatile
Resolves  #2492 [xCAT] Set BMC ipsrc dhcp failed using ipmitool

whowutwut, 2017-11-09:
@xuweibj Where in the xCAT code is this invoked?

ratagupt, 2018-03-14:
This defect is to track the failure message which is coming once user apply the changes with ipmitool raw 0x06 0x40

dkodihal, 2018-03-14:
Closing based on previous comment, since this is being addressed by #2932.

closed 2018-03-14 by dkodihal

rfrandse, 2018-03-14:
<https://gerrit.openbmc-project.xyz/7817> Apply network settings only if the channel settings is non-volatile
Resolves  #2492 Set BMC ipsrc dhcp failed using ipmitool