openbmc/openbmc #2493: [xCAT] Sometime set VLAN failed using ipmitool

xuweibj, 2017-10-19:
Sometime we got "Unable to Set Channel Access(volatile) for channel 1" response when we want to set LAN IP.
```
[xCAT Genesis running on mid05tor12cn16 /]# /bin/bmcsetup
ipmitool version 1.8.18
Invalid channel: 16
/bin/bmcsetup: line 150: 2900 Terminated allowcred.awk
xcat.genesis.bmcsetup: BMC Information obtained from xCAT
xcat.genesis.bmcsetup: NUMBMCS=1 ==> BMC IP=172.11.139.16/255.255.0.0, GW=172.11.253.27, VLAN=11
xcat.genesis.bmcsetup: IPMIVER=2.0, IPMIMFG=42817, XPROD=16975
xcat.genesis.bmcsetup: Auto detecting LAN channel...
xcat.genesis.bmcsetup: Detected LAN channel 1
Setting LAN IP Address to 172.11.139.16
Setting LAN Subnet Mask to 255.255.0.0
Setting LAN Default Gateway IP to 172.11.253.27
IPMI command failed: Command response could not be provided
Unable to Set Channel Access(volatile) for channel 1
Get Device ID command failed: 0xce Command response could not be provided
[xCAT Genesis running on mid05tor12cn16 /]# ipmitool lan print 1
Set in Progress : Set Complete
Auth Type Support : MD5
Auth Type Enable : Callback : MD5
: User : MD5
: Operator : MD5
: Admin : MD5
: OEM : MD5
IP Address Source : Static Address
IP Address : 172.11.139.16
Subnet Mask : 255.255.0.0
MAC Address : 70:e2:84:14:30:bf
Default Gateway IP : 172.11.253.27
802.1q VLAN ID : Disabled
Cipher Suite Priv Max : Not Available
Bad Password Threshold : Not Available
```

The most time it's OK.
```
[xCAT Genesis running on mid05tor12cn16 /]# /bin/bmcsetup
ipmitool version 1.8.18
Invalid channel: 16
/bin/bmcsetup: line 150: 3049 Terminated allowcred.awk
xcat.genesis.bmcsetup: BMC Information obtained from xCAT
xcat.genesis.bmcsetup: NUMBMCS=1 ==> BMC IP=172.11.139.16/255.255.0.0, GW=172.11.253.27, VLAN=11
xcat.genesis.bmcsetup: IPMIVER=2.0, IPMIMFG=42817, XPROD=16975
xcat.genesis.bmcsetup: Auto detecting LAN channel...
xcat.genesis.bmcsetup: Detected LAN channel 1
Setting LAN IP Address to 172.11.139.16
Setting LAN Subnet Mask to 255.255.0.0
Setting LAN Default Gateway IP to 172.11.253.27
IPMI command failed: Command response could not be provided
Unable to Set Channel Access(non-volatile) for channel 1
Get Device ID command failed: 0xce Command response could not be provided
[xCAT Genesis running on mid05tor12cn16 /]# ipmitool lan print 1
Set in Progress : Set Complete
Auth Type Support : MD5
Auth Type Enable : Callback : MD5
: User : MD5
: Operator : MD5
: Admin : MD5
: OEM : MD5
IP Address Source : Static Address
IP Address : 172.11.139.16
Subnet Mask : 255.255.0.0
MAC Address : 70:e2:84:14:30:bf
Default Gateway IP : 172.11.253.27
802.1q VLAN ID : 11
Cipher Suite Priv Max : Not Available
Bad Password Threshold : Not Available
```

@whowutwut @zet809

whowutwut, 2017-10-30:
@zet809 @ratagupt Did we have a conclusion for this error message when we were investigating? 
I forgot whether it was due to some version of IPMI.. or that was a VLAN message...don't remember. .  It seems like the error message will print, but it actually does kick off the BMC to do some action.

ratagupt, 2017-10-30:
@whowutwut :- I am not sure why this is coming something new for me,```Get Device ID command failed: 0xce Command response could not be provided```

ratagupt, 2017-10-30:
@prkatti1 :- Can you verify it on master?

whowutwut, 2017-11-02:
@ratagupt  I don't think these messages mean anything regarding VLAN, on scenario w/o VLAN passed in, we also see the messages: 

```
[Wed Nov  1 17:11:52 2017]
[Wed Nov  1 17:12:00 2017]IPMI command failed: Command response could not be provided
[Wed Nov  1 17:12:00 2017]Unable to Set Channel Access(volatile) for channel 1
[Wed Nov  1 17:12:06 2017]Get Device ID command failed: 0xce Command response could not be provided
``

whowutwut, 2017-11-02:
@xuweibj  We need to create all issues in CQ and it will be bridges automatically to this GitHub repo if it is in fact an issue, otherwise we will not get the correct focus on this.  

https://w3.rchland.ibm.com/projects/bestquest/?defect=SW406813 

@ratagupt  we need you guys to verify that this is just a print issue with "lan print 1" and that VLAN is infact working and set on the BMC, if it's just a display error, then it's not a big deal, but if we are not consistently setting VLAN, then it is a very big deal.

ratagupt, 2017-11-02:
@whowutwut :- I am thinking why this message is always ```Get Device ID command failed: 0xce Command response could not be provided``` as part of set channel access.

prkatti1, 2017-11-02:
I tried multiple times on latest GHE: I did not see this issue
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="ibm-v1.99.10"
VERSION_ID="ibm-v1.99.10-113-g65edf7d-r9-0-g67617be"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) ibm-v1.99.10"
BUILD_ID="ibm-v1.99.10-113-g65edf7d-r9"

Commands I used:

root@witherspoon:/tmp# /tmp/ipmitool -I dbus lan set 1 ipsrc static
root@witherspoon:/tmp# /tmp/ipmitool -I dbus lan set 1 ipaddr 10.5.5.4
Setting LAN IP Address to 10.5.5.4
root@witherspoon:/tmp#
root@witherspoon:/tmp# /tmp/ipmitool -I dbus lan set 1 netmask 255.255.255.0
Setting LAN Subnet Mask to 255.255.255.0
root@witherspoon:/tmp#
root@witherspoon:/tmp# /tmp/ipmitool -I dbus lan set 1 defgw ipaddr 10.5.5.1
Setting LAN Default Gateway IP to 10.5.5.1
root@witherspoon:/tmp# /tmp/ipmitool -I dbus lan set 1 vlan id 50
root@witherspoon:/tmp# /tmp/ipmitool -I dbus raw 0x06 0x40
[10024.710000] ftgmac100 1e660000.ethernet eth0: NCSI interface up

root@witherspoon:/tmp# /tmp/ipmitool -I dbus lan print
Set in Progress         : Set Complete
Auth Type Support       : MD5
Auth Type Enable        : Callback : MD5
                        : User     : MD5
                        : Operator : MD5
                        : Admin    : MD5
                        : OEM      : MD5
IP Address Source       : Static Address
IP Address              : 10.5.5.4
Subnet Mask             : 255.255.255.0
MAC Address             : 70:e2:84:14:16:6c
[10073.800000] ucd9000 11-0064: Device ID UCD90160|2.3.4.0000|110603
[10076.040000] ucd9000 11-0064: Device ID UCD90160|2.3.4.0000|110603

Default Gateway IP      : 10.5.5.1
802.1q VLAN ID          : 50
Cipher Suite Priv Max   : Not Available
Bad Password Threshold  : Not Available

ratagupt, 2017-11-02:
@whowutwut :- Hopefully you are on the GHE build

xuweibj, 2017-11-03:
Hi @ratagupt , I tested on ``ibm-v1.99.10-113-g65edf7d-r3-0-g9e4f715``. Do you mean that maybe this issue has been solved on ``ibm-v1.99.10-113-g65edf7d-r9-0-g67617be``?

tomjoseph83, 2017-11-03:
@xuweibj Please test with the latest GHE..and I think it is `r10` is the latest one.

whowutwut, 2017-11-03:
@tomjoseph83 Check out above https://github.com/openbmc/openbmc/issues/2493#issuecomment-341307022 

> I am thinking why this message is always Get Device ID command failed: 0xce Command response could not be provided as part of set channel access.
Might be it is due to some latest changes..

-r10 is still being respun supposedly, we could grab it and put it on our machine. 

The one difference here is that the testing is being done on the BMC , where our testing is being done on the host.  @prkatti1 Can you boot the host and also try inband from host? 

If using RHEL: 
```
yum -y install ipmitool
modprobe ipmi_devintf
ipmitool lan print 1
```

ratagupt, 2017-11-04:
@whowutwut :- Can you please share the commands which is being run as part of bmcsetup script?

ratagupt, 2017-11-04:
@whowutwut :- https://github.com/openbmc/openbmc/issues/2492 and https://github.com/openbmc/openbmc/issues/2493 are related to that set channel access on is broken..

Please try with raw command..
ipmitool -I dbus raw 0x06 0x40 (apply changes)

whowutwut, 2017-11-06:
@xuweibj  Do you recall when the VLAN failed to set, was the BMC still ping-able with the VLAN IP address? 

Also, how often did you see this issue? 

@ratagupt  Here's the gist of it... 
```
        while ! ipmitool -d $idev lan set $LANCHAN ipsrc static; do
        while ! ipmitool -d $idev lan set $LANCHAN ipaddr $b; do
        while ! ipmitool -d $idev lan set $LANCHAN netmask $m; do
            while ! ipmitool -d $idev lan set $LANCHAN defgw ipaddr $g; do
        while ! ipmitool -d $idev lan set $LANCHAN ipsrc $IPCFGMETHOD; do
    while ! ipmitool raw 0xc 1 $LANCHAN 0x14 0 0; do
        while ! ipmitool -d $idev lan set $LANCHAN vlan id $b; do
    ipmitool -d 0 lan set $LANCHAN access on
```

@zet809, do you recall why we changed the raw command back to `access on` ... ?

ratagupt, 2017-11-06:
@whowutwut :- please try with raw command and share the result,On my setup also,I can see that set access on is broken...

xuweibj, 2017-11-06:
@whowutwut , no, when set VLAN failed, the BMC is not ping-able with the VLAN IP address.
I tried almost 10 times, and met once.

@ratagupt 
I replaced ``ipmitool -d 0 lan set $LANCHAN access on`` to ``ipmitool -d 0 raw 0x06 0x40``. And result is below:
```
[xCAT Genesis running on mid05tor12cn16 /]# /bin/bmcsetup
ipmitool version 1.8.18
Invalid channel: 16
/bin/bmcsetup: line 150:  3949 Terminated              allowcred.awk
xcat.genesis.bmcsetup: BMC Information obtained from xCAT
xcat.genesis.bmcsetup: NUMBMCS=1 ==> BMC IP=172.11.139.16/255.255.0.0, GW=172.11.253.27, VLAN=11
xcat.genesis.bmcsetup: IPMIVER=2.0, IPMIMFG=42817, XPROD=16975
xcat.genesis.bmcsetup: Auto detecting LAN channel...
xcat.genesis.bmcsetup: Detected LAN channel 1
Setting LAN IP Address to 172.11.139.16
Setting LAN Subnet Mask to 255.255.0.0
Setting LAN Default Gateway IP to 172.11.253.27
Unable to send RAW command (channel=0x0 netfn=0x6 lun=0x0 cmd=0x40 rsp=0xce): Command response could not be provided
Get Device ID command failed: 0xce Command response could not be provided
[xCAT Genesis running on mid05tor12cn16 /]# ipmitool lan print 1
Set in Progress         : Set Complete
Auth Type Support       : MD5
Auth Type Enable        : Callback : MD5
                        : User     : MD5
                        : Operator : MD5
                        : Admin    : MD5
                        : OEM      : MD5
IP Address Source       : Static Address
IP Address              : 172.11.139.16
Subnet Mask             : 255.255.0.0
MAC Address             : 70:e2:84:14:30:bf
Default Gateway IP      : 172.11.253.27
802.1q VLAN ID          : 11
Cipher Suite Priv Max   : Not Available
Bad Password Threshold  : Not Available
```

zet809, 2017-11-06:
hi, @whowutwut , in the issue #1565 , it mentioned that the `set access on` interface maybe ready and a NCIS up event happen on the BMC, so we switch.

whowutwut, 2017-11-07:
@ratagupt @tomjoseph83 can you look at the above issue mentioned by @zet809 

Seems like the "access on" was implemented and now the recommendation is to go back to using the raw command.....

tomjoseph83, 2017-11-07:
@whowutwut @zet809 `access on` has a issue which i would fix up. After which the raw command option would not work. Our preference would be to continue using the `access on` option.

amboar, 2017-11-13:
@tomjoseph83 @xuweibj @whowutwut @ratagupt @prkatti1 This issue has been tagged as `ibm-internal` because it references IBM-internal tags, commits, machines or issues. This is a public issue tracker and is for the benefit of the community. If any of the details are not publicly accessible then the issue should be filed elsewhere, or reworded to reflect the relevant publicly accessible details.

whowutwut, 2018-03-06:
The internal defect is rejected for this one.  This is not easily re-creatable for us because we have to be watching the discovery process since the ipmi output is not saved in syslog.  

So feel free to close this as I don't think anything will be done at this point .

Ramkosh, 2018-03-13:
@whowutwut hi! we're actually working on the similair issue right now.
To be specific, I'm fixing a couple of issues with phosphor-host-ipmid regarding how it handles IP (and related) object creation.
For example, here's our patch for IP object handling: https://gerrit.openbmc-project.xyz/#/c/9409/

As for VLAN handling, I saw you mention earlier (#2185 ) that VLAN is created as a new object with it's own path, not as a child object for physical interface. So when ipmid will use this path (e.g.: `/eth0/ipv4/` instead of `/eth0_<vlanID>/ipv4/`) it will use only IP objects from that path, not the one provided via ipmitool. 
I guess I'll take on #2407 as a part of a fix, because we either need to change vlan dbus object path generation or add support for current implementation.

whowutwut, 2018-03-13:
Hi @Ramkosh! Interesting! I've opened these issues on behalf of the xcat project that manages the OpenBMC boxes but currently we interface in exclusively REST. The ipmi issues was mostly during discovery of the hardware, we configured the bmc via inband communication using ipmitool. 

We will re-visit the performance of the ipmi 2.0 support when it's available and see how it compares to the REST, if it's faster, we may switch back to that.. 

But this is great that you are looking into this issue! Thanks!

closed 2018-07-18 by rfrandse