openbmc/openbmc #2494: BMC software manager inventory empty

gkeishin, 2017-10-19:
`Driver Info`
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.10-162"
VERSION_ID="v1.99.10-162-g965231b"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.10-162"
BUILD_ID="v1.99.10"
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-1af701e56cd6bd5399b517e276d479482b702839 #1 Thu Oct 19 03:45:32 UTC 2017 armv6l GNU/Linux
root@witherspoon:~#
```

`Software enumerate`

```
Executing: GET  /xyz/openbmc_project/software/735d47c9

/xyz/openbmc_project/software/735d47c9
{
    "Activation": "xyz.openbmc_project.Software.Activation.Activations.Active",
    "Path": "",
    "Priority": 0,
    "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.BMC",
    "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.None",
    "Version": "v1.99.10-162-g965231b",
    "associations": [
        [
            "inventory",
            "activation",
            "/xyz/openbmc_project/inventory/system/chassis/motherboard/boxelder/bmc"
        ]
    ]
}
-bash-4.1$ ./rest_cmd -i wsbmc011 -o GET -u /xyz/openbmc_project/software/enumerate
Executing: GET  /xyz/openbmc_project/software/enumerate

/xyz/openbmc_project/software/enumerate
{
    "/xyz/openbmc_project/software/06edbc22": {
        "Activation": "xyz.openbmc_project.Software.Activation.Activations.Active",
        "ExtendedVersion": "op-build-v1.7-1421-g2f4cbfe,buildroot-2017.08-6-g319c6e1,skiboot-v5.8-130-g712837cedca0,hostboot-8e7e2ad,linux-4.13.4-openpower1-p1c668f0,petitboot-v1.6.0-p85a1eab,machine-xml-1bd247d-p7c6c471,occ-60c09cd,hostboot-binaries-b7d8417,capp-ucode-p9-dd2-v1,sbe-ff713df",
        "Path": "",
        "Priority": 0,
        "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.Host",
        "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.None",
        "Version": "IBM-witherspoon-ibm-OP9_v1.19_1.52",
        "associations": [
            [
                "inventory",
                "activation",
                "/xyz/openbmc_project/inventory/system/chassis"
            ]
        ]
    },
    "/xyz/openbmc_project/software/06edbc22/inventory": {
        "endpoints": [
            "/xyz/openbmc_project/inventory/system/chassis"
        ]
    },
    "/xyz/openbmc_project/software/06edbc22/software_version": {
        "endpoints": [
            "/xyz/openbmc_project/software"
        ]
    },
    "/xyz/openbmc_project/software/3eb1a8be": {
        "Activation": "xyz.openbmc_project.Software.Activation.Activations.Active",
        "Path": "",
        "Priority": 1,
        "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.BMC",
        "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.None",
        "Version": "v1.99.10-160-g4cacdbd",
        "associations": [
            [
                "inventory",
                "activation",
                "/xyz/openbmc_project/inventory/system/chassis/motherboard/boxelder/bmc"
            ]
        ]
    },
    "/xyz/openbmc_project/software/3eb1a8be/software_version": {
        "endpoints": [
            "/xyz/openbmc_project/software"
        ]
    },
    "/xyz/openbmc_project/software/735d47c9": {
        "Activation": "xyz.openbmc_project.Software.Activation.Activations.Active",
        "Path": "",
        "Priority": 0,
        "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.BMC",
        "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.None",
        "Version": "v1.99.10-162-g965231b",
        "associations": [
            [
                "inventory",
                "activation",
                "/xyz/openbmc_project/inventory/system/chassis/motherboard/boxelder/bmc"
            ]
        ]
    },
    "/xyz/openbmc_project/software/735d47c9/software_version": {
        "endpoints": [
            "/xyz/openbmc_project/software"
        ]
    },
    "/xyz/openbmc_project/software/active": {
        "endpoints": [
            "/xyz/openbmc_project/software/06edbc22",
            "/xyz/openbmc_project/software/735d47c9",
            "/xyz/openbmc_project/software/3eb1a8be"
        ]
    },
    "/xyz/openbmc_project/software/functional": {
        "endpoints": [
            "/xyz/openbmc_project/software/06edbc22",
            "/xyz/openbmc_project/software/735d47c9"
        ]
    }
}
```

gkeishin, 2017-10-19:
[x] Journad log 
[BMC_software_inventory.txt](https://github.com/openbmc/openbmc/files/1398352/BMC_software_inventory.txt)

gtmills, 2017-10-19:
Caused by an error in the mapper call.

From the journal:
```Oct 12 18:41:02 witherspoon phosphor-image-updater[1175]: Error in mapper GetSubTreePath```

https://github.com/openbmc/phosphor-bmc-code-mgmt/blob/master/item_updater.cpp#L508

gtmills, 2017-10-19:
curl -c cjar -b cjar -k -H "Content-Type: application/json" -X GET  https://${BMC_IP}/xyz/openbmc_roject/software/735d47c9/inventory
{
  "data": {
    "endpoints": [
      ""
    ]
  }, 
  "message": "200 OK", 
  "status": "ok"
}

gkeishin, 2017-10-23:
Any new outlook  fix on this one? @gtmills

gtmills, 2017-10-25:
Next sprint probably. I am full this sprint.

rfrandse, 2017-12-07:
<https://gerrit.openbmc-project.xyz/8121> Software.BMC.Updater: Add dependency to op-vpd-parser
Resolves  #2494 BMC software manager inventory missing

closed 2018-01-18 by geissonator