openbmc/openbmc #2495: Time Manager: terminate called after throwing an instance of 'sdbusplus::xyz::openbmc_project::Time::Internal::Error::MethodError'

gkeishin, 2017-10-19:
```
Oct 19 15:36:00 witherspoon phosphor-timemanager[1349]: Failed to invoke Dbus method
Oct 19 15:36:00 witherspoon phosphor-timemanager[1349]: terminate called after throwing an instance of 'sdbusplus::xyz::openbmc_project::Time::Internal::Error::MethodError'
Oct 19 15:36:00 witherspoon phosphor-timemanager[1349]:   what():  xyz.openbmc_project.Time.Internal.Error.MethodError: Failed to invoke Dbus method
Oct 19 15:36:00 witherspoon systemd[1]: Started Process Core Dump (PID 1353/UID 0).
Oct 19 15:36:00 witherspoon systemd[1]: xyz.openbmc_project.Time.Manager.service: Main process exited, code=dumped, status=6/ABRT
Oct 19 15:36:00 witherspoon systemd[1]: Failed to start Phosphor Time Manager daemon.
Oct 19 15:36:00 witherspoon systemd[1]: xyz.openbmc_project.Time.Manager.service: Unit entered failed state.
Oct 19 15:36:00 witherspoon systemd[1]: xyz.openbmc_project.Time.Manager.service: Failed with result 'core-dump
```
```
------------------------------------------------------------------------ 
DRIVER INFO : FW Level	Executed : cat /etc/os-release
------------------------------------------------------------------------ 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.10-165"
VERSION_ID="v1.99.10-165-g2cb2894"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.10-165"
BUILD_ID="v1.99.10"
```
- [x] Journald 
[Time_manager_crashed.txt](https://github.com/openbmc/openbmc/files/1399164/Time_manager_crashed.txt)

vishwabmc, 2017-10-19:
Likely a version bump needed for phosphor-time-manager looks like

gkeishin, 2017-10-19:
Pervasive on the master latest build.

mine260309, 2017-10-20:
The crash is due to DBus method error is that time manager fails to get Host State service.

When time manager starts, it need to check the host state, previously it is using power service to check pgood to determine host status, now it's changed to use host state service.

While on BMC starts, time manager starts before host state service, and that's why it gets crash.

So there could be two fixes:
1. Adjust time manager service's unit file to make it starts after state service;
2. Handle the exception and retry to get host state service in time manager.

I would prefer option 1, but it looks a bit weird for time manager service to depend on host state service.

Any better idea? @vishwabmc @bradbishop

mine260309, 2017-10-20:
I just find @bradbishop has already put a fix in https://gerrit.openbmc-project.xyz/#/c/7447/
It adds dependency to host state service, great :)

rfrandse, 2017-10-20:
<https://gerrit.openbmc-project.xyz/7447> timemgr: Use correct timemanager deps
Resolves  #2495 sdbusplus::xyz::openbmc_project::Time::Internal::Error::MethodError

closed 2017-10-20 by geissonator

gkeishin, 2017-10-20:
Verified on
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.10-167"
VERSION_ID="v1.99.10-167-gb08d68f"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.10-167"
BUILD_ID="v1.99.10"
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-1af701e56cd6bd5399b517e276d479482b702839 #1 Thu Oct 19 03:24:58 UTC 2017 armv6l GNU/Linux
root@witherspoon:~#
```