openbmc/openbmc #2497: Poweron watchdog logic

yongli3, 2017-10-20:
I received the “phosphor-watchdog Timer Expired” message during host power on. I would like to know more detailed information about such watchdog.

It seems that the watchdog will be enabled at the beginning of host poweron, but I did find how to disable it. Is it disabled by other service? When to disable it?

mine260309, 2017-10-20:
Yes, the watchdog is started on host power on, which is used to monitor is host is alive.
In normal case, when host starts, it will send IPMI commands to kick watchdog and so everything would work fine.
If host fails to start, the watchdog eventually timeout.

You can manually disable watchdog by systemctl, e.g. `systemctl stop phosphor-watchdog@poweron.service`

yongli3, 2017-10-20:
@mine260309 Thanks for your comment. Regarding the "host will send IPMI commands".  Is there any running application on the host? Or just send out from the BIOS?
Could you share more information about the IPMI command?  is it just disable timer, or set the interval to UINT64_MAX ? I would like to trace the system, to check if there is any command is sent out.

edtanous, 2017-10-21:
Yong, feel free to give me a call Monday and I can give you a rundown of how the host watchdog is intended to work.  Based on what I see, watchdog is working as intended, it just needs some tuning for the platform we're working on.

yongli3, 2017-10-22:
Thanks @edtanous !

rfrandse, 2017-11-03:
Reopen if there still and issue.

closed 2017-11-03 by rfrandse