openbmc/openbmc #2498: VFS: Cannot open root device "ubiblock0_0" or unknown-block(0,0): error -6

geissonator, 2017-10-20:
Was using w54 today for some debug.  I had created some /usr/local/bin and lib directories to test out patches, and had done a bunch of boot tests.  

I then wanted to clean up, so I kicked off the jenkins job to update the system to the latest openpower.xyz image (#1048-b08d68fa39b9d4, https://openpower.xyz/job/openbmc-build/1048/)

The update seemed to go ok, but then it got stuck in an endless reboot.

We've seen this elsewhere, not sure if we had an issue tracking it or not.
```
U-Boot 2016.07 (Aug 29 2017 - 21:30:45 +0000)

       Watchdog enabled
DRAM:  496 MiB
Flash: 32 MiB
In:    serial
Out:   serial
Err:   serial
Net:   aspeednic#0
Hit any key to stop autoboot:  0 
ubi0: attaching mtd1
ubi0: scanning is finished

. done
ubi0: attached mtd1 (name "mtd=2", size 31 MiB)
ubi0: PEB size: 65536 bytes (64 KiB), LEB size: 65408 bytes
ubi0: min./max. I/O unit sizes: 1/1, sub-page size 1
ubi0: VID header offset: 64 (aligned 64), data offset: 128
ubi0: good PEBs: 504, bad PEBs: 0, corrupted PEBs: 0
ubi0: user volume: 4, internal volumes: 1, max. volumes count: 128
ubi0: max/mean erase counter: 177/153, WL threshold: 4096, image sequence number: 1358235831
ubi0: available PEBs: 48, total reserved PEBs: 456, PEBs reserved for bad PEB handling: 0
Read 0 bytes from volume kernel-bcd6d72f to 80800000
No size specified -> Using max size (2114676)
## Loading kernel from FIT Image at 80800000 ...
   Using 'conf@1' configuration
   Trying 'kernel@1' kernel subimage
     Description:  Linux kernel
     Type:         Kernel Image
     Compression:  uncompressed
     Data Start:   0x8080012c
     Data Size:    2078664 Bytes = 2 MiB
     Architecture: ARM
     OS:           Linux
     Load Address: 0x80001000
     Entry Point:  0x80001000
     Hash algo:    sha1
     Hash value:   d5f53074beb20733cec4d4a75171f99c18255068
   Verifying Hash Integrity ... sha1+ OK
## Loading fdt from FIT Image at 80800000 ...
   Using 'conf@1' configuration
   Trying 'fdt@1' fdt subimage
     Description:  Flattened Device Tree blob
     Type:         Flat Device Tree
     Compression:  uncompressed
     Data Start:   0x809fb9e8
     Data Size:    34115 Bytes = 33.3 KiB
     Architecture: ARM
     Hash algo:    sha1
     Hash value:   82f97a3ec535e66f3c0c3c14e1cd76c1c5a31f89
   Verifying Hash Integrity ... sha1+ OK
   Booting using the fdt blob at 0x809fb9e8
   Loading Kernel Image ... OK
   Loading Device Tree to 9eb7a000, end 9eb85542 ... OK

Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.10.17-1af701e56cd6bd5399b517e276d479482b702839 (fspcibld@tulpok01p4) (gcc version 6.2.0 (GCC) ) #1 Thu Oct 19 03:45:32 UTC 2017
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[    0.000000] OF: fdt:Machine model: Witherspoon BMC
[... 199 lines cut ...]
[   15.470000]  (driver?)
[   15.470000] 1f06             128 mtdblock6 
[   15.470000]  (driver?)
[   15.530000] 1f07           32256 mtdblock7 
[   15.530000]  (driver?)
[   15.530000] 1f08          131072 mtdblock8 
[   15.530000]  (driver?)
[   15.530000] fe00           14860 ubiblock4_0 
[   15.530000]  (driver?)
[   15.530000] Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)
[   15.530000] CPU: 0 PID: 1 Comm: swapper Not tainted 4.10.17-1af701e56cd6bd5399b517e276d479482b702839 #1
[   15.530000] Hardware name: ASpeed SoC
[   15.530000] [<80010080>] (unwind_backtrace) from [<8000d934>] (show_stack+0x20/0x24)
[   15.530000] [<8000d934>] (show_stack) from [<801dbf2c>] (dump_stack+0x20/0x28)
[   15.530000] [<801dbf2c>] (dump_stack) from [<800a4a40>] (panic+0xc8/0x264)
[   15.530000] [<800a4a40>] (panic) from [<805884f8>] (mount_block_root+0x214/0x300)
[   15.530000] [<805884f8>] (mount_block_root) from [<80588740>] (mount_root+0x60/0x7c)
[   15.530000] [<80588740>] (mount_root) from [<805888c8>] (prepare_namespace+0x16c/0x1d0)
[   15.530000] [<805888c8>] (prepare_namespace) from [<805880b0>] (kernel_init_freeable+0x184/0x1cc)
[   15.530000] [<805880b0>] (kernel_init_freeable) from [<803fdc60>] (kernel_init+0x18/0x104)
[   15.530000] [<803fdc60>] (kernel_init) from [<8000a5e8>] (ret_from_fork+0x14/0x2c)

U-Boot 2016.07 (Aug 29 2017 - 21:30:45 +0000)

```

geissonator, 2017-10-20:
```
ast# printenv
0348592f=2
062c11d3=2
14572ff5=2
14960761=2
16b40ebe=2
1dc52b48=2
261e8789=2
2a685513=2
2f8083e2=2
31ecc836=2
3563ed7a=2
379bdef3=2
3eb1a8be=2
457d3fd2=2
470a1084=2
4b16b3f3=2
57f8fc93=2
5c7b7ade=2
5f19f34d=2
5f60a712=2
613d4e17=2
616f60e8=2
63e69906=2
6670a0a3=2
67d1a965=2
69bc82cf=2
6c2d4b9e=2
735d47c9=2
765c626c=1
77b5d0f4=2
7a23bd7a=2
7c535624=2
84afb9e1=2
84b49aab=2
865ca9d5=2
868f546d=2
87796854=2
8981856a=2
8d93cc83=2
8f1b17a7=2
91c5117a=2
92861bce=2
9897095e=2
99e47709=2
9a0de8e9=2
9d6e71ce=2
9dd49241=2
9e91f471=2
a72dd72a=2
ab23271a=2
adf763d5=2
aebc3de9=2
b15dfb54=2
baudrate=115200
bcd6d72f=0
beb33ef3=2
bootargs=console=ttyS4,115200n8 ubi.mtd=obmc-ubi,0,0,0 ubi.mtd=alt-obmc-ubi,0,0,4 ro rootfstype=squashfs
bootcmd=run set_bootargs; run obmc_bootcmd
bootdelay=2
c63c3271=2
cfe4715f=2
d1281ac6=2
d361b844=2
da042ae7=2
dd14b06d=2
ddf04517=2
e9073611=2
eab06bee=2
ethact=aspeednic#0
ethaddr=70:e2:84:14:23:e1
f6a9c4b4=2
fa42a010=2
fde6eb2b=2
fde70ad8=2
fe60a46c=2
ff782109=2
kernelname=kernel-bcd6d72f
loadaddr=80800000
mtddevname=u-boot
mtddevnum=0
mtdids=nor0=bmc
mtdparts=mtdparts=bmc:384k(u-boot),128k(u-boot-env),-(obmc-ubi)
obmc_bootcmd=ubi part obmc-ubi; ubi read ${loadaddr} ${kernelname}; bootm ${loadaddr} 
partition=nor0,0
root=/dev/ubiblock0_0
set_bootargs=setenv bootargs console=ttyS4,115200n8 ubi.mtd=obmc-ubi,0,0,0 ubi.mtd=alt-obmc-ubi,0,0,4 ro rootfstype=squashfs ubi.block=${ubiblock} root=${root} 
spi_dma=yes
stderr=serial
stdin=serial
stdout=serial
ubiblock=4,0
verify=yes

Environment size: 1695/65531 bytes
```

mdmillerii, 2017-10-20:
This seems to be breakage in our code update process.  We set `ubi.block=4,0` to say create `/dev/ubiblock4_0` but still have `root=/dev/ubiblock0_0`.
```
[    0.000000] Kernel command line: console=ttyS4,115200n8 ubi.mtd=obmc-ubi,0,0,0 ubi.mtd=alt-obmc-ubi,0,0,4 ro rootfstype=squashfs ubi.block=4,0 root=/dev/ubiblock0_0
```

which comes from
```
root=/dev/ubiblock0_0
ubiblock=4,0
```

How did we get to an environment with these two mismatched?

On a secondary note, does anyone know where these hash=2 environment variables are coming from?

```
fde6eb2b=2
fde70ad8=2
fe60a46c=2
```

amboar, 2017-10-22:
> On a secondary note, does anyone know where these hash=2 environment variables are coming from?

> fde6eb2b=2
> fde70ad8=2
> fe60a46c=2

I assumed they were mapping priority value per image. Though given the number of them, it feels like they're not being cleaned up with the removal of the associated image.

saqibkh, 2017-10-22:
The version priorities are stored in /var and a copy of that is stored under u-boot-env(BMC ubi volume that stores uboot env variables). When the version is deleted, the priority file under /var is cleared but u-boot-env variables are not cleared.

We need to explicitly call "fw_setenv <versionID>" to be able to delete them inside the deleteinterface.

anoo1, 2017-10-23:
> We've seen this elsewhere, not sure if we had an issue tracking it or not.

First reported here: https://github.com/openbmc/openbmc/issues/2498

saqibkh, 2017-10-31:
This defect was fixed by https://gerrit.openbmc-project.xyz/#/c/7649/

geissonator, 2017-10-31:
Still hitting this with master code that has that fix :(

anoo1, 2018-02-27:
Please reopen if seen with latest master.

closed 2018-02-27 by anoo1