openbmc/openbmc #2501: Setting NTP as mode and HOST as owner at the same time not throwing error

rahulmah, 2017-10-23:
I believe only BMC can be set to NTP (this is how it use to work in old ORG interface). So setting HOST as owner and NTP as mode at the same time should not be allowed. But at present in xyz, it is not throwing error while setting these settings.

```
bash-4.1$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -d "{\"data\": \"xyz.openbmc_project.Time.Synchronization.Method.NTP\"}" -X PUT  https://<BMC_IP>/xyz/openbmc_project/time/sync_method/attr/TimeSyncMethod
{
  "data": null,
  "message": "200 OK",
  "status": "ok"
}bash-4.1$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -d "{\"data\": \"xyz.openbmc_project.Time.Owner.Owners.Host\"}" -X PUT  https://<BMC_IP>xyz/openbmc_project/time/owner/attr/TimeOwner
{
  "data": null,
  "message": "200 OK",
  "status": "ok"
}bash-4.1$
bash-4.1$ curl -c cjar -b cjar -k -X GET  https://<BMC_IP>/xyz/openbmc_project/time/owner/attr/TimeOwner
{
  "data": "xyz.openbmc_project.Time.Owner.Owners.Host",
  "message": "200 OK",
  "status": "ok"
}bash-4.1$ curl -c cjar -b cjar -k -X GET  https://<BMC_IP>/xyz/openbmc_project/time/sync_method/attr/TimeSyncMethod
{
  "data": "xyz.openbmc_project.Time.Synchronization.Method.NTP",
  "message": "200 OK",
  "status": "ok"
}bash-4.1$
```

mine260309, 2017-10-25:
There will be no error log in such case.
Each settings combinations are allowed which is set by user.
The different combinations result in different behavior on settings BMC/Host time, as https://github.com/openbmc/phosphor-time-manager/blob/master/README.md#time-settings describes.

rahulmah, 2017-10-30:
Okay. Closing the issue.

closed 2017-10-30 by rahulmah