openbmc/openbmc #2503: phosphor-reboot-host@0.service: Main process exited, code=exited, status=1/FAILURE

gkeishin, 2017-10-23:
r/r BMC test

`System state`
```
root@witherspoon:~# obmcutil power
pgood = 0
state = 0
pgood_timeout = 10
root@witherspoon:~# obmcutil state
CurrentBMCState     : xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState   : xyz.openbmc_project.State.Chassis.PowerState.Off
CurrentHostState    : xyz.openbmc_project.State.Host.HostState.Off
root@witherspoon:~#
```

`Journald snippet`
```
root@witherspoon:~# systemctl status phosphor-reboot-host@0.service | cat
● phosphor-reboot-host@0.service - Reboot host0
   Loaded: loaded (/lib/systemd/system/obmc-host-reboot@0.target.requires/../phosphor-reboot-host@.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Thu 2017-10-12 12:05:40 UTC; 7min ago
  Process: 1944 ExecStart=/bin/systemctl start obmc-host-start@%i.target (code=exited, status=1/FAILURE)
 Main PID: 1944 (code=exited, status=1/FAILURE)

Oct 12 12:05:22 witherspoon systemd[1]: Started Reboot host0.
Oct 12 12:05:39 witherspoon systemctl[1944]: Job for obmc-host-start@0.target canceled.
Oct 12 12:05:40 witherspoon systemd[1]: phosphor-reboot-host@0.service: Main process exited, code=exited, status=1/FAILURE
Oct 12 12:05:41 witherspoon systemd[1]: phosphor-reboot-host@0.service: Unit entered failed state.
Oct 12 12:05:41 witherspoon systemd[1]: phosphor-reboot-host@0.service: Failed with result 'exit-code'.
root@witherspoon:~#

```

`Driver info`
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.10-174"
VERSION_ID="v1.99.10-174-gc9f3377"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.10-174"
BUILD_ID="v1.99.10"
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-da78428ac543d6d9ac2493dfa77364eb300a47cb #1 Fri Oct 20 20:00:41 UTC 2017 armv6l GNU/Linux
root@witherspoon:~# cat /media/pnor-ro-b0507b6c/VERSION
IBM-witherspoon-ibm-OP9_v1.19_1.68
        op-build-v1.7-1473-g6f195a1
        buildroot-2017.08-8-g5e23247
        skiboot-v5.9-rc3
        hostboot-b01e6bb
        linux-4.13.5-openpower1-pc511380
        petitboot-v1.6.0-p3f89369
        machine-xml-389c56f
        occ-ba4e81e
        hostboot-binaries-da1227e
        capp-ucode-p9-dd2-v2
        sbe-374add4
root@witherspoon:~#
```

- [x] Journald log 20171023140748328367_Myffdc  
[RR_BMC_error.txt](https://github.com/openbmc/openbmc/files/1408497/RR_BMC_error.txt)

gkeishin, 2017-10-23:
Post the above, we can see for the log, the subsequent failures immediately. 
```
root@witherspoon:~# systemctl status pcie-slot-detect@0.service | cat
● pcie-slot-detect@0.service - PCIE Slot Presence Detection For Chassis0
   Loaded: loaded (/lib/systemd/system/obmc-host-start@0.target.wants/../pcie-slot-detect@.service; disabled; vendor preset: enabled)
   Active: failed (Result: signal) since Thu 2017-10-12 12:05:42 UTC; 1min 43s ago
  Process: 2172 ExecStart=/usr/bin/env pcie_slot_present.exe (code=killed, signal=TERM)
 Main PID: 2172 (code=killed, signal=TERM)

Oct 12 12:05:36 witherspoon systemd[1]: Starting PCIE Slot Presence Detection For Chassis0...
Oct 12 12:05:42 witherspoon systemd[1]: pcie-slot-detect@0.service: Main process exited, code=killed, status=15/TERM
Oct 12 12:05:42 witherspoon systemd[1]: Stopped PCIE Slot Presence Detection For Chassis0.
Oct 12 12:05:42 witherspoon systemd[1]: pcie-slot-detect@0.service: Unit entered failed state.
Oct 12 12:05:42 witherspoon systemd[1]: pcie-slot-detect@0.service: Failed with result 'signal'
```

 ```
root@witherspoon:~# systemctl status fsi-scan@0.service | cat
● fsi-scan@0.service - Scan FSI devices
   Loaded: loaded (/lib/systemd/system/fsi-scan@.service; disabled; vendor preset: enabled)
   Active: failed (Result: signal) since Thu 2017-10-12 12:05:43 UTC; 2min 10s ago
  Process: 2170 ExecStart=/usr/bin/env openpower-proc-control scanFSI (code=killed, signal=TERM)
 Main PID: 2170 (code=killed, signal=TERM)

Oct 12 12:05:36 witherspoon systemd[1]: Starting Scan FSI devices...
Oct 12 12:05:43 witherspoon systemd[1]: fsi-scan@0.service: Main process exited, code=killed, status=15/TERM
Oct 12 12:05:43 witherspoon systemd[1]: Stopped Scan FSI devices.
Oct 12 12:05:43 witherspoon systemd[1]: fsi-scan@0.service: Unit entered failed state.
Oct 12 12:05:43 witherspoon systemd[1]: fsi-scan@0.service: Failed with result 'signal
```

gkeishin, 2017-10-23:
Post the above failure, the subsequent failures started appearing as well

```root@witherspoon:~# systemctl status pcie-slot-detect@0.service | cat
● pcie-slot-detect@0.service - PCIE Slot Presence Detection For Chassis0
   Loaded: loaded (/lib/systemd/system/obmc-host-start@0.target.wants/../pcie-slot-detect@.service; disabled; vendor preset: enabled)
   Active: failed (Result: signal) since Thu 2017-10-12 12:05:42 UTC; 1min 43s ago
  Process: 2172 ExecStart=/usr/bin/env pcie_slot_present.exe (code=killed, signal=TERM)
 Main PID: 2172 (code=killed, signal=TERM)

Oct 12 12:05:36 witherspoon systemd[1]: Starting PCIE Slot Presence Detection For Chassis0...
Oct 12 12:05:42 witherspoon systemd[1]: pcie-slot-detect@0.service: Main process exited, code=killed, status=15/TERM
Oct 12 12:05:42 witherspoon systemd[1]: Stopped PCIE Slot Presence Detection For Chassis0.
Oct 12 12:05:42 witherspoon systemd[1]: pcie-slot-detect@0.service: Unit entered failed state.
Oct 12 12:05:42 witherspoon systemd[1]: pcie-slot-detect@0.service: Failed with result 'signal'
```

 ```root@witherspoon:~# systemctl status fsi-scan@0.service | cat
● fsi-scan@0.service - Scan FSI devices
   Loaded: loaded (/lib/systemd/system/fsi-scan@.service; disabled; vendor preset: enabled)
   Active: failed (Result: signal) since Thu 2017-10-12 12:05:43 UTC; 2min 10s ago
  Process: 2170 ExecStart=/usr/bin/env openpower-proc-control scanFSI (code=killed, signal=TERM)
 Main PID: 2170 (code=killed, signal=TERM)

Oct 12 12:05:36 witherspoon systemd[1]: Starting Scan FSI devices...
Oct 12 12:05:43 witherspoon systemd[1]: fsi-scan@0.service: Main process exited, code=killed, status=15/TERM
Oct 12 12:05:43 witherspoon systemd[1]: Stopped Scan FSI devices.
Oct 12 12:05:43 witherspoon systemd[1]: fsi-scan@0.service: Unit entered failed state.
Oct 12 12:05:43 witherspoon systemd[1]: fsi-scan@0.service: Failed with result 'signal```

geissonator, 2017-10-23:
I think the main issue here is that pgood is 0 on the bmc reboot with the host up:
```
Oct 12 12:01:51 witherspoon power_control.exe[1294]: Pgood state: 0
```

Reminds me of bugs we had in the past here where the gpio is not preserved across a BMC reboot.

geissonator, 2017-10-23:
I don't see any obvious changes in this area recently @amboar but gonna send your way first to see if there's anything with the gpio dts configs or preservation across reboots.

amboar, 2017-10-24:
@geissonator I think you're confusing the issue here, at least, I don't see evidence of a BMC reboot, and from @gkeishin 's log extracts I don't think that's what he was testing. I think @gkeishin is complaining that a *host* reboot failed.

Regardless, I can't recreate either problem on w33 (host instability on BMC reboot, or BMC instability on host reboot) with openbmc/openbmc@c9f3377. Possibly some hardware issue on @gkeishin 's machine?

gkeishin, 2017-10-24:
@amboar I can recreate the issue this real quick if you want to take a look at it. This system is running for Master has been good so far.

gkeishin, 2017-10-24:
@amboar and I recreate the issue.  
```
Oct 12 12:05:38 witherspoon systemd[1]: phosphor-reboot-host@0.service: Main process exited, code=exited, status=1/FAILURE
Oct 12 12:05:38 witherspoon systemd[1]: phosphor-reboot-host@0.service: Unit entered failed state.
Oct 12 12:05:38 witherspoon systemd[1]: phosphor-reboot-host@0.service: Failed with result 'exit-code'.
Oct 12 12:05:38 witherspoon systemd[1]: Starting Assert power_on LED...
Oct 12 12:05:38 witherspoon systemd[1]: Starting Soft power off of the host...
Oct 12 12:05:38 witherspoon systemd[1]: Starting OpenPOWER OCC Active Disable...
Oct 12 12:05:38 witherspoon systemd[1]: Stopping Phosphor poweron watchdog...
Oct 12 12:05:38 witherspoon power_control.exe[1279]: PowerControl: pgood: 1, setting reset BMC_CP0_RESET_N to 1
Oct 12 12:05:38 witherspoon witherspoon-psu-monitor[1308]: The power supply indicated that it is not on when it should be.
Oct 12 12:05:38 witherspoon witherspoon-psu-monitor[1309]: The power supply indicated that it is not on when it should be.
Oct 12 12:05:38 witherspoon system_manager.py[1173]: Running System State: HOST_POWERED_ON
Oct 12 12:05:38 witherspoon phosphor-gevent[1150]: ::60f9:6a76:8e91:f402%1995832092 - - [2017-10-12 12:05:38] "PUT /xyz/openbmc_project/state/host0//attr/RequestedHostTransition HTTP/1.1" 200 169 0.963367
Oct 12 12:05:39 witherspoon systemd[1]: Stopping Phosphor GPIO checkstop monitor...
Oct 12 12:05:39 witherspoon systemd[1]: Stopped Witherspoon Fan Watchdog Reset.
Oct 12 12:05:39 witherspoon systemd[1]: Stopped Assert power_on LED.
Oct 12 12:05:39 witherspoon systemd[1]: Stopped Phosphor poweron watchdog.
Oct 12 12:05:39 witherspoon systemd[1]: Stopped Phosphor GPIO checkstop monitor.
Oct 12 12:05:39 witherspoon systemd[1]: Stopped Wait for Power0 to turn on.
Oct 12 12:05:39 witherspoon systemd[1]: witherspoon-pseq-monitor-pgood.service: Main process exited, code=killed, status=15/TERM
Oct 12 12:05:39 witherspoon systemd[1]: Stopped Witherspoon Power Sequencer Power-on Monitor.
Oct 12 12:05:39 witherspoon systemd[1]: witherspoon-pseq-monitor-pgood.service: Unit entered failed state.
Oct 12 12:05:39 witherspoon systemd[1]: witherspoon-pseq-monitor-pgood.service: Failed with result 'signal
```

Turns out this is just because of the host watchdog timeout and we need to understand why the host times out.

`Services`
```
root@witherspoon:~# cat /lib/systemd/system/phosphor-reboot-host@.service
[Unit]
Description=Reboot host%i
Wants=obmc-host-stop@%i.target
After=obmc-host-stop@%i.target

[Service]
ExecStart=/bin/systemctl start obmc-host-start@%i.target

[Install]
WantedBy=obmc-host-reboot@%i.target
```
and

```
oot@witherspoon:~# cat /lib/systemd/system/obmc-host-start@.target
[Unit]
Description=Start Host%i
Wants=obmc-standby.target
After=obmc-standby.target
Wants=mapper-wait@-org-openbmc-control-chassis%i.service
After=mapper-wait@-org-openbmc-control-chassis%i.service
Conflicts=obmc-host-stop@%i.target
RefuseManualStop=yes
OnFailure=obmc-host-quiesce@%i.target
OnFailureJobMode=flush
```

```
root@witherspoon:~# ls -1  /lib/systemd/system/obmc-host-start@0.target.wants
obmc-enable-host-watchdog@0.service
pcie-slot-detect@0.service
phosphor-gpio-monitor@checkstop.service
phosphor-watchdog@poweron.service
```
`Host SOL log`
```
[   75.460578342,5] IPMI: Soft shutdown requested
[   10.646420] reboot: Power down
[   75.492610604,5] OPAL: Shutdown request type 0x0...
```
Host has send down the poweroff request for the reboot host request from BMC.

gkeishin, 2017-10-24:
Attached FFDC and bus.log zip for ur reference on the recreate

- [x] Bus log
[bus.log.gz](https://github.com/openbmc/openbmc/files/1410048/bus.log.gz)

- [x] FFDC 
[20171024081813093969_MyFFDCLogs.zip](https://github.com/openbmc/openbmc/files/1410052/20171024081813093969_MyFFDCLogs.zip)

amboar, 2018-04-16:
@gkeishin is this still relevant?

gkeishin, 2018-04-16:
Well not now.. I no longer see this now..

closed 2018-04-16 by gkeishin